pymodbus-dev/pymodbus #231: Modbus Module not found

oict18, 2017-10-23:
### Versions

* Python: 3.5.3
* OS: raspbian, rpi 3 b
* Pymodbus: 1.3.2
* Modbus Hardware (if used): 

### Description

(I am new to Python and r-pi's.  Thanks in advance for putting up with a learner)
Looking to do a quick test on a local server before testing to PLC, but not getting very far.  python can't find the pymodbus module.

### Code and Logs

// copied from web, put into python
from pymodbus.client.sync import ModbusTcpClient

client = ModbusTcpClient('127.0.0.1')
client.write_coil(1, True)
result = client.read_coils(1,1)
print (result.bits[0])
client.close()

error thrown is:
Traceback (most recent call last):
File "/home/pi/Desktop/projects/testPy.py", line 1, in <module>
from pymodbus.client.sync import ModbusTcpClient
ImportError: No module named 'pymodbus'

So I thought maybe Python can't find the pymodbus reference in its path.  
pi@raspberrypi:~ $ python
Python 2.7.13 (default, Jan 19 2017, 14:48:08) 
[GCC 6.3.0 20170124] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> print sys.path
['', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-arm-linux-gnueabihf', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/home/pi/.local/lib/python2.7/site-packages', '/usr/local/lib/python2.7/dist-packages', '/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.2-py2.7.egg', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/gtk-2.0']
>>> 

not sure why I'm missing.  Very appreciative of any input... 
```

dhoomakethu, 2017-10-23:
How did you install pymodbus ?

oict18, 2017-10-23:
well... I botched it the first time, then I installed following the instructions at https://github.com/riptideio/pymodbus

git clone git://github.com/bashwork/pymodbus.git
cd pymodbus
python setup.py install

dhoomakethu, 2017-10-23:
You can install with `pip install pymodbus` , which should actually take care of installing the latest stable version  correctly on to your site-packages.

closed 2017-10-23 by dhoomakethu

oict18, 2017-10-23:
executed pip install pymodbus (actually did that previously thinking it
would repair the install)
ran pip freeze and pymodbus==1.3.2 is in the list.

the test program throws the same error:
Traceback (most recent call last):
File "/home/pi/Desktop/project/testPy.py", in line 1, in <module>
from pymodbus.client.sync import ModbusTcpClient
ImportError: No module named 'pymodbus'

program has accessed other modules in same location.  Is there more
required in the import function then *from pymodbus.client.sync import
Modbus?*

Lee Clore
Onyx Industries
651.260.7115

dhoomakethu, 2017-10-23:
That should work, Could you please paste the output of the following commands
```
pip show pymodbus
echo $PYTHONPATH
echo $PATH
```

oict18, 2017-10-23:
something wrong with the path?

pi@raspberrypi:~ $ pip show pymodbus
Name: pymodbus
Version: 1.3.2
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/riptideio/pymodbus/
Author: dhoomakethu
Author-email: otlasanju@gmail.com
License: BSD
Location: /home/pi/.local/lib/python2.7/site-packages
Requires: pyserial, twisted
pi@raspberrypi:~ $ echo $PYTHONPATH

pi@raspberrypi:~ $ echo $PYTHONPATH

pi@raspberrypi:~ $ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
pi@raspberrypi:~ $

oict18, 2017-10-23:
so looking at this, the location is set to:

Location: /home/pi/.local/lib/python2.7/site-packages

again, excuse my lack of experience in this platform please... I've
been looking in /usr/local/python2.7/dist-packages/pymodbus.

learning the pi and raspbian.  is the modules supposed to install in
both locations?  do i need to do something with the path & which
should it point too?

dhoomakethu, 2017-10-23:
just do `export PYTHONPATH=/home/pi/.local/lib/python2.7/site-packages` before running your script. That should fix the issue.

oict18, 2017-10-23:
great, thanks.  I will run that later today