pymodbus-dev/pymodbus #232: Modbus error while communicating with Trio Motion MC403-Z

linhanh123, 2017-10-26:
Hi,

Her is my setup
Server - running on Raspberry Pi, with Python 2.7
Client - running on Trio Motion controller MC403-Z.

When the controller sends requests to PC, it works at first. But after a random period of time, the server displays errors. Sometimes, it display error(1), sometimes it displays error(2), even though the controller just sends the same read/write request all the time.

###Error 1
ERROR:pymodbus.server.sync:Socket exception occurred Traceback (most recent call last):
  File "/home/pi/Desktop/pymodbus/server/sync.py", line 163, in handle
    self.framer.processIncomingPacket(data, self.execute)
  File "/home/pi/Desktop/pymodbus/transaction.py", line 458, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/pi/Desktop/pymodbus/transaction.py", line 470, in _process
    raise InvalidMessageRecievedException(result)
InvalidMessageRecievedException: Modbus Error: [Invalid Message] <pymodbus.pdu.IllegalFunctionRequest object at 0x766c7b30>

###Error 2
ERROR:pymodbus.server.sync:Socket exception occurred Traceback (most recent call last):
  File "/home/pi/Desktop/pymodbus/server/sync.py", line 163, in handle
    self.framer.processIncomingPacket(data, self.execute)
  File "/home/pi/Desktop/pymodbus/transaction.py", line 457, in processIncomingPacket
    if self.__header['len'] < 2:
KeyError: 'len'

Any suggestions on how to fix this?

Thank you so much for your help!

dhoomakethu, 2017-10-27:
Are both server and client from pymodbus ? Is this a Modbus RTU issue or Modbus TCP issue ? It would be good if you use the issue template for better assistance.

linhanh123, 2017-10-27:
The server is from pymodbus, the client is from the trioBasic Modbus functions. I'm using Modbus TCP, connecting over an Ethernet cable.
On the server, I use the synchronous server example for the code. On the client, I just open the client connection, and try read/write to the sever.
Does this make the question clearer? Sorry I'm still pretty new to this.

dhoomakethu, 2017-10-28:
@linhanh123 thanks, could you use the pymodbus from dev branch and verify this issue.This could be same as #216 and there is a fix for it in [dev](https://github.com/riptideio/pymodbus/tree/dev) branch. Let me know if you need assistance with building the pymodbus library from dev branch.

linhanh123, 2017-10-30:
@dhoomakethu I download the pymodbus library from the dev branch and replaced the old one with this one. It still shows the same error though...

dhoomakethu, 2017-10-30:
@linhanh123 when you say download, did you git cloned the dev branch and created an installer with `python setup.py sdist` or somethin similar and did a `pip install` of the sdist ?

linhanh123, 2017-10-30:
@dhoomakethu I went into the branch dev, and there's a "Clone or download button", and when I clicked on it, I got the file "pymodbus-dev.zip"

dhoomakethu, 2017-10-30:
Please share the logs with debug enabled.

linhanh123, 2017-10-30:
Here are the logs with the 2 errors.

[Error1.log](https://github.com/riptideio/pymodbus/files/1427125/Error1.log)
[Error2.log](https://github.com/riptideio/pymodbus/files/1427126/Error2.log)

dhoomakethu, 2017-10-30:
do you have the logs after the errors have occured ?

linhanh123, 2017-10-30:
Sorry I don't :/ After the error occurred, there was no more logging.

shayden--, 2018-01-18:
Hello,

Our group is currently using a Trio MC4N-ECAT and we are encountering the same errors. I've taken many packet captures and it appears that the Trio TCP stack will occasionally fragment a Modbus ADU across two TCP segments. If the fragment is an incomplete packet but larger than 0x7, then Error1 occurrs. If the fragments is less than 0x7, it causes Error2.

I forked the repo and created an initial fix: 
https://github.com/shayden--/pymodbus/commit/4fa747a967021cd44c7d944c7a53c68bc333d6b8

I haven't tested this much yet, but will create unit tests, test it further and then create a PR.

Let me know if you want more logs or packet captures to verify my findings.

dhoomakethu, 2018-01-19:
@shayden-- Thanks for the patch, will have a look in to it. Please raise a PR against `dev` branch when ready.