pymodbus-dev/pymodbus #235: Timeout issue with ModbusTCPClient sync client when processing response from write multiple registers and using ModbusRTU framer.

dhewgill, 2017-11-02:
* Python: 3.6.3 [64-bit]
* OS: Windows 10 [64-bit]
* Pymodbus: 1.3.2
* Modbus Hardware (if used): 

### Description
I create a client like so:

```python
from pymodbus.client.sync import ModbusTcpClient
from pymodbus.transaction import ModbusRtuFramer

ip = "127.0.0.1"
port = 7701

client = ModbusTcpClient(ip, port, framer=ModbusRtuFramer)
addr = 411
vals_to_wr = (0x1,)
unit_id = 255

wr = client.write_registers(address=addr-1, values=vals_to_wr, unit=unit_id)
```

Yields the following output:
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0xff 0x10 0x1 0x9a 0x0 0x1 0x2 0xbe 0xef 0xd3 0x22
DEBUG:pymodbus.transaction:Transaction failed. (timed out)
DEBUG:pymodbus.transaction:send: 0xff 0x10 0x1 0x9a 0x0 0x1 0x2 0xbe 0xef 0xd3 0x22
DEBUG:pymodbus.transaction:Transaction failed. (timed out)
DEBUG:pymodbus.transaction:send: 0xff 0x10 0x1 0x9a 0x0 0x1 0x2 0xbe 0xef 0xd3 0x22
DEBUG:pymodbus.transaction:Transaction failed. (timed out)
DEBUG:pymodbus.transaction:getting transaction 2

Ultimately I get a time out error waiting for the response. Note that my device supports both ModbusTCP [on port 502] and ModbusRTU over TCP [on port 7701] communications. This works fine if I use the ModbusSocketFramer [on port 502].

I have tracked it down to transaction.ModbusTransactionManager._recv. With my own debug messaging I have found that the response message was actually received, but _recv doesn't know to quit because it's expecting up to 1024 bytes and the subsequent _recv on the socket times out.
I notice that in _recv you break from the receive loop if you find a ModbusSocketFramer.
I fixed this locally by adding another break condition in _recv:
```python
if isinstance(self.client, pymodbus.client.sync.ModbusTcpClient) and isinstance(self.client.framer, ModbusRtuFramer):
    break
```

dhoomakethu, 2017-11-03:
Why do you want to use RTU framer with TCP client? `ModbusSocketFramer` is the one that needs to be used.

dhewgill, 2017-11-03:
The device that I am interfacing with supports the usual ModbusTCP as well as the gateway-like ModbusRTU over TCP where a full ModbusRTU request is wrapped in a TCP frame, then unwrapped by the device and forwarded along unmodified over a serial line. For this, I need the RTU framer.

dhoomakethu, 2017-11-03:
I am sorry, pymodbus does not support that feature as of now,  you can do that with a modbus server (refer https://github.com/riptideio/pymodbus/blob/master/examples/contrib/serial-forwarder.py)

dhoomakethu, 2017-11-03:
You can override the Socket framer to meet your needs .

closed 2017-11-03 by dhoomakethu

dhewgill, 2017-11-03:
Thanks for the info.