pymodbus-dev/pymodbus #238: help wanted

dmdelorme, 2017-11-15:


dhoomakethu, 2017-11-16:
@dmdelorme what kind of help are you seeking?

dmdelorme, 2017-11-16:
woops i for got to include the info
it seems ever other time i run the code it has an issue connecting. see debug i included 2 debug outputs when i ran the code twice. 
pymodbus ver 1.3.2
OC Linux openplotter 4.9.13-v7+ #974 SMP Wed Mar 1 20:09:48 GMT 2017 armv7l GNU/Linux
On a RPi3
python version 2.7
the Modbus device is a epsolar tracer A
The Code
```
#import logging
#logging.basicConfig()
#log = logging.getLogger()
#log.setLevel(logging.DEBUG)
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
#from pymodbus.mei_message import *
import socket
client = ModbusClient(method='rtu', port='/dev/ttyXRUSB0', baudrate=115200, stopbits = 1, bytesize = 8, timeout = 1)
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
client.connect()
#request = ReadDeviceInformationRequest(unit=1)
#response = client.execute(request)
#print repr(response.information)

result = client.read_input_registers(0x3100,10,unit=1)
pvVolts = float(result.registers[0] / 100.0)
pvAmps = float(result.registers[1] / 100.0)
batVolts = float(result.registers[4] / 100.0)
batchgAmps = float(result.registers[5] / 100.0)
ldVolts = float(result.registers[8] / 100.00)
ldAmps = float(result.registers[9] / 100.00)
ldnetAmps = batchgAmps-ldAmps
result2 = client.read_input_registers(0x3110,1,unit=1)
batTemp = float((result2.registers[0] / 100)+ 273.15)
#print 'send'
SignalK = '{"updates": [{"source": {"type": "CHARGER","src" : "epsolar"},"values":['
SignalK += '{"path": "electrical.chargers.tracer.batteries.main.voltage","value":'+str(round(batVolts, 2))+'}'
SignalK += ',{"path": "electrical.chargers.tracer.batteries.charge.current","value":'+str(batchgAmps)+'}'
SignalK += ',{"path": "electrical.chargers.tracer.pv.voltage","value":'+str(round(pvVolts,2))+'}'
SignalK += ',{"path": "electrical.chargers.tracer.pv.current","value":'+str(pvAmps)+'}'
SignalK += ',{"path": "electrical.chargers.tracer.load.voltage","value":'+str(ldVolts)+'}'
SignalK += ',{"path": "electrical.chargers.tracer.load.current","value":'+str(ldAmps)+'}'
SignalK += ',{"path": "electrical.chargers.tracer.loadnet.current","value":'+str(ldnetAmps)+'}'
SignalK += ',{"path": "electrical.chargers.tracer.batteries.temp","value":'+str(batTemp)+'}'
SignalK += ']}]}\n'
sock.sendto(SignalK, ('localhost', 55559))
#print pvVolts
#print pvAmps
#print result2.registers[0]/100
#print batteryAmps
print (batTemp)
client.close()
```
Debug output
```
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x31 0x0 0x0 0xa 0x7e 0xf1
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x14 0x3 0x58 0x0 0x0
DEBUG:pymodbus.factory:Factory Response[1]
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(40)) 
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x31 0x0 0x0 0xa 0x7e 0xf1
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x14 0x3 0x59 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xe8 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xe8 0x0 0x17 0x92 0x45
DEBUG:pymodbus.factory:Factory Response[1]
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(240)) 
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x31 0x0 0x0 0xa 0x7e 0xf1
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x14 0x3 0x57 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xe8 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xe8 0x0 0x18 0xb8 0xa8
DEBUG:pymodbus.factory:Factory Response[1]
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(440)) 
DEBUG:pymodbus.transaction:getting transaction 1
Traceback (most recent call last):
  File "sktest.py", line 16, in <module>
    pvVolts = float(result.registers[0] / 100.0)
```
debug 2 working
```
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x31 0x0 0x0 0xa 0x7e 0xf1
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x14 0x0 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xe2 0x0 0x0 0x0 0x0 0x0 0x0 0x4 0xe2 0x0 0x27 0x96 0x2d
DEBUG:pymodbus.factory:Factory Response[4]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 1
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x31 0x10 0x0 0x1 0x3e 0xf3
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x2 0x7 0xce 0x3a 0x94
DEBUG:pymodbus.factory:Factory Response[4]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 2
292.15
```

Is there are some out standing issues with pymodbus or is my code bad. do i need to do more cleanup besides client.close()

dhoomakethu, 2017-11-16:
You can start with lower baudrates(9600, 19200, 38400 etc) and increased read timeout . 
`client = ModbusClient(method='rtu', port='/dev/ttyXRUSB0', baudrate=115200, stopbits = 1, bytesize = 8, timeout = 1)
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)`

dmdelorme, 2017-11-16:
tried that baud is non adjustable time out did not help nether. why every every second time?

dhoomakethu, 2017-11-16:
You are receiving incomplete response and incorrect response in your failure case from the device. Is the wirings solid (correct termination installed, no loose connections ?) . Could you share the manual for the device , I could not find any on the net with the modbus specs for the device you have mentioned.

dmdelorme, 2017-11-16:
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] ReadBitResponse(432) i think it is incorrect?
I have tried different ports but no change.
https://github.com/kasbert/epsolar-tracer/blob/master/archive/1733_modbus_protocol.pdf

dhoomakethu, 2017-11-16:
The Invalid Message error is raised when it is expecting a valid response but received a different one after decoding. I will run some manual decoding to see if something is wrong from the pymodbus side , but from the logs it looks like it is not. Let me double confirm.

dmdelorme, 2017-11-17:
I might have a clue i am using a usb /rs485 cable it uses a exar driver.
`xr_usb_serial_common: Exar USB UART (serial port) driver`
I change to this version [https://www.exar.com/design-tools/software-drivers](url) 1.b connects but data is trash every time.
changed to [https://github.com/kasbert/epsolar-tracer](url)  	xr_usb_serial_common-1a
then every other read is garbage:) 100% improvement but works 50% of the time.
I also have a 10 amp Tracer and did the test with 1A same result.
I ordered a second cable and a wifi device.
the last resort is to run the program twice and take the good value. kinda like flipping a rigged coin. first toss 50/50 second toss 100% its dirty but it could work since the issue is so repeatable. 
I keep thinking that the stream is out of alignment and the error resets everything. byte alignment?
Is there a way i can connect to the device and just stream raw data out of it? I know $hit .....
did some more testing. got it read every time. when i just read 1 value? there are some blank/undocumented registers in the array could this be my problem?. also i have some issues with 0/100 when  volts/amps are 0... 

Thanks for your help?

dhoomakethu, 2017-11-17:
@dmdelorme please refer #239, which is similar to what you are facing , the issues could be due to electro magnetic interference as well. Please check and revert

dmdelorme, 2017-11-17:
I will look into that. I am not dealing with a lot of current  but have seen a few references to this issue. 
Thanks

dmdelorme, 2017-11-19:
yup it looks like it was some loose wires causing noise. not 100% yet but better.

closed 2017-11-19 by dmdelorme