pymodbus-dev/pymodbus #239: Sync read_input_registers() from serial RTU device gives random results

fluffyfury, 2017-11-16:
### Versions

* Python: 2.7.9
* OS: Raspbian Jessie
* Pymodbus: 1.3.2
* Modbus Hardware (if used): Raspberry Pi with LB-LINK RB-RS485 board + Smartprocess SMARTRAIL X835

### Description

I am trying to read the input registers of a power meter and I noticed that as I increased the "count", errors started to appear. So asking for 2 registers seemed fairly reliable, but when I tried 6 or 12, it sometimes gives me the values; but more often than not, it will spit out random errors. See some examples below.

### Code and Logs

`import sys

address = int(sys.argv[1])
count = int(sys.argv[2])

from pymodbus.client.sync import ModbusSerialClient as ModbusClient

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient(method = "rtu", port="/dev/ttyAMA0", stopbits = 1, bytesize = 8, parity = 'N', baudrate = 9600)
connection = client.connect()
print (connection)

result = client.read_input_registers(address, count, unit=0x01)
print (result)

client.close()`

Here I got the "ReadBitResponse(%d)" several times with various numbers:

`pi@raspberrypi:~/AUX/modbus $ python3 ReadWrite_Client.py 0 12
Address =  0
Data = 12
True
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x0 0x0 0x0 0xc 0xf0 0xf
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x18 0x43 0x6c 0x56 0xa8 0x43 0x9f 0x6d 0x55 0x0 0x0 0x0 0x0 0x3e 0xec 0xaa 0xaa 0x61 0xa4
DEBUG:pymodbus.factory:Factory Response[1]
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(152))
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x0 0x0 0x0 0xc 0xf0 0xf
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x18 0x43 0x6c 0x5f 0x50 0x43 0x6c 0x6a 0xd8 0x43 0x6c 0x6d 0xa0 0x3e 0xa0 0x1a 0xaa 0x0 0x0 0x0 0x0 0x3e 0xeb 0x78 0x0 0x9b 0xd1
DEBUG:pymodbus.factory:Factory Response[1]
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(384))
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x0 0x0 0x0 0xc 0xf0 0xf
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x18 0x43 0x6c 0x5f 0x50 0x43 0x6c 0x6a 0xd8 0x43 0x6c 0x6d 0xa0 0x3e 0xa0 0x1a 0xaa 0x0 0x0 0x0 0x0 0x3e 0xeb 0x78 0x0 0x9b 0xd1
DEBUG:pymodbus.factory:Factory Response[1]
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(616))
DEBUG:pymodbus.transaction:getting transaction 1
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] ReadBitResponse(616)`

Here I got no response at all:

`pi@raspberrypi:~/AUX/modbus $ python3 ReadWrite_Client.py 0 12
True
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x0 0x0 0x0 0xc 0xf0 0xf
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv:
DEBUG:pymodbus.transaction:getting transaction 1
Modbus Error: [Input/Output] No Response received from the remote unit`

And then it will work once in a while:

`pi@raspberrypi:~/AUX/modbus $ python3 ReadWrite_Client.py 0 12
True
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x0 0x0 0x0 0xc 0xf0 0xf
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x18 0x43 0x6b 0xd5 0xc8 0x43 0x6b 0xe0 0xd8 0x43 0x6b 0xe3 0x20 0x3e 0xa7 0x22 0xaa 0x0 0x0 0x0 0x0 0x3e 0xf0 0xe2 0xaa 0xd3 0x73
DEBUG:pymodbus.factory:Factory Response[4]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 1
ReadRegisterResponse (12)`

dhoomakethu, 2017-11-16:
Could you try with an explicit timeout for reads while creating the client ?
`client = Modbus client(method= "rtu", timeout=3,....)`

fluffyfury, 2017-11-16:
Thank you for looking into this. Yes, I did try timeout = 3 and timeout = 0 in the past. Nothing seemed to change.

Also, a new error I am seeing now when printing result.registers is "Exception Response(%d, %d, None)", again with random numbers.

fluffyfury, 2017-11-17:
I believe I found the issue. I still get errors like "Unable to decode request", "ReadBitResponse()" and "No response received from the remote unit", but a lot less than before. 
The problem is electromagnetic interference. The raspberry pi is connected in a lab, lying near other cables and not using a proper twisted pair shielded connection. By moving the raspberry pi away from other equipment, the reading of the input registers has improved significantly. Hence, I'll close the issue, but I hope others might find this useful during their debugging.

closed 2017-11-17 by fluffyfury

dhoomakethu, 2017-11-17:
@fluffyfury that was what I was about to recommend (check line terminations, ground and cable) ,glad that you figured it out. Thanks for the confirmation