pymodbus-dev/pymodbus #247: Seems impossible to create a MODBUS/TCP server on port 502

ltcasg, 2017-11-30:
Hello

Using pymodbus 1.3.2 on a Raspberry Pi 3 with Raspbian and Python 2.7 installed, I tried to create a MODBUS/TCP server from the "Synchronous Server Example

To be able to communicate with a MODBUS/TCP client I replaced the line:
```python,
StartTcpServer(context, identity=identity, address=("localhost", 5020))
```
with
```python
StartTcpServer(context, identity=identity, address=("192.168.0.42", 502))
```
But unfortunately, launching the modified script generates a "Permission denied" error in socket.py.

If I replace 502 with 5020, the server is functionnal and responds correclty

Is it a known issue in pymodbus, python 2.7 socket.py, ...  ?

Thanks

dhoomakethu, 2017-11-30:
Run as sudo

ltcasg, 2017-11-30:
Thanks for your answer

But it seems I have another issue when I try to use ```sido```
[image]

dhoomakethu, 2017-11-30:
You probably need to set the `PYTHONPATH` ,so that the python interpreter can find the installed pymodbus package.

closed 2017-12-02 by dhoomakethu

ltcasg, 2017-12-07:
For all having the same issue as mine, here is the way I used to solve it:

In the server script, initialize the server with its IP address and a port > 1024
```
StartTcpServer(context, identity=identity, address=("192.168.0.42", 502))
```

open a LXTerminal window and execute:
```
sudo iptables -t nat -A PREROUTING -i eth0 -p tcp --dport 502 -j REDIRECT --to-port 5020
```
then launch the server script.

Test with a MODBUS/TCP client and result on wireshark
[image]

Server responds correctly and Wireshark is able to recognize MODBUS frames as port 502 is used!