pymodbus-dev/pymodbus #248: Writing a value (float) over holding registers (localhost, 502) using Modbus TCP version

josemariasecurity, 2017-11-30:
### Versions

* OS: Raspbian / Debian

### Description

I would need an example to write a value (float) over holding registers (localhost, 502) using Modbus TCP version. I did not find this example. Please, could you give me a simple example?

dhoomakethu, 2017-11-30:
Please refer to example https://github.com/riptideio/pymodbus/blob/master/examples/common/modbus-payload.py

josemariasecurity, 2017-12-01:
Thank you dhoomakethu.

Regards,

closed 2017-12-01 by dhoomakethu

josemariasecurity, 2017-12-14:
Hi!

Firstly, congratulations for your library.

Sorry, I integrated some lines of your example (https://github.com/riptideio/pymodbus/blob/master/examples/common/modbus-payload.py) to write a float. But I have the following error. The output (script) is as follows.

Received data from USB: 25.3486633300
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x1e
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x1e
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 1
DEBUG:pymodbus.transaction:getting transaction 1
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x1
DEBUG:pymodbus.transaction:recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xa
DEBUG:pymodbus.transaction:0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xa
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 2
DEBUG:pymodbus.transaction:getting transaction 2
DEBUG:pymodbus.transaction:Running transaction 3
Traceback (most recent call last):
  File "pymodbus-USB-temperature.py", line 105, in <module>
    rq = client.write_register(2, payload, skip_encode=True, unit=1)
  File "build/bdist.linux-armv7l/egg/pymodbus/client/common.py", line 87, in write_register
  File "build/bdist.linux-armv7l/egg/pymodbus/client/sync.py", line 85, in execute
  File "build/bdist.linux-armv7l/egg/pymodbus/transaction.py", line 125, in execute
  File "build/bdist.linux-armv7l/egg/pymodbus/transaction.py", line 498, in buildPacket
  File "build/bdist.linux-armv7l/egg/pymodbus/register_write_message.py", line 40, in encode
TypeError: cannot concatenate 'str' and 'list' objects

My code (script) is as follows.

#!/usr/bin/env python

from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadBuilder
  
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

import serial

ser = serial.Serial('/dev/ttyUSB0', 115200,timeout=.5) 

client = ModbusClient('localhost', port=502)
client.connect()
  
while True:
    incoming = ser.readline().strip()
    if len(incoming) > 5:
    	print 'Received data from USB: '+ incoming
	
	rr = client.read_holding_registers(0, 1, unit=1)
	rr2 = client.read_holding_registers(1, 1, unit=1)
	#client.read_holding_register(address, count, unit)
    
	builder = BinaryPayloadBuilder(endian=Endian.Big)
	builder.add_32bit_float(float(incoming))
	payload = builder.build()
	rq = client.write_register(2, payload, skip_encode=True, unit=1)
	#client.write_register(address, value, unit)		# comment
	#rr3 = client.read_holding_registers(2, 1, unit=1)
	#assert(rq.function_code < 0x80)     			# test that we are not an error
	#assert(rr3.registers[0] == int(float(incoming)))      	# test the expected value
	
	print 'Write to a holding register and read back'
	print 'Read holding register value (superior limit): '
	print rr.registers[0]
	print 'Read holding register value (inferior limit): '
	print rr2.registers[0]
	print 'Write holding register value (temperature value): '
	print float(incoming)
	print '\n'
    
client.close()

dhoomakethu, 2017-12-15:
Hello @josemariasecurity sorry for the delayed response. You are using `write_register` method from the client instead of `write_registers`. `write_register` is to be used only when writing a single register (16 bits/2 bytes max) , you are creating a 32 bit float which basically is 2 registers (32 bits/4 bytes).
Here is the change you need to make

`rq = client.write_registers(2, payload, skip_encode=True, unit=1)`