pymodbus-dev/pymodbus #250: To scrape a real PLC in realtime and use that data to create NEW updating server

MazoSergey, 2017-12-05:
### Versions

* Python: 2.7
* OS: Linux
* Pymodbus: 1.3.2
* Modbus Hardware (if used): RSIM 

### Description

I am trying to make a clone of a real PLC with a help of Scraper example, and then use that data to create MODBUS server... If i understand correct I have to: scrape  a real PLC --> save the data to file --> create server --> load data from file to server. It looks like I do not understand how this staff works :( If it is possible please help me, may be some examples of this kind of data flow. Thank you!

### Code and Logs

```python
 # code and logs here.

```

dhoomakethu, 2017-12-05:
@MazoSergey what exactly is the problem ? Did you run the scraper example agains the PLC and encountered any error ?

MazoSergey, 2017-12-05:
@dhoomakethu thank you for support! Yes, I tried to run example, I've got a file (dump of data from PLC). Now I am trying to run "Modbus Simulator Example" with -c option, and getting "Configuration Error: Invalid Configuration" :(

dhoomakethu, 2017-12-05:
Could you upload the saved the configuration here please ?

MazoSergey, 2017-12-05:
For sure! Here it is:
[conf.txt](https://github.com/riptideio/pymodbus/files/1531316/conf.txt)

rahulraghu94, 2017-12-06:
Hello @MazoSergey,

Could you please try [this branch](https://github.com/riptideio/pymodbus/tree/%23250-fix-scraper-simulator) on your config file?
I contains a fix that worked. Please let us know :)

MazoSergey, 2017-12-06:
Hello @rahulraghu94,
Thank you very much guys for your support. I tried to use an updated modbus-simulator. Now it is started, but it seems that simulator does not take correct date from map dump. I've make a dump of PLC with 400001 = 111, 400002 = 222, but in simulator i see only zeros :(

rahulraghu94, 2017-12-22:
Hello @MazoSergey , can you provide code excerpts, output and logs appended to the problem you're facing?

MazoSergey, 2017-12-23:
Again thanks guys! Tomorrow i will be on the work and will provide all needed information...

MazoSergey, 2017-12-26:
Hello and sorry for delay guys. So I am trying to use files from #250-fix-scraper-simulator brunch: `pymodbus/examples/contrib/modbus-scraper.py -c data -s 172.27.248.33` against MOD_RSim Version 8.20. After that I am executing `pymodbus/examples/contrib/modbus-simulator.py -c data -D` and trying to connect with ModScan32. But as the result I did success to connect to the simulator but all data is "0". Can you explain please what else i should to provide for you? Thank you very much!
[scraper_log.txt](https://github.com/riptideio/pymodbus/files/1587508/scraper_log.txt)
[server_log.txt](https://github.com/riptideio/pymodbus/files/1587509/server_log.txt)
[data.txt](https://github.com/riptideio/pymodbus/files/1587510/data.txt)

MazoSergey, 2017-12-26:
After I tried to scrape a real PLC I've got an error: 

> Unhandled error in Deferred

[scraper_log (real_PLC).txt](https://github.com/riptideio/pymodbus/files/1587516/scraper_log.real_PLC.txt)

rahulraghu94, 2017-12-26:
@MazoSergey , thanks for the information! I'll have a closer look ASAP and get back to you :)

rahulraghu94, 2018-01-08:
Hello @MazoSergey,

I'm using the config file you have uploaded here and am checking few of the values. They seem to be holding values. Are you using the message-parsers unser `examples/contrib/message-parser.py` to decode your message?

```
(modbus) [~/pymodbus/examples/contrib]$ ./message-parser.py -b -t -p tcp -m "0x0 0xc 0x0 0x0 0x0 0x13 0x1 0x3 0x10 0x0 0xa 0x0 0xa 0x0 0xa 0x0 0xa 0x0 0xa 0x0 0xa 0x0 0xa 0x0 0xa"                            *[#250-fix-scraper-simulator]
================================================================================
Decoding Message 000c00000013010310000a000a000a000a000a000a000a000a
================================================================================
ServerDecoder
--------------------------------------------------------------------------------
ClientDecoder
--------------------------------------------------------------------------------
name            = ReadHoldingRegistersResponse
protocol_id     = 0x0
unit_id         = 0x1
skip_encode     = 0x0
registers       =
                . [10, 10, 10, 10, 10, 10, 10, 10]
check           = 0x0
transaction_id  = 0xc
documentation   =
    This function code is used to read the contents of a contiguous block
    of holding registers in a remote device. The Request PDU specifies the
    starting register address and the number of registers. In the PDU
    Registers are addressed starting at zero. Therefore registers numbered
    1-16 are addressed as 0-15.
```

closed 2018-01-11 by rahulraghu94