pymodbus-dev/pymodbus #256: Modbus TCP Transaction ID >255

Kudie86, 2018-01-04:
### Versions

* Python: 2.7
* OS: 
* Pymodbus: 1.4

### Pymodbus Specific
* Server: tcp - async
* Client: tcp - sync

### Description

Hello,

I have a async modbus TCP server running on a RaspberryPi and a windows computer as a sync client. With pymodbus 1.4.0 the server stops responding after several time, which is not the case with pymodbus 1.3.2. I checked with wireshark and found out that the server does not respond to Transaction IDs larger than 255. When I start a new client, which starts with Transaction ID 1, the server is again running until Transaction ID 255.

Best regards,
Jörg

dhoomakethu, 2018-01-04:
Could you please attach the debug logs for both client and server.

Kudie86, 2018-01-04:
[logging_client.log](https://github.com/riptideio/pymodbus/files/1604001/logging_client.log)
[logging_server.log](https://github.com/riptideio/pymodbus/files/1604002/logging_server.log)

Attached the logs.
The client has a timeout at transaction 256 (row 1793). The position where the server stops is in row 2983 in the server log.
I use a single thread (threading library) to read data out of the data store in the server and put it to a digital-analog-converter.

dhoomakethu, 2018-01-04:
Got it thanks, I could reproduce the b issue and found out the reason. Will give you a patch by tomorrow

dhoomakethu, 2018-01-05:
@Kudie86 Please use this branch and see if it fixes your issue.
https://github.com/riptideio/pymodbus/tree/%23256-Fix-TCP-Server-Transactions

Kudie86, 2018-01-05:
Great, it is working now! Thanks!

closed 2018-01-05 by Kudie86