pymodbus-dev/pymodbus #257: PEP-8 Compliant code

dhoomakethu, 2018-01-05:
Make Pymodbus PEP-8 compliant

rahulraghu94, 2018-01-05:
Yikes. This going to take a while :P

swamper123, 2021-12-08:
What can be done is throwing [black](https://black.readthedocs.io/en/stable/) on it (but pls. with 120 chars per line, unless you realy want that stuff shall looks nice on the terminal), see what stuff was changed (and if tests are still running fine) and merge a PR.
Then add flake8 tests on the Github Actions, kick all PRs again in Github Actions and newer tests have to pass it as well.

swamper123, 2021-12-08:
In the same breath we may add sth. that the coverage will not drop and contributers are forced to add tests for new PRs.

janiversen, 2022-05-22:
This is now solved with 3.0.0dev4.

closed 2022-05-22 by janiversen