pymodbus-dev/pymodbus #261: Serial Server RS485 - Interrupted receivemessage

ruwido, 2018-01-09:
Hi,

i am simulating a modbus slave on my computer with pymodbus. The slave is connected over a usb to 485 converter to a real modbus master.

 ### Versions
* Python: 3.4
* OS: Win7 Pro
* Pymodbus: 1.4
* Modbus Hardware (if used): real master

### Pymodbus Specific
* Server: serial - sync

### Description
Everything works fine, but if i simulate a slave with more datapoints (e.g. 24 32bit input register = 48 modbus register) i got this error:
[image]

I have a socket timeout of 0.2s. I think this is really short, but if i am use 1 oder 2s i become many more crc errors.
The whole setting and the hardware works very fine with tcp. No problems. Also with a real modbus slave there are no probs. If i simulate one or many slaves with less then 10 input registers there is also no ERROR message.

As i can expect, the received message broke after two byte. So it cannot be unpacked. In the next receive the message is sent again. Is the receiver going in an timeout?

Please help me...

### Code and Logs
The whole program has more then 2500 lines (with GUI), so i cannot post it completely
Here is my server object:
[image]
Here is the datablock:
[image]
..and the server thread:
[image]

rahulraghu94, 2018-01-10:
@ruwido , could you provide more information on how you're generating the slaves (`self.slaves`) while creating the `ModbusServerContext`?

ruwido, 2018-01-10:
Thanks for your answere!

Her is a Screenshot:
[image]

I have a dict with single datablocks. Cause i am simulating more slaves on one com-port i put the dicts from every slave in a list (self.list_DataBlockDisc).
Also i have a list with all Slave id's.
In the Screenshot you can see that i am iterating over the lists to generate the slaves.

closed 2021-03-08 by github-actions[bot]