pymodbus-dev/pymodbus #272: pymodbus 1.3.2 source not available on PyPI

duliba, 2018-02-06:
### Versions

* Python: pypy 5.6.0
* OS: MacOS Sierra
* Pymodbus: 1.3.2

### Pymodbus Specific
pymodbus version 1.3.2

### Description
I was trying to download the source for pymodbus 1.3.2 via pip, but apparently it's not available in the worldwide PyPI index servers. Here's what I wanted to do and failed:
pypy -m pip download --no-binary :all: pymodbus==1.3.2

To work around it, I'm using this:
pypy -m pip download --no-binary :all: -e git+https://github.com/riptideio/pymodbus.git@v1.3.2#egg=pymodbus

I expect the first command to be successful once the pymodbus package source is uploaded there.

dhoomakethu, 2018-02-06:
It is available on pypi https://pypi.python.org/pypi/pymodbus/1.3.2 .

duliba, 2018-02-06:
It's available there, but it's not the source version - it's a wheel (binary). Earlier versions, like pymodbus 1.2.0, have a tar.gz file instead (which is source code). It doesn't have to do with pypy compatibility, we're just talking about downloading the package.

dhoomakethu, 2018-02-07:
@duliba thanks for the clarification, starting from 1.3.x we are only uploading universal wheels to pypi. You can download the source from the [releases](https://github.com/riptideio/pymodbus/releases) . Is there a reason why you need the source . Let me know if you are having any issues using the wheel.

dhoomakethu, 2018-02-09:
@duliba I will be closing this issue. Please feel free to re-open if you don't find the answer satisfactory.

closed 2018-02-09 by dhoomakethu

yegorich, 2018-03-13:
We are trying to integrate this package into [Buildroot](https://buildroot.org/). Buildroot doesn't use pip, but downloads source code (PyPI tar.gz, git etc.) and performs `python setup.py install`. The whole dependencies  management occurs in Buildroot's own recipe files. This is needed because of the cross-compilation framework. So putting a tarball on PyPI would help us a lot.

dhoomakethu, 2018-03-13:
@yegorich will you not be able to use the source distribution from [pymodbus github release ](https://github.com/riptideio/pymodbus/releases) ? I see from this [example](https://github.com/rancher/buildroot/blob/master/docs/manual/adding-packages-python.txt) , All you need is a path on the web from where you need to download the source tar ball. I am not so familiar with buildroot ,so I might be missing some obvious things with what you are asking, please correct me if I am wrong.

yegorich, 2018-03-13:
@dhoomakethu it is possible to use github, but PyPI approach is more straight forward. So far I can see the vast majority of the Python packages distributes the source code in a tarball (tar.gz).

dhoomakethu, 2018-05-03:
@yegorich , @duliba the source dist is also available on pypi, starting with [v1.5.0](https://pypi.org/project/pymodbus/#files)

yegorich, 2018-05-03:
@dhoomakethu Great! Thanks.

FYI: the 1.4.0 is already in Buildroot.

duliba, 2018-05-03:
Thank you :) that's awesome