pymodbus-dev/pymodbus #274: field device hangs - physical reset required.

timothyarnold1983, 2018-02-20:
### Versions

* Python: 2.6
* OS: Linux (Centos 7)
* Pymodbus: 1.1.0
* Modbus Hardware (if used):  Starline CPM

### Pymodbus Specific
* Server: tcp - sync
* Client: rtu/ascii - sync

### Description

We have been successfully monitoring a significant number of modbus devices using Moxa TCP/IP Gateways and RTU Field units. We've recently integrated starline CPM modules and we've found after a random period of time these devices are "locking up" and will no longer respond to any modbus requests. The only option is to physically reset the devices and will start working again.

This is not isolated to a single string as we have multiple affected, and not isolated to individual units however, some units will reset sooner, than others.

The vendor has our code, and will be testing in their lab, but wanted to check if someone has seen this type of issue before? As stated before we've not seen any issue on any other vendor's equipment so suspect it's something odd within the field device.

Thanks!

dhoomakethu, 2018-02-21:
@timothyarnold1983 Are you having problems with both pymodbus client and server ? Any logs would be helpful.

Also note, you are using a very old version of pymodbus (1.1.0) and unsupported version of python (2.6). You can still go ahead and use pymodbus 1.3.2 but it is not tested on 2.6 so things can break.

timothyarnold1983, 2018-02-21:
The only symptom is the remote field device is hanging at random times and is unrecoverable - simply no longer responds to any Modbus requests  until physically reset. We have 250+ other manufacturers field devices and have had zero issues!

Noted on the old version and we will test the new version at some point. It's a case that it currently works. We are successfully polling 5000+ registers every minute and pymodbus has never missed a beat... If it's not broke.. :-)

dhoomakethu, 2018-02-21:
@timothyarnold1983 Thanks, I will just keep this issue open for some more time . Please share any logs if available.

timothyarnold1983, 2018-03-19:
We've been progressing this issue with the device manufacturer and they have fixed the issue their side - disabling their receiver when transmitting the response. Apparently the device was getting a buffer overflow of some description.

As suspected, no issue with the fantastic pymodbus stack.!

dhoomakethu, 2018-03-19:
@timothyarnold1983 thanks for the confirmation, will close this issue.

closed 2018-03-19 by dhoomakethu