pymodbus-dev/pymodbus #276: version 1.4.0 TCP connection closed after 256 messages

muhlbaier, 2018-03-03:
### Versions

* Python: 2.7.14
* OS: Windows 7 64 bit
* Pymodbus: 1.4.0 (this issue is introduced with 1.4.0)
* Modbus Hardware (if used): none, TCP / localhost

### Pymodbus Specific
* Server: tcp - async
* Client: tcp - async

### Description

I have a client that connects to a server and reads a bunch of registers using:

# count is between 1 and 4
rr = client.read_input_registers(address, count, unit=1)

On the 256th transaction the connection is closed. This was not the case for the same exact code using pymodbus 1.3.x.

I added the following workaround to my code and it resumed expected functionality:

### Code and Logs

```python
MODBUS_140_WORKAROUND = True
transaction_count = 0
    for something in somethingelse:
        if MODBUS_140_WORKAROUND:
            transaction_count += 1
            if transaction_count > 200:
                client.close()
                time.sleep(0.1)
                client = ModbusClient('localhost', port=5020)
                client.connect()
                transaction_count = 0
```
# last few log messages on the server:
DEBUG:pymodbus.server.async:send: 00fd000000050103020000
DEBUG:pymodbus.server.async:0x0 0xfe 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0xd5 0x0 0x1
DEBUG:pymodbus.transaction:0x0 0xfe 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0xd5 0x0 0x1
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 726:1
DEBUG:pymodbus.datastore.context:getValues[3] 726:1
DEBUG:pymodbus.server.async:send: 00fe000000050103020000
DEBUG:pymodbus.server.async:0x0 0xff 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0xd6 0x0 0x1
DEBUG:pymodbus.transaction:0x0 0xff 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0xd6 0x0 0x1
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 727:1
DEBUG:pymodbus.datastore.context:getValues[3] 727:1
DEBUG:pymodbus.server.async:send: 00ff000000050103020000
DEBUG:pymodbus.server.async:0x1 0x0 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0xe1 0x0 0x2
DEBUG:pymodbus.server.async:Client Disconnected: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.]
DEBUG:pymodbus.server.async:Client Connected [IPv4Address(TCP, '127.0.0.1', 5020)]

dhoomakethu, 2018-03-04:
This is same as #256 and #260 . A fix is available on this [branch](https://github.com/riptideio/pymodbus/tree/%23256-Fix-TCP-Server-Transactions) . I will try to get this in to the next release asap. Please give a try on the above mentioned branch . I will be closing this issue as of now .

closed 2018-03-04 by dhoomakethu

muhlbaier, 2018-03-07:
Sorry about that, I didn't look through the closed issues so I didn't notice #256 or #260. Thanks for the fix.