pymodbus-dev/pymodbus #281: ModbusTCP  sync client return ModbusIOException object instead of raising it

memetb, 2018-03-15:
### Versions

* Python: 3.6.0 (anaconda)
* OS: linux debian 4.9.0-5
* Pymodbus: 1.4.0
* Using Modsim32 as master

### Pymodbus Specific
* Client: tcp - sync

### Description

When running modbus tcp client in sync mode, calling `client.read_holding_registers(...)` with the underlying connection broken results in a *return* value of `ModbusIOException` instead of an actual exception being raised. 

### Code and Logs

```python
 # code and logs here.

while True:
  response = self.client.read_holding_registers(self._start, self._length, unit=self.unit)

  log.info(type(response))

  raw_data = response.registers
  function_code = response.function_code
  log.info( "function_code: %d" % function_code)

```

log output:

```INFO:Modbus@client:<class 'pymodbus.register_read_message.ReadHoldingRegistersResponse'>
INFO:Modbus@client:function_code: 3
INFO:Modbus@client:class 'pymodbus.register_read_message.ReadHoldingRegistersResponse'>
INFO:Modbus@client:function_code: 3
INFO:Modbus@client:class 'pymodbus.exceptions.ModbusIOException'>
WARNING:Modbus@client: ModbusIOException' object has no attribute 'registers'```
```

From casual inspection of code, I believe at the very least [the source code here](https://github.com/riptideio/pymodbus/blob/4ed1f1a9750bc931c100ed24f5cc0fc032ab5f4d/pymodbus/register_read_message.py#L133) should be:

```python
if not (1 <= self.count <= 0x7d):
  raise self.doException(merror.IllegalValue)
if not context.validate(self.function_code, self.address, self.count):
  raise self.doException(merror.IllegalAddress)
```

dhoomakethu, 2018-03-19:
@memetb This was done to address #175 .

memetb, 2018-03-19:
Understood. It would seems to me that the solution to #175 was the raise an exception instead of returning NoneType. Returning an exception type is not very pythonic. 

I have worked around the issue in my code for the time being.

closed 2018-03-19 by memetb

dhoomakethu, 2018-03-19:
@memetb I really don't remember the reason why we went with returning the ModbusException object.  But since this is introduced in 1.3.x ,changing this might break existing behaviour . I will think about this when releasing 2.x series of pymodbus. But as of now there won't be any change in the existing behaviour.

memetb, 2018-03-20:
No problem @dhoomakethu: I completely understand the backward compatibility issue.

techietoday, 2019-05-15:
> Understood. It would seem to me that the solution to #175 was the raise an exception instead of returning NoneType. Returning an exception type is not very pythonic.
> 
> I have worked around the issue in my code for the time being.

I have been facing the same error, how did you fix it?

Ansari28, 2022-07-27:
> > Understood. It would seem to me that the solution to #175 was the raise an exception instead of returning NoneType. Returning an exception type is not very pythonic.
> > I have worked around the issue in my code for the time being.
> 
> I have been facing the same error, how did you fix it?

Even i am also facing same issue. if you found any solution please let me know about it. i am  struggling with this error since a very long time.

janiversen, 2022-07-27:
It has been solved on dev for a very long time.

Ansari28, 2022-07-27:
> It has been solved on dev for a very long time.

i am newbie in python, can you elaborate me how can i fix this problem.
Thank You!!!

janiversen, 2022-07-27:
use dev instead 2.5.3