pymodbus-dev/pymodbus #282: Pymodbus Sync Server - Config Issue & Not able to connect to Client Port

kcsekhar14, 2018-03-27:
### Versions

* Python: 3.5
* OS: Ubuntu
* Pymodbus: Latest
* Modbus Hardware (if used): No

### Pymodbus Specific
* Server: tcp - sync

### Description

I am using the Pymodbus server application to communicate with the client. I my client i want to read first 16 words and write next 16 words of 32 length data. 

1.  How can i input the which data variable are read type (function code = 4) and which variable are write type (function code = 16).
2. Pymodbus sync server is keep on scanning for the connection port (it has identified the client address), even through i have mentioned the port number.

> DEBUG:pymodbus.server.sync:Client Disconnected [100.168.96.17:1170]
> DEBUG:pymodbus.server.sync:Started thread to serve client at ('100.168.96.17', 1171)

3. There is error throwing continuously when running the software

> AttributeError: 'dict' object has no attribute 'validate'  

### Console Log
```
ERROR:pymodbus.pdu:Exception Response F(4) E(4)
DEBUG:pymodbus.server.sync:send: b'000000000003018404'
DEBUG:pymodbus.server.sync:
DEBUG:pymodbus.transaction:
DEBUG:pymodbus.server.sync:Client Disconnected [100.168.96.17:1170]
DEBUG:pymodbus.server.sync:Started thread to serve client at ('100.168.96.17', 1171)
DEBUG:pymodbus.server.sync:Client Connected [100.168.96.17:1171]
DEBUG:pymodbus.server.sync:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.transaction:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.factory:Factory Request[4]
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'dict' object has no attribute 'validate'; Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/server/sync.py", line 60, in execute
    response = request.execute(context)
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/register_read_message.py", line 184, in execute
    if not context.validate(self.function_code, self.address, self.count):
AttributeError: 'dict' object has no attribute 'validate'

ERROR:pymodbus.pdu:Exception Response F(4) E(4)
DEBUG:pymodbus.server.sync:send: b'000000000003018404'
DEBUG:pymodbus.server.sync:
DEBUG:pymodbus.transaction:
DEBUG:pymodbus.server.sync:Client Disconnected [100.168.96.17:1171]
DEBUG:pymodbus.server.sync:Started thread to serve client at ('100.168.96.17', 1172)
DEBUG:pymodbus.server.sync:Client Connected [100.168.96.17:1172]
DEBUG:pymodbus.server.sync:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.transaction:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.factory:Factory Request[4]
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'dict' object has no attribute 'validate'; Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/server/sync.py", line 60, in execute
    response = request.execute(context)
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/register_read_message.py", line 184, in execute
    if not context.validate(self.function_code, self.address, self.count):
AttributeError: 'dict' object has no attribute 'validate'

ERROR:pymodbus.pdu:Exception Response F(4) E(4)
DEBUG:pymodbus.server.sync:send: b'000000000003018404'
DEBUG:pymodbus.server.sync:
DEBUG:pymodbus.transaction:
DEBUG:pymodbus.server.sync:Client Disconnected [100.168.96.17:1172]
DEBUG:pymodbus.server.sync:Started thread to serve client at ('100.168.96.17', 1173)
DEBUG:pymodbus.server.sync:Client Connected [100.168.96.17:1173]
DEBUG:pymodbus.server.sync:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.transaction:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.factory:Factory Request[4]
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'dict' object has no attribute 'validate'; Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/server/sync.py", line 60, in execute
    response = request.execute(context)
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/register_read_message.py", line 184, in execute
    if not context.validate(self.function_code, self.address, self.count):
AttributeError: 'dict' object has no attribute 'validate'

ERROR:pymodbus.pdu:Exception Response F(4) E(4)
DEBUG:pymodbus.server.sync:send: b'000000000003018404'
DEBUG:pymodbus.server.sync:
DEBUG:pymodbus.transaction:
DEBUG:pymodbus.server.sync:Client Disconnected [100.168.96.17:1173]
DEBUG:pymodbus.server.sync:Started thread to serve client at ('100.168.96.17', 1174)
DEBUG:pymodbus.server.sync:Client Connected [100.168.96.17:1174]
DEBUG:pymodbus.server.sync:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.transaction:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.factory:Factory Request[4]
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'dict' object has no attribute 'validate'; Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/server/sync.py", line 60, in execute
    response = request.execute(context)
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/register_read_message.py", line 184, in execute
    if not context.validate(self.function_code, self.address, self.count):
AttributeError: 'dict' object has no attribute 'validate'

ERROR:pymodbus.pdu:Exception Response F(4) E(4)
DEBUG:pymodbus.server.sync:send: b'000000000003018404'
DEBUG:pymodbus.server.sync:
DEBUG:pymodbus.transaction:
DEBUG:pymodbus.server.sync:Client Disconnected [100.168.96.17:1174]
DEBUG:pymodbus.server.sync:Started thread to serve client at ('100.168.96.17', 1175)
DEBUG:pymodbus.server.sync:Client Connected [100.168.96.17:1175]
DEBUG:pymodbus.server.sync:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.transaction:0x0 0x0 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x20
DEBUG:pymodbus.factory:Factory Request[4]
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'dict' object has no attribute 'validate'; Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/server/sync.py", line 60, in execute
    response = request.execute(context)
  File "/usr/local/lib/python3.5/dist-packages/pymodbus/register_read_message.py", line 184, in execute
    if not context.validate(self.function_code, self.address, self.count):
AttributeError: 'dict' object has no attribute 'validate'
```

### Code and Logs

```
# --------------------------------------------------------------------------- # 
# import the various server implementations
# --------------------------------------------------------------------------- # 
from pymodbus.server.sync import StartTcpServer

from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

# --------------------------------------------------------------------------- # 
# configure the service logging
# --------------------------------------------------------------------------- # 
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_server():
    
    # ----------------------------------------------------------------------- # 
    # initialize your data store
    # ----------------------------------------------------------------------- #
    block1 = ModbusSequentialDataBlock(0, [888]*32)
    store1 = ModbusSlaveContext(hr=block)
	
	block21 = ModbusSequentialDataBlock(0, [888]*16)
	block22 = ModbusSequentialDataBlock(16, [0]*16)
	store2 = ModbusSlaveContext(hr=block21,ir=block22)
	
    slaves  = {
               0x01: store2,
              }
    context = ModbusServerContext(slaves=slaves, single=True)
    
    # ----------------------------------------------------------------------- # 
    # initialize the server information
    # ----------------------------------------------------------------------- # 
    # If you don't set this or any fields, they are defaulted to empty strings.
    # ----------------------------------------------------------------------- # 
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '1.0'

    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- # 
    # Tcp:
    StartTcpServer(context, identity=identity, address=('0.0.0.0', 255))
    

if __name__ == "__main__":
    run_server()
```

dhoomakethu, 2018-03-27:
@kcsekhar14  what client are you using ? Is it a pymodbus client or from other modbus stack ?

dhoomakethu, 2018-03-27:
@kcsekhar14 your problem is that you are declaring ServerContext with single set to True but you are passing a dict as store. Please change it to 
`context = ModbusServerContext(slaves= store2, single=True)` to get it working.

closed 2018-03-27 by dhoomakethu

kcsekhar14, 2018-03-27:
@dhoomakethu : I am using the third party client. 
I have changed code to  `context = ModbusServerContext(slaves=slaves, single=False)` to make it work.

If i changed my code to `context = ModbusServerContext(slaves=store2, single=True)` to make it work. Does it consider the slaves id's number to communicate ? 
```
 slaves  = {
               0x01: store2,
              }
```
**I want to print the values of block21 and block22, how can i access these variables after server configuration is established and communicating with the client. I am to print in console the values of block21 and block22, so that i can observe the change in values.**

```
block21 = ModbusSequentialDataBlock(0, [888]*16)
block22 = ModbusSequentialDataBlock(16, [0]*16)
```

dhoomakethu, 2018-03-27:
If you want to use the slave id passed as a key to the dict, use `single=False`. What exactly you want to print with your block, if you enable debug logging, they get printed to console by default.

kcsekhar14, 2018-03-27:
According to the Program, Server has to read from Client [0-15] and server is sending data to Client [16-31] of value 323 initialized. I am able to receive from server to client data of value 323, but i am not able to read data from client to server.

### Console is printing below msg continuously :
```
DEBUG:pymodbus.server.sync:0x1 0x3f 0x0 0x0 0x0 0x9 0x1 0xf 0x0 0x0 0x0 0x10 0x2 0xff 0xff
DEBUG:pymodbus.transaction:0x1 0x3f 0x0 0x0 0x0 0x9 0x1 0xf 0x0 0x0 0x0 0x10 0x2 0xff 0xff
DEBUG:pymodbus.factory:Factory Request[15]
DEBUG:pymodbus.datastore.context:validate[15] 1:16
DEBUG:pymodbus.datastore.context:setValues[15] 1:16
DEBUG:pymodbus.server.sync:send: b'013f00000006010f00000010'
DEBUG:pymodbus.server.sync:0x1 0x40 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x10 0x0 0x10
DEBUG:pymodbus.transaction:0x1 0x40 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x10 0x0 0x10
DEBUG:pymodbus.factory:Factory Request[4]
DEBUG:pymodbus.datastore.context:validate[4] 17:16
DEBUG:pymodbus.datastore.context:getValues[4] 17:16
DEBUG:pymodbus.server.sync:send: b'0140000000230104200143014301430143014301430143014301430143014301430143014301430143'
```
### Client Data Log :
[image]

dhoomakethu, 2018-03-27:
If you are trying to write , your need to use function code 0x10 (Write Multiple Registers) , You are currently using 0x0F (Write Coils).

kcsekhar14, 2018-03-27:
Where i have to use function 0x0F - 
```
	block21 = ModbusSequentialDataBlock(0x00, [888]*0x0F)
	block22 = ModbusSequentialDataBlock(0x10, [737]*0x1F)
        store2 = ModbusSlaveContext(hr=block21,ir=block22)
```
hr is Holding register for writing data of function code 16 and ir is input register to read data of function code 4. [0x00, 0x0F] are start and end of data array. Is my understanding is correct ? or i have to input the function code sepeately.

Is there any way to read and write the data values while server is running and print these values online

dhoomakethu, 2018-03-27:
I would suggest you to go through the [specs](http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) first. Your client should send the correct requests , for writing to holding registers you need to send  (0x10) Write Multiple registers request, to read the holding registers send  (0x03) Read Holding Registers and to read input registers  (0x04) Read Input Registers.

kcsekhar14, 2018-03-27:
I understand, but now client and server both are running with out errors. My server script I have already shared initially just few corrections only.

I want to write some loopback function in server itself, something like -
just like read_server_data & write_server_data any function or script with can access the variables. (Client side I am able to read and write it back, I want to do the same on server side also)
```
def loopback :
  x = read_server_data(server_id=1, function_code=4, address=0)
  y = x+1
  write_server_data(server_id=1, function_code=16, address=18,value=y)
if __name__ = __main__
  loopback()
  runserver()
```

dhoomakethu, 2018-03-27:
You can not write from server directly, use pymodbus client in your code to read the server data. Refer example 
https://github.com/riptideio/pymodbus/blob/master/examples/common/synchronous_client.py

kcsekhar14, 2018-03-28:
Actually, I want to replace a third party modbus sever application where I can manipulate the data read and write by server. Similar, I cannot do that using pymodbus.

On the client side anyway, I will do the data manipulation by reading and writing using pymodbus or other client devices.

Let's suppose, I am running a pymodbus server in PC and client as HMI. Here I have send data to HMI to visualize and read commands send by HMI.

dhoomakethu, 2018-03-28:
If you are unable to use pymodbus client to query/write registers on the server, the only option I could think is to access teh ModbusServerContext object directly. You will have to run the server in a separate thread 
```
from pymodbus.server.sync import ModbusTcpServer

server = ModbusTcpServer(context, framer, identity, address, **kwargs)

# to access the blocks for slave 1
store_1 = server.context[1]  

# to read from the block
store_1.getValues(function_code, address, count)

# For ex. store_1.getValues(3, 0, 10) would read 10  holding registers starting at 0

# to write to the block
store_1.setValues(function_code, address, values)
# For ex. store_1.getValues(0x10, 0, [1, 1]) would write to 2 holding registers starting at 0 with value 1
```
For list of function codes refer modbus specs .

kcsekhar14, 2018-03-28:
I have used the following code for testing, this is the latest code now running on server side.

Client side, i am running the third party application to receive the data. i have shared the screen shot of this below.

With the present server code, i am not able to read the data from client and print it on console.

### Client Side
[image]

### Server Code
```
# --------------------------------------------------------------------------- # 
# import the various server implementations
# --------------------------------------------------------------------------- # 
from pymodbus.server.sync import StartTcpServer

from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

from twisted.internet.task import LoopingCall

# --------------------------------------------------------------------------- # 
# configure the service logging
# --------------------------------------------------------------------------- # 
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_server():
    
    # ----------------------------------------------------------------------- # 
    # initialize your data store
    # ----------------------------------------------------------------------- #
    
    #   di=block, co=block, hr=block, ir=block
    #   block = ModbusSequentialDataBlock(0x00, [123]*0x20)
    #   store = ModbusSlaveContext(hr=block)
    
    block1 = ModbusSequentialDataBlock(0x00, [717]*0x0F)
    block2 = ModbusSequentialDataBlock(0x10, [323]*0x1F)
    store2 = ModbusSlaveContext(hr=block1,ir=block2)    

    slaves  = {
                       0x01: store2,
                    }
    
    context = ModbusServerContext(slaves=slaves, single=False)

    #   print(block1.values)
    #   print(block2.values)    
    
    # ----------------------------------------------------------------------- # 
    # initialize the server information
    # ----------------------------------------------------------------------- # 
    # If you don't set this or any fields, they are defaulted to empty strings.
    # ----------------------------------------------------------------------- # 
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '1.0'

    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- # 
    # Tcp:
    # server = StartTcpServer(context, identity=identity, address=('0.0.0.0', 255))

    # start server in a separate thread so that is not blocking
    # server.start_server()

    # to access the blocks for slave 1
    #store_1=server.context[1]

    # to read from the block
    #print("------")
    #print(store_1.getValues(4,0,32))
    
    # to write to the block
    #store_1.setValues(0x0F, 0, [111, 121, 122, 123, 124])
	
    # Type-2 Implementationt
    time = 2
    #loop = LoopingCall(f=updatevalues, a=(context,))
    #loop.start(time, now=True)
    server = StartTcpServer(context, identity=identity, address=('0.0.0.0', 255))
    loop = LoopingCall(f=updatevalues, a=(server,))
    loop.start(time, now=True)

def updatevalues(a):
    print("------------START----------")
    #contxt = a[1]
    contxt = a.context[1] 
    rfuncode = 4
    wfuncode = 15
    slave_id = 0x01
    address = 0x00
    values = contxt[slave_id].getValues(rfuncode, address, count=32)
    print(values)
    contxt[slave_id].setValues(wfuncode, address, values)
    print("-------------END-------------")

if __name__ == "__main__":
    run_server()

```

dhoomakethu, 2018-03-28:
Multiple issues.

1. You are doing `StartTcpServer` which is blocking and the control never reaches to your LoopingCall, You need to use `ModbusTcpServer` and run `server.serve_forever()` in a separate thread to start the server
2. In your `updaetvalues` function `values = contxt[slave_id].getValues(rfuncode, address, count=32)` This would fail, ModbusSlaveContext does not support indexing . You can directly access it `values = contxt.getValues(rfuncode, address, count=32)`. The same applies with `setValues` code in the next line
3. Last but not least, you need to start twisted reactor with `reactor.run()` to get the LoopingCall work.

kcsekhar14, 2018-03-28:
I have tried many possibilities to run function updatevalues while running background function server.serve_forever(). But when ever execution reaches server.serve_forever(), it is not interrupting or not allowing other functions to execute. 

Do you have any idea on how to execute the updatevalues(server) function or while running the server.serve_forever() function in background or any other possibilities.

```
    print("1")
    time = 1
    server = ModbusTcpServer(context, identity=identity, address=('0.0.0.0', 255))
    loop1 = LoopingCall(updatevalues, server)
    loop1.start(0.5, now=True)
    #reactor.callInThread(updatevalues,server)
    #reactor.callLater(2,updatevalues,server)
    #reactor.run()
    #reactor.callWhenRunning(ModbusTcpServer(context, identity=identity, address=('0.0.0.0', 255)))
    #reactor.callWhenRunning(updatevalues,server)
    #reactor.run()
    #print("1")
    #p1 = Process(target=updatevalues,args=server)
    #p2 = Process(target=server.serve_forever, args=())
    #processes = list()
    #processes.append(p1)
    #processes.append(p2)
    #for p in processes:
    #    p.start()
    #updatevalues(server)
    #x = Thread(updatevalues(server))
    #x.start()
    #server.serve_forever()
    #y = Thread(server.serve_forever())
    #y.start()
    Timer(0.5,updatevalues(server))
    Timer(0.5,server.serve_forever())
```

dhoomakethu, 2018-03-28:
```
# --------------------------------------------------------------------------- #
# import the various server implementations
# --------------------------------------------------------------------------- #
from pymodbus.server.sync import StartTcpServer, ModbusTcpServer

from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

from twisted.internet.task import LoopingCall
from twisted.internet import reactor
import threading
# --------------------------------------------------------------------------- #
# configure the service logging
# --------------------------------------------------------------------------- #
import logging

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_server():
    # ----------------------------------------------------------------------- #
    # initialize your data store
    # ----------------------------------------------------------------------- #

    #   di=block, co=block, hr=block, ir=block
    #   block = ModbusSequentialDataBlock(0x00, [123]*0x20)
    #   store = ModbusSlaveContext(hr=block)

    block1 = ModbusSequentialDataBlock(0x00, [717] * 0x0F)
    block2 = ModbusSequentialDataBlock(0x10, [323] * 0x1F)
    store2 = ModbusSlaveContext(hr=block1, ir=block2)

    slaves = {
        0x01: store2,
    }

    context = ModbusServerContext(slaves=slaves, single=False)

    #   print(block1.values)
    #   print(block2.values)

    # ----------------------------------------------------------------------- #
    # initialize the server information
    # ----------------------------------------------------------------------- #
    # If you don't set this or any fields, they are defaulted to empty strings.
    # ----------------------------------------------------------------------- #
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '1.0'

    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- #
    # Tcp:
    # server = StartTcpServer(context, identity=identity, address=('0.0.0.0', 255))

    # start server in a separate thread so that is not blocking
    # server.start_server()

    # to access the blocks for slave 1
    # store_1=server.context[1]

    # to read from the block
    # print("------")
    # print(store_1.getValues(4,0,32))

    # to write to the block
    # store_1.setValues(0x0F, 0, [111, 121, 122, 123, 124])

    # Type-2 Implementationt
    interval = 2
    # loop = LoopingCall(f=updatevalues, a=(context,))
    # loop.start(time, now=True)
    server = ModbusTcpServer(context, identity=identity,
                            address=('0.0.0.0', 5020))
    t = threading.Thread(target=server.serve_forever, daemon=True)
    t.start()
    loop = LoopingCall(f=updatevalues, a=server)
    loop.start(interval, now=True)
    reactor.run()

def updatevalues(a):
    print("------------START----------")
    # contxt = a[1]
    rfuncode = 3
    wfuncode = 16
    slave_id = 0x01
    address = 0x00
    contxt = a.context[slave_id]
    values = contxt.getValues(rfuncode, address, count=32)
    print(values)
    values = [val+1 for val in values]
    contxt.setValues(wfuncode, address, values)
    print("-------------END-------------")

if __name__ == "__main__":
    run_server()
```

And the output I see, Note I am reading and updating holding registers 

```
DEBUG:pymodbus.datastore.context:getValues[3] 1:32
DEBUG:pymodbus.datastore.context:setValues[16] 1:14
------------START----------
[717, 717, 717, 717, 717, 717, 717, 717, 717, 717, 717, 717, 717, 717]
-------------END-------------
DEBUG:pymodbus.datastore.context:getValues[3] 1:32
DEBUG:pymodbus.datastore.context:setValues[16] 1:14
------------START----------
[718, 718, 718, 718, 718, 718, 718, 718, 718, 718, 718, 718, 718, 718]
-------------END-------------
DEBUG:pymodbus.datastore.context:getValues[3] 1:32
DEBUG:pymodbus.datastore.context:setValues[16] 1:14
------------START----------
[719, 719, 719, 719, 719, 719, 719, 719, 719, 719, 719, 719, 719, 719]
-------------END-------------
------------START----------
[720, 720, 720, 720, 720, 720, 720, 720, 720, 720, 720, 720, 720, 720]
-------------END-------------
DEBUG:pymodbus.datastore.context:getValues[3] 1:32
DEBUG:pymodbus.datastore.context:setValues[16] 1:14
DEBUG:pymodbus.datastore.context:getValues[3] 1:32
DEBUG:pymodbus.datastore.context:setValues[16] 1:14
------------START----------
[721, 721, 721, 721, 721, 721, 721, 721, 721, 721, 721, 721, 721, 721]
-------------END-------------
```

kcsekhar14, 2018-03-29:
@dhoomakethu Works like a charm. Now i am able to read and write on both server and client side. Yesterday, i have tried many combinations, to work the both LoopingCall & ModbusTcpServer parallel but not able to converge. 

Thanks a lot for your contribution & time. Your quick response helped me to solve my problem easily.

MaximKolbin, 2021-02-08:
Hello, Thank you for the example. I took your code as a basis, but I redid the update values function. In "class pymodbus.datastore.ModbusSequentialDataBlock" there are functions: getvalue, setvalue. Which allow you to change the data in input registers.
`loop = LoopingCall(f=updatevalues, a=block1)
    loop.start(interval, now=True)
    reactor.run()

def updatevalues(a):
         b= a.getValues(0x00, count=10)
         a.setValues(0x00, 1)

`

aasick, 2022-05-10:
@dhoomakethu I have situation like need to read two different message, first payload (start with 0x64002) contain power meter information and second payload (start with 0x66) contain power information (voltage). below is my code

    block1 = ModbusSequentialDataBlock(64000, dev_info)
    block2 = ModbusSequentialDataBlock(65, dev_info)
    store = ModbusSlaveContext(hr=block1, ir=block2)
    slaves = {0x01: store}
    context = ModbusServerContext(slaves=slaves, single=False)
    StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port='COM5', stopbits=1, bytesize=8, parity='E', baudrate=9600, timeout=1)

below is output log from server:
  2022-05-10 23:50:40,890 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0xfa 0x1 0x0 0x1b
2022-05-10 23:50:40,890 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-05-10 23:50:40,894 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-05-10 23:50:40,898 MainThread      DEBUG    context        :63       validate: fc-[3] address-64002: count-27
2022-05-10 23:50:40,898 MainThread      DEBUG    context        :77       getValues fc-[3] address-64002: count-27
2022-05-10 23:50:40,900 MainThread      DEBUG    sync           :144      send: [ReadHoldingRegistersResponse (27)]- b'0103362a00374b4d323230302d32454134302d3148413100004c514e2f32313032303537323032383900025603001100000000f600000000005b6c'
2022-05-10 23:50:41,908 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x0 0x41 0x0 0x2
2022-05-10 23:50:41,909 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-05-10 23:50:41,913 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-05-10 23:50:41,914 MainThread      DEBUG    context        :63       validate: fc-[3] address-66: count-2
2022-05-10 23:50:41,918 MainThread      ERROR    pdu            :107      Exception Response(131, 3, IllegalAddress)
2022-05-10 23:50:41,920 MainThread      DEBUG    sync           :144      send: [Exception Response(131, 3, IllegalAddress)]- b'018302c0f1'

aasick, 2022-05-11:
@dhoomakethu any idea on this

dhoomakethu, 2022-05-11:
Try using hex decimal values instead of decimal
```
block1 = ModbusSequentialDataBlock(64000, dev_info)
block2 = ModbusSequentialDataBlock(65, dev_info)
```
to
```
block1 = ModbusSequentialDataBlock(0x64000, dev_info)
block2 = ModbusSequentialDataBlock(0x65, dev_info)
```
And see that helps.

aasick, 2022-05-11:
after changing to 0x64000, 0x65 working address 64000 also throwing error @dhoomakethu 

2022-05-11 10:24:56,654 MainThread      DEBUG    sync           :46       Client Connected [COM9:COM9]
2022-05-11 10:24:56,655 MainThread      DEBUG    sync           :581      Started thread to serve client
2022-05-11 10:24:58,677 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0xfa 0x1 0x0 0x1b
2022-05-11 10:24:58,678 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-05-11 10:24:58,685 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-05-11 10:24:58,686 MainThread      DEBUG    context        :63       validate: fc-[3] address-64002: count-27
2022-05-11 10:24:58,688 MainThread      ERROR    pdu            :107      Exception Response(131, 3, IllegalAddress)
2022-05-11 10:24:58,690 MainThread      DEBUG    sync           :144      send: [Exception Response(131, 3, IllegalAddress)]- b'018302c0f1'
2022-05-11 10:25:01,715 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0xfa 0x1 0x0 0x1b
2022-05-11 10:25:01,716 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-05-11 10:25:01,720 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-05-11 10:25:01,724 MainThread      DEBUG    context        :63       validate: fc-[3] address-64002: count-27
2022-05-11 10:25:01,726 MainThread      ERROR    pdu            :107      Exception Response(131, 3, IllegalAddress)
2022-05-11 10:25:01,727 MainThread      DEBUG    sync           :144      send: [Exception Response(131, 3, IllegalAddress)]- b'018302c0f1'

dhoomakethu, 2022-05-11:
You will have to share your complete code @aasick . Looks like your data blocks are not configured properly

aasick, 2022-05-11:
Please take a look, following code is power meter server code

`

  def message_generator():
      builder = BinaryPayloadBuilder(byteorder=Endian.Little,
                                     wordorder=Endian.Little)
      payload = message_generator(builder)
  
      # Write to registers slave 1 ####
      slave_id = 0x01
      log.debug(f"::: Make payload to SLAVE={slave_id} :::")
  
      # Total energy
      builder = BinaryPayloadBuilder(byteorder=Endian.Little, wordorder=Endian.Little)
      builder.add_16bit_uint(0x0103)
      builder.add_16bit_uint(0x36)
      builder.add_16bit_uint(0x2A)
      builder.add_string('7KM2200-2EA40-1HA1')
      builder.add_16bit_int(0x0000)
      builder.add_string('LQN/210205720289')
      builder.add_8bit_int(0x00)
      builder.add_8bit_int(0x02)
      builder.add_8bit_int(0x56)
      builder.add_8bit_int(0x03)
      builder.add_8bit_int(0x00)
      builder.add_8bit_int(0x11)
  
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0xF6)
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0x01)
      builder.add_16bit_int(0x01)
      builder.add_16bit_int(0x1E)
      builder.add_16bit_int(0x87)
      builder.add_16bit_int(0xCD)
  
      dev_info = builder.to_registers()
  
      log.debug(f"::: Write registers to SLAVE={slave_id} :::")
      context[slave_id].setValues(register, 0x64001, dev_info)
      context[slave_id].setValues(register, 0x011e, phase1)
  
      return context
  
  
  def siemens_PAC2200(comport):
      register = 1
      # Write to registers slave 1 ####
      slave_id = 0x01
      log.debug(f"::: Make payload to SLAVE={slave_id} :::")
  
      # Total energy
      builder = BinaryPayloadBuilder(byteorder=Endian.Little, wordorder=Endian.Little)
      builder.add_16bit_uint(0x03)
      builder.add_16bit_uint(0x36)
      builder.add_16bit_uint(0x2A)
      builder.add_string('7KM2200-2EA40-1HA1')
      builder.add_16bit_int(0x0000)
      builder.add_string('LQN/210205720289')
      builder.add_8bit_int(0x00)
      builder.add_8bit_int(0x02)
      builder.add_8bit_int(0x56)
      builder.add_8bit_int(0x03)
      builder.add_8bit_int(0x00)
      builder.add_8bit_int(0x11)
  
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0xF6)
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0x00)
      builder.add_16bit_int(0x01)
      builder.add_16bit_int(0x01)
      builder.add_16bit_int(0x1E)
      builder.add_16bit_int(0x87)
      builder.add_16bit_int(0xCD)
      builder.add_bits([0, 1, 0, 1, 1, 0, 1, 0])
      builder.add_8bit_int(-0x12)
      builder.add_8bit_uint(0x12)
      builder.add_16bit_int(-0x5678)
      builder.add_16bit_uint(0x1234)
      builder.add_32bit_int(-0x1234)
      builder.add_32bit_uint(0x12345678)
      builder.add_16bit_float(12.34)
      builder.add_16bit_float(-12.34)
      builder.add_32bit_float(22.34)
      builder.add_32bit_float(-22.34)
      builder.add_64bit_int(-0xDEADBEEF)
      builder.add_64bit_uint(0x12345678DEADBEEF)
      builder.add_64bit_uint(0xDEADBEEFDEADBEED)
      builder.add_64bit_float(123.45)
      builder.add_64bit_float(-123.45)
      dev_info = builder.to_registers()
  
      # log.debug(f"::: payload to SLAVE={builder.to_registers()} :::")
  
      # builder2 = BinaryPayloadBuilder(byteorder=Endian.Little, wordorder=Endian.Little)
      builder.add_16bit_uint(0x03)  #01 03 04 42 0E 4B BC B8 C9
      builder.add_16bit_uint(0x04)
      builder.add_16bit_uint(0x42)
      builder.add_16bit_uint(0x0E)
      builder.add_16bit_uint(0x4B)
      builder.add_16bit_uint(0xBC)
  
      builder.add_32bit_float(35.57)
      builder.add_16bit_uint(0xB8)
      builder.add_16bit_uint(0xC9)
      # builder.add_32bit_float(0x0BDE)
      vol1 = builder.to_registers()
      # log.debug(f"::: payload to SLAVE={builder.to_registers()} :::")
  
      block1 = ModbusSequentialDataBlock(64000, dev_info)
      block2 = ModbusSequentialDataBlock(65, vol1)
      store = ModbusSlaveContext(hr=block1, ir=block2)
      slaves = {0x01: store}
      context = ModbusServerContext(slaves=slaves, single=False)
  
      # log.debug(f"::: Write registers to SLAVE={block} :::")
      # log.debug(f"::: Write registers to SLAVE={context} :::")
      # context[slave_id].setValues(register, 0x64000, dev_info)
      # context[slave_id].setValues(register, 0x65, vol1)
      #
      # log.debug(f"::: context to SLAVE={context[slave_id]} :::")
  
      # ----------------------------------------------------------------------- #
      # initialize the server information
      # ----------------------------------------------------------------------- #
      # If you don't set this or any fields, they are defaulted to empty strings.
      # ----------------------------------------------------------------------- #
      identity = ModbusDeviceIdentification()
      identity.VendorName = 'Siemens AG'
      identity.OrderNo = '7KM2200-2EA40-1HA1'
      identity.SerialNumber = 'LQN/210205720289'
      identity.HardwareVersion = '2'
      identity.FirmwareVersion = 'V 3.0.17'
      # ----------------------------------------------------------------------- #
      # run the server you want
      # ----------------------------------------------------------------------- #
      StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port=comport, stopbits=1, bytesize=8,
                        parity='E', baudrate=9600, timeout=1)
`

aasick, 2022-05-11:
@dhoomakethu above is code what I have prepared for power meter server. current I tried with two block data one is for device info(block1 -> 64000) another one is voltage value(block2 -> 65). and need to add another blocks for different voltage levels

aasick, 2022-05-12:
hello @dhoomakethu

swamper123, 2022-05-12:
hey @aasick,
please remember and respect that this is a voluntary project. We won't provide support 24/7 and won't provide immediately support (unless you got a contract with one of us somehow).
I am sure that dhoomakethu read your post above and will respond to you.

aasick, 2022-05-12:
Yeah, Cool!

dhoomakethu, 2022-05-12:
@aasick 
```
2022-05-10 23:50:40,900 MainThread DEBUG sync :144 send: [ReadHoldingRegistersResponse (27)]- b'0103362a00374b4d323230302d32454134302d3148413100004c514e2f32313032303537323032383900025603001100000000f600000000005b6c'
2022-05-10 23:50:41,908 MainThread DEBUG rtu_framer :185 Getting Frame - 0x3 0x0 0x41 0x0 0x2
2022-05-10 23:50:41,909 MainThread DEBUG factory :137 Factory Request[ReadHoldingRegistersRequest: 3]
2022-05-10 23:50:41,913 MainThread DEBUG rtu_framer :107 Frame advanced, resetting header!!
2022-05-10 23:50:41,914 MainThread DEBUG context :63 validate: fc-[3] address-66: count-2
2022-05-10 23:50:41,918 MainThread ERROR pdu :107 Exception Response(131, 3, IllegalAddress)
2022-05-10 23:50:41,920 MainThread DEBUG sync :144 send: [Exception Response(131, 3, IllegalAddress)]- b'018302c0f1'
```

You are initialzing address 65 to `input_registers` and trying to read with `read_holding_register` request. Try `read_input_registers` instead.

aasick, 2022-05-12:
how to include read_input_registers ? we have ony di, co, hr, ir right

aasick, 2022-05-12:
we have only one holding register right for creating data block for slave

dhoomakethu, 2022-05-12:
For your case, you need use `ModbusSparseDataBlock`

```
# ----------------------------------------------------------------------- #
    # initialize your data store
    # ----------------------------------------------------------------------- #
    # The datastores only respond to the addresses that they are initialized to
    # Therefore, if you initialize a DataBlock to addresses of 0x00 to 0xFF, a
    # request to 0x100 will respond with an invalid address exception. This is
    # because many devices exhibit this kind of behavior (but not all)::
    #
    #     block = ModbusSequentialDataBlock(0x00, [0]*0xff)
    #
    # Continuing, you can choose to use a sequential or a sparse DataBlock in
    # your data context.  The difference is that the sequential has no gaps in
    # the data while the sparse can. Once again, there are devices that exhibit
    # both forms of behavior::
    #
    #     block = ModbusSparseDataBlock({0x00: 0, 0x05: 1})
    #     block = ModbusSequentialDataBlock(0x00, [0]*5)
    #
    # Alternately, you can use the factory methods to initialize the DataBlocks
    # or simply do not pass them to have them initialized to 0x00 on the full
    # address range::
    #
    #     store = ModbusSlaveContext(di = ModbusSequentialDataBlock.create())
    #     store = ModbusSlaveContext()
    #
    # Finally, you are allowed to use the same DataBlock reference for every
    # table or you may use a separate DataBlock for each table.
    # This depends if you would like functions to be able to access and modify
    # the same data or not::
    #
    #     block = ModbusSequentialDataBlock(0x00, [0]*0xff)
    #     store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
    #
    # The server then makes use of a server context that allows the server to
    # respond with different slave contexts for different unit ids. By default
    # it will return the same context for every unit id supplied (broadcast
    # mode).
    # However, this can be overloaded by setting the single flag to False and
    # then supplying a dictionary of unit id to context mapping::
    #
    #     slaves  = {
    #         0x01: ModbusSlaveContext(...),
    #         0x02: ModbusSlaveContext(...),
    #         0x03: ModbusSlaveContext(...),
    #     }
    #     context = ModbusServerContext(slaves=slaves, single=False)
    #
    # The slave context can also be initialized in zero_mode which means that a
    # request to address(0-7) will map to the address (0-7). The default is
    # False which is based on section 4.4 of the specification, so address(0-7)
    # will map to (1-8)::
    #
    #     store = ModbusSlaveContext(..., zero_mode=True)
    # ----------------------------------------------------------------------- #
```

aasick, 2022-05-12:
can we use like,  
block = ModbusSparseDataBlock({0x64000: dev_info, 0x65: vol1})

dhoomakethu, 2022-05-12:
yes 
```
block = ModbusSparseDataBlock({0x64000: dev_info, 0x65: vol1})
store = ModbusSlaveContext(hr=block)
```

aasick, 2022-05-12:
I have changed payload like this

    dev_info = [0x03, 0x36, 0x2A, 0x11, 0x00, 0x23, 0x00, 0x02, 0x56, 0x03, 0x00, 0x11, 0x57, 0x20, 0x56, 0x33, 0x2E, 0x30, 0x2E, 
    0x31, 0x31, 0x2E, 0x30, 0x5F, 0x31, 0x2E, 0x31, 0x2E]
    msg_payload = [0x03, 0x04, 0x43, 0x6A, 0x0B, 0xDE, 0x48, 0xC3]

    block = ModbusSparseDataBlock({0x64000: dev_info, 0x65: msg_payload})
    store = ModbusSlaveContext(hr=block)
    slaves = {0x01: store}
    context = ModbusServerContext(slaves=slaves, single=False)

aasick, 2022-05-12:
I have error on 64000 it self

2022-05-12 15:17:30,623 MainThread      DEBUG    sync           :46       Client Connected [COM9:COM9]
2022-05-12 15:17:30,629 MainThread      DEBUG    sync           :581      Started thread to serve client
2022-05-12 15:17:31,633 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0xfa 0x1 0x0 0x1b
2022-05-12 15:17:31,633 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2022-05-12 15:17:31,636 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2022-05-12 15:17:31,641 MainThread      DEBUG    context        :63       validate: fc-[3] address-64002: count-27
2022-05-12 15:17:31,642 MainThread      ERROR    pdu            :107      Exception Response(131, 3, IllegalAddress)
2022-05-12 15:17:31,645 MainThread      DEBUG    sync           :144      send: [Exception Response(131, 3, IllegalAddress)]- b'018302c0f1'

dhoomakethu, 2022-05-12:
@aasick you are initializing your address in hex and requesting in decimal which is obviously going to fail. Please pay attention to the error codes and please do read the specs.