pymodbus-dev/pymodbus #286: Pymodbus buffer receiving

NagarjunaReddy1242, 2018-04-12:
Hi!!!!!
I have one problem that is when i was send query through GPIO (BeagleBoneBlack P8-18&P8-30). when i was send query before i enabled GPIO is HIGH(both). after sending query i make it GPIO LOW , than after i would get responds. but receive responds is not happening please solve that problem. 
Program:
import struct
import Adafruit_BBIO.GPIO as GPIO
import pymodbus
import sys
import struct

import serial
import time
from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance
from pymodbus.transaction import ModbusRtuFramer
GPIO.setup("P8_19", GPIO.OUT)
GPIO.setup("P8_30", GPIO.OUT)

import time,os

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client= ModbusClient(method = "rtu",port='/dev/ttyS1', stopbits = 1, bytesize = 8, parity = 'N', baudrate=9600)
#connect to the serial modbus server
connection = client.connect()
message='hello'
print(message)
print connection

def _send(self, packet):
    GPIO.output("P8_19", GPIO.HIGH)
    GPIO.output("P8_30", GPIO.HIGH)
            
    packett=client.read_holding_registers(0x009C,0x016,unit=0x01)
    
    GPIO.output("P8_19", GPIO.LOW)
    GPIO.output("P8_30", GPIO.LOW)
    print packet

client.close()

Note1: before Sending time both GPIO pins are make it HIGH
Note2: before Receiving time both GPIO pins are make it LOW

dhoomakethu, 2018-04-13:
please post the pymodbus logs from your program.

NagarjunaReddy1242, 2018-04-13:
Thank you for replay to me. I got the answer.

closed 2018-04-13 by dhoomakethu