pymodbus-dev/pymodbus #290: Sync Modbus RTU client always waits for timeout- When writing multiple Analog Holding Registers

pokornyElgas, 2018-04-23:
### Versions

* Python: 2.7.9
* OS: Windows 7 Professional
* Pymodbus: 1.4.0
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: rtu - sync
* Client: rtu - sync

### Description
Sync Modbus RTU client always waits for timeout- When writing multiple Analog Holding Registers. Similar to Issue 76
I have solved it by adding method get_response_pdu_size to the class WriteMultipleRegistersRequest in module register_write_message.py.
Is it right? If so, could You include it in new pymodbus version?

### Code and Logs

```python
    def get_response_pdu_size(self):
        """
        FuncCode (1 Byte) + Data Address (2 Bytes) + Num of Regs Written (2 Bytes)
        :return:
        """
        return (1 + 2 + 2)

```

dhoomakethu, 2018-04-23:
This is already fixed as part of upcoming v1.5.0, you can give a try of the [pre-release ](https://github.com/riptideio/pymodbus/releases/tag/v1.5.0.rc2).

pokornyElgas, 2018-04-24:
OK, clear, thank You for answer. And for Your work on pymodbus too!

closed 2018-04-24 by pokornyElgas