pymodbus-dev/pymodbus #314: A program using ModbusTcpSlave doesn't exit

yegorich, 2018-07-09:
### Versions

* Python: 3.6.5
* OS: Ubuntu 18.04 LTS
* Pymodbus: 1.5.2
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp - sync/async

### Description

A threaded class creates and starts ModbusSlave, but invoking StopServer() makes the main process stuck. So after "OK" the program doesn't return to the prompt. Both sync and asynch versions showed the same behavior. Am I missing something?

### Code and Logs

```python
import threading
import time
from pymodbus.server.async import StartTcpServer, StopServer
from pymodbus.server.async import StartSerialServer

from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

from pymodbus.transaction import ModbusRtuFramer

class MRunnner(threading.Thread):

    def __init__(self, **kwargs):
        threading.Thread.__init__(self)
        store = ModbusSlaveContext(
            di=ModbusSequentialDataBlock(0, [17]*100),
            co=ModbusSequentialDataBlock(0, [17]*100),
            hr=ModbusSequentialDataBlock(0, [17]*100),
            ir=ModbusSequentialDataBlock(0, [17]*100))

        self.context = ModbusServerContext(slaves=store, single=True)

        self.identity = ModbusDeviceIdentification()
        self.identity.VendorName = 'Pymodbus'
        self.identity.ProductCode = 'PM'
        self.identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
        self.identity.ProductName = 'Pymodbus Server'
        self.identity.ModelName = 'Pymodbus Server'
        self.identity.MajorMinorRevision = '1.5'

        self.kwargs = kwargs

    def run(self):
        StartTcpServer(self.context,
                       identity=self.identity,
                       **self.kwargs)

    def stop(self):
        '''Stop Modbus server.'''
        StopServer()

slave = MRunnner(address=('0.0.0.0', 5020))
slave.start()
time.sleep(2)
slave.stop()

print("OK")
```

dhoomakethu, 2018-07-09:
It could be the way how twisted reactor runs in thread, a simple work around would be to make the thread `daemon` so that it wont block when the main program exits. Just add `self.daemon = True` in the `__init__` method after `threading.Thread.__init__(self)`.

yegorich, 2018-07-10:
`self.daemon = True` helped. Thanks.

closed 2018-07-10 by yegorich