pymodbus-dev/pymodbus #315: ERROR:pymodbus.server.sync:can't concat NoneType to bytes

MAKOMO, 2018-07-14:
### Versions

* Python: 3.6/3.7
* OS: Mac OS X
* Pymodbus: 1.5.2 (todays trunk)
* Modbus Hardware (if used): external software client

### Pymodbus Specific
* Server: tcp(UDP) - sync
* Client: external software client

### Description

Running this pymodbus based sync server via UDP I receive above error on connecting an external (software) client. It seems this error is thrown in ModbusDisconnectedRequestHandler during the call to processIncomingPacket if the data extracted from the request is None. Maybe in this case, processIncomingPacket should not be called and the pymodbus server could silently accept this request, even if broken.

### Code and Logs

```
DEBUG:pymodbus.server.sync:Started thread to serve client at ('192.168.2.105', 52789)
DEBUG:pymodbus.server.sync:Client Connected [192.168.2.105:52789]
data b'\x00\xb1\x00\x00\x00\x06\x00\x030\x82\x00\x01' <class 'bytes'>
DEBUG:pymodbus.server.sync:Handling data2: 0x0 0xb1 0x0 0x0 0x0 0x6 0x0 0x3 0x30 0x82 0x0 0x1
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0xb1 0x0 0x0 0x0 0x6 0x0 0x3 0x30 0x82 0x0 0x1
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 12419:1
DEBUG:pymodbus.datastore.context:getValues[3] 12419:1
DEBUG:pymodbus.server.sync:send: b'00b1000000050003020000'
DEBUG:pymodbus.server.sync:Handling data: 
DEBUG:pymodbus.framer.socket_framer:Processing: 
ERROR:pymodbus.server.sync:can't concat NoneType to bytes
DEBUG:pymodbus.server.sync:Client Disconnected [192.168.2.105:52789]
```

chintal, 2018-07-18:
processIncomingPacket could simply return if data is None, but what are the conditions under which you actually get valid TCP packets but result in zero length data? 

If this is happening at the moment of disconnection or so, does it warrant special handling, given your client is going away and the request will have to be purged anyway? More specifically, does it just pollute the log or does it also cause the server to become non-functional henceforth?

If it is the latter, it may be worthwhile to invest in a more robust and error agnostic server reset mechanism that triggers a recovery process on *any* unhandled error.

MAKOMO, 2018-07-18:
It does not render the server non-functional, but this issue did not occur with previous versions.

janiversen, 2022-05-25:
solved on current dev. 

Feel free to open a new PR, if you still experience problems.

closed 2022-05-25 by janiversen