pymodbus-dev/pymodbus #316: Request for feature of native or a RFC 2217 implementation of the Serial class, depending on port/url.

madhujk, 2018-07-24:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```

dhoomakethu, 2018-07-24:
@madhujk Could you provide more info on your request ?

madhujk, 2018-07-24:
It is a request to support both, local ports and remote ports servers supporting rfc2217  format ,
format:
rfc2217://<host>:<port>[?<option>[&<option>...]]
socket://<host>:<port>[?logging={debug|info|warning|error}]
loop://[?logging={debug|info|warning|error}]
hwgrep://<regexp>[&skip_busy][&n=N]
spy://port[?option[=value][&option[=value]]]
alt://port?class=<classname>

Pyserial support above url handler to connect to remote ports with "serial_for_url()".

pymodbus dymanically  can connect to local port or remote port based on input to the server

isj4, 2019-05-14:
I need this, specifically the "hwgrep://" syntax.
Scenario: Remote box, two FTDI RS-485 USB adapters, other USB devices also show up as /dev/ttyUSB*, so I cannot rely on the device names as they change on reboot.
So ideally I would specify the serial ports to be:
"hwgrep://0403:6001" and "hwgrep://0403:6001&n=2"

I hacked the pymodbus module, sync.py, ModbusSerialClient.connect()
Instead of:

            self.socket = serial.Serial(port=self.port,
                                        timeout=self.timeout,
                                        bytesize=self.bytesize,
                                        stopbits=self.stopbits,
                                        baudrate=self.baudrate,
                                        parity=self.parity)

I use this:

            self.socket = serial.serial_for_url(self.port,
                                                timeout=self.timeout,
                                                bytesize=self.bytesize,
                                                stopbits=self.stopbits,
                                                baudrate=self.baudrate,
                                                parity=self.parity)

That's it. Works for me.

dhoomakethu, 2019-05-15:
@isj4 Thanks for the workaround, I will see if this is something we can support from the pymodbus library directly

isj4, 2021-03-03:
I still need this.

commit e9fda05e (2021-03-09): Fix #316 support serial for url
closed 2021-04-01 by dhoomakethu