pymodbus-dev/pymodbus #317: Detected bugs in multiple files

joseluisfloresbarroso, 2018-07-31:
I was conducting a set of security tests using different software packages in the industrial field, and one of them was this modbus server. I have used MTF (https://github.com/avoyages/etfa2015) as fuzzer to find bugs. 

As a a result we have found several bug in different files:
* bit_read_message.py
* bit_write_message.py
* diag_message.py
* factory.py
* file_message.py
* mei_message.py
* register_read_message.py
* register_write_message.py
* transaction.py

The main problem, that is, the corresponding to the first 7 files, corresponds to buffer overflows, whereas the last ones corresponds to errors checking the length of the arrays. In the logs you can check the details.

With the information that I have provided regarding the buzzer, you can reproduce the bugs, if you need some information do not hesitate to contact me (I'll be online from the last week of August). It takes more or less 3-4 hours.

On the other hand, one of the reasons to contact you is that this work for finding bugs was a part of a research work of secure architecture in industrial environments that I'm going to show in September 4-7.  In my presentation, I will omit deliberately the specific details of the bugs and although the paper includes the details, the organization requires much more time to publish the proceedings, this can give you more time to patch the code when required.

It is only an idea but I think it can be useful for you to use this fuzzer to test the robustness of the software, as a part of the software validation process, I think some else can use the same for other not so clear reasons.

Sincerely,

Jose Luis.

dhoomakethu, 2018-07-31:
Hello @joseluisfloresbarroso , Thanks for your testing. If you have any reports handy from your tests ,please share it here.

joseluisfloresbarroso, 2018-08-13:
Hi

I have used this software package: mtf-2015 (
https://github.com/avoyages/etfa2015), as a fuzzer with the following
parameters:

*python ./mtf.py -i <IP Address> -s -z -f*

Where the IP corresponds to the modbus server IP address.

And after several hours we have obtained several bugs with specific python
function, on the other hand we have modified the fuzzer to increase the
logging level to debug, in this way we know with precision the exact data
pattern and at the same time in the modbus server the corresponding
function.

I hope this helps.

Regards,

JL

closed 2021-03-08 by github-actions[bot]