pymodbus-dev/pymodbus #318: Dev branch - different binary parsing?

patrickfuller, 2018-08-17:
I maintain and use [numat/clickplc](https://github.com/numat/clickplc) for lab-scale automation projects.

I've been using the `ReconnectingAsyncioModbusTcpClient`, which was only available on the `python3` branch. This has since moved to `dev`. I changed the import ([here](https://github.com/numat/clickplc/commit/f7c3d8a8145f905b89f631065caffaac0d1a04ee)), but now the `BinaryPayloadDecoder` ([here](https://github.com/numat/clickplc/blob/master/clickplc/driver.py#L244-L249)) is no longer returning reasonable floats.

Has something significant changed in `BinaryPayloadDecoder` since the python3 branch?

dhoomakethu, 2018-08-18:
`[BinaryPayloadDecoder](https://github.com/riptideio/pymodbus/blob/dev/pymodbus/payload.py#L261)` now also takes in to consideration `wordorder )` and `byteorder` . I see that you are using the defaults for both of those arguements (`Endian.Little for byteorder` and `Endian.Big for wordorder`) Check if that is something that is giving you a wrong output. It would helpful, If you can share more information on what was the earlier value (expected) and what is the (wrong) value you are getting for a given set of registers.

patrickfuller, 2018-08-20:
Thanks for the information. Adding `byteorder` and `wordorder` args is definitely cleaner ([fix](https://github.com/numat/clickplc/commit/542a4eab6bbd3e1f729ce2385a5fe28a889d7fb7)).

I know this branch is for 2.0.0, so breaking changes should be expected. However, I'm curious why this branch changes the default endianness from Little/Little to Little/Big. My two cents - It's fine if there's a good reason, but it's a subtle breakage that (I think) will result in a lot more emails/issues once you push to master.

dhoomakethu, 2018-08-20:
Thanks for your suggestion.Please note,  the `byteorder` was set to `Endian.Little` by default in the earlier versions as well. There were few bug fixes the way words were handled when wordcount>1 (16 bits) with the newer version.

closed 2018-08-20 by dhoomakethu