pymodbus-dev/pymodbus #319: `asyncio.async` deprecated since python 3.4.4 and was removed in 3.7

cbergmiller, 2018-08-21:
### Versions

* Python: 3.7
* OS: Windows
* Pymodbus: async branch

### Description

`asyncio.async` has been deprecated since python 3.4.4 and was removed in 3.7. 
See: [https://docs.python.org/3.5/library/asyncio-task.html#asyncio.async](https://docs.python.org/3.5/library/asyncio-task.html#asyncio.async)

The alias `ensure_future` should be used instead. See [https://docs.python.org/3.5/library/asyncio-task.html#asyncio.ensure_future](https://docs.python.org/3.5/library/asyncio-task.html#asyncio.ensure_future)

This of course would only support python >= 3.4.4
I can create an PR if desired.

dhoomakethu, 2018-08-21:
@cbergmiller , Thanks. A PR is appreciated

dhoomakethu, 2019-01-14:
Now part of branch [pymodbus-2.2.0](https://github.com/riptideio/pymodbus/tree/pymodbus-2.2.0), please give it a try and feedback

dhoomakethu, 2019-01-16:
Fixed and available as part of [v2.2.0rc1](https://github.com/riptideio/pymodbus/releases/tag/v2.2.0rc1).
`pip install pymodbus==2.2.0rc1`

closed 2019-01-16 by dhoomakethu