pymodbus-dev/pymodbus #320: async branch is not compatible with python 3.7

cbergmiller, 2018-08-21:
`async` is a not allowed in Python 3.7 module paths.

### Versions

* Python: 3.7
* OS: Windows
* Pymodbus: async branch

### Description

The async branch can not be used with Python 3.7
A solution would be to rename `pymodbus.client.async` with `pymodbus.client.aio` or `pymodbus.client.asynchronous`.

I can create a PR if desired.

### Code and Logs

```python
  File "async_asyncio_client.py", line 18
    from pymodbus.client.async.tcp import AsyncModbusTCPClient as ModbusClient
                             ^
SyntaxError: invalid syntax
```

dhoomakethu, 2018-08-21:
@cbergmiller thanks for bringing this to notice, do you have any more hints on why this is happening ? Is `async` a reserved keyword starting with 3.7 ?

cbergmiller, 2018-08-21:
@dhoomakethu yes `async` and `await` are reserved keywords since 3.7:
[https://docs.python.org/3/whatsnew/3.7.html](https://docs.python.org/3/whatsnew/3.7.html)

dhoomakethu, 2018-08-21:
@cbergmiller Appreciate if you can raise a PR . Thanks.

cbergmiller, 2018-08-21:
It would be great if Python 3.7 could be added to the Travis CI test environments.

jessekerkhoven, 2018-11-27:
Any update on merging `dev3.7` branch in to master for a new release? See also #346

dhoomakethu, 2018-11-27:
@jessekerkhoven apologies for the delay, I am yet to do a code review . Will try to complete this asap.

dhoomakethu, 2019-01-14:
For any one wanting support on python 3.7, please give a try on branch [pymodbus-2.2.0](https://github.com/riptideio/pymodbus/tree/pymodbus-2.2.0)

dhoomakethu, 2019-01-16:
Fixed and available as part of [v2.2.0rc1](https://github.com/riptideio/pymodbus/releases/tag/v2.2.0rc1).
`pip install pymodbus==2.2.0rc1`

closed 2019-01-16 by dhoomakethu