pymodbus-dev/pymodbus #322: How to use REUSEADDR in ModbusTcpServer?

agn-7, 2018-08-28:
### Versions

* Python: 3.6.5
* OS: Ubuntu 16.04
* Pymodbus: 1.5.2
* Modbus Hardware (if used): No

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

I'm using ModbusTcpServer sync library to create a Modbus slave, here's this code:

```python
from pymodbus.server.sync import StartTcpServer, ModbusTcpServer

from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

from twisted.internet.task import LoopingCall
from twisted.internet import reactor
import threading
import logging

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_server():
    block1 = ModbusSequentialDataBlock(0x00, [717] * 0x0F)
    block2 = ModbusSequentialDataBlock(0x10, [323] * 0x1F)
    store2 = ModbusSlaveContext(hr=block1, ir=block2)

    slaves = {
        0x01: store2,
    }

    context = ModbusServerContext(slaves=slaves, single=False)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '1.0'

    interval = 2
    server = ModbusTcpServer(context,
                             identity=identity,
                             address=('0.0.0.0', 5021))
    thread_ = threading.Thread(target=server.serve_forever, daemon=True)
    thread_.start()
    loop = LoopingCall(f=update_values, a=server)
    loop.start(interval, now=True)
    reactor.run()

def update_values(a):
    print("-----------START-----------")
    rfuncode = 3
    wfuncode = 16
    slave_id = 0x01
    address = 0x00
    context_ = a.context[slave_id]
    values = context_.getValues(rfuncode, address, count=32)
    print(values)
    values = [val+1 for val in values]
    context_.setValues(wfuncode, address, values)
    print("------------END------------")

if __name__ == "__main__":
    run_server()

```

When I close this code (with <kbd>Ctrl</kbd>+<kbd>C</kbd>) and run again encountered with this error:

`OSError: [Errno 98] Address already in use
`
I know that in socket creation we can use `socket.SO_REUSEADDR` for overcome on this issue.

Also, I can `.close()` connection in client side to resolve this issue but I want have a stable server. 

My question is, is there a built-in way to overcome this? I find out this argument (`socket.SO_REUSEADDR`) in `async.py`  but there isn't in `sync.py`

dhoomakethu, 2018-08-29:
You will have to override `ModbusTcpServer` class and set the property `self.allow_resuse_address` to `True` to make it work.
```
class ReusableModbusTcpServer(ModbusTcpServer):

    def __init__(self, context, framer=None, identity=None,
                 address=None, handler=None, **kwargs):
        self.allow_reuse_address = True
        ModbusTcpServer.__init__(self, context, framer, identity, address, handler, **kwargs)
```
https://docs.python.org/2/library/socketserver.html#SocketServer.BaseServer.allow_reuse_address

dhoomakethu, 2018-08-29:
@agn-7  I will be closing this issue. Please re-open the issue if that does not resolve the issue.

closed 2018-08-29 by dhoomakethu

richaeell, 2019-05-31:
> ### Versions
> * Python: 3.6.5
> * OS: Ubuntu 16.04
> * Pymodbus: 1.5.2
> * Modbus Hardware (if used): No
> 
> ### Pymodbus Specific
> * Server: tcp - sync
> * Client: tcp - sync
> 
> I'm using Synchronous ModbusTcpServer to create a Modbus slave, here's the code:
> 
> ```python
> from pymodbus.server.sync import StartTcpServer, ModbusTcpServer
> 
> from pymodbus.device import ModbusDeviceIdentification
> from pymodbus.datastore import ModbusSequentialDataBlock
> from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
> 
> from twisted.internet.task import LoopingCall
> from twisted.internet import reactor
> import threading
> import logging
> 
> logging.basicConfig()
> log = logging.getLogger()
> log.setLevel(logging.DEBUG)
> 
> 
> def run_server():
>     block1 = ModbusSequentialDataBlock(0x00, [717] * 0x0F)
>     block2 = ModbusSequentialDataBlock(0x10, [323] * 0x1F)
>     store2 = ModbusSlaveContext(hr=block1, ir=block2)
> 
>     slaves = {
>         0x01: store2,
>     }
> 
>     context = ModbusServerContext(slaves=slaves, single=False)
> 
>     identity = ModbusDeviceIdentification()
>     identity.VendorName = 'Pymodbus'
>     identity.ProductCode = 'PM'
>     identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
>     identity.ProductName = 'Pymodbus Server'
>     identity.ModelName = 'Pymodbus Server'
>     identity.MajorMinorRevision = '1.0'
> 
>     interval = 2
>     server = ModbusTcpServer(context,
>                              identity=identity,
>                              address=('0.0.0.0', 5021))
>     thread_ = threading.Thread(target=server.serve_forever, daemon=True)
>     thread_.start()
>     loop = LoopingCall(f=update_values, a=server)
>     loop.start(interval, now=True)
>     reactor.run()
> 
> 
> def update_values(a):
>     print("-----------START-----------")
>     rfuncode = 3
>     wfuncode = 16
>     slave_id = 0x01
>     address = 0x00
>     context_ = a.context[slave_id]
>     values = context_.getValues(rfuncode, address, count=32)
>     print(values)
>     values = [val+1 for val in values]
>     context_.setValues(wfuncode, address, values)
>     print("------------END------------")
> 
> 
> if __name__ == "__main__":
>     run_server()
> ```
> 
> When I close this code (with Ctrl+C) and run again encountered with this error:
> 
> `OSError: [Errno 98] Address already in use `
> I know that in socket creation we can use `socket.SO_REUSEADDR` for overcome on this issue.
> 
> Also, I can `.close()` connection in client side to resolve this issue but I want have a stable server.
> 
> # Question:
> Is there a built-in way to overcome this? I find out this argument (`socket.SO_REUSEADDR`) in Asynchronous ModbusTcpServer (in `async.py`) but there isn't in Synchronous ModbusTcpServer (`sync.py`).

Where did you find the async.py with the socket.SO_REUSEADDR??