pymodbus-dev/pymodbus #324: Unable to close Synchronous ModbusTcpServer

agn-7, 2018-08-30:
### Versions

* Python: 3.6.6
* OS: Ubuntu 16.04
* Pymodbus: 1.5.2
* Modbus Hardware (if used): No

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

When I start  a Synchronous ModbusTcpServer with `.server_forever()` in a thread, I can't close it with <kbd>Ctrl</kbd>+<kbd>C</kbd> when a synchronous client connected (`client.connect()`) to it and not closed  (`client.close()`) and stuck in terminal.

### Code and Logs

```python
from pymodbus.server.sync import ModbusTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from twisted.internet.task import LoopingCall
from twisted.internet import reactor
from threading import Thread
import logging

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

class ReusableModbusTcpServer(ModbusTcpServer):
    def __init__(self, context, framer=None, identity=None, address=None, handler=None, **kwargs):
        self.allow_reuse_address = True
        super().__init__(context, framer, identity, address, handler, **kwargs)

def run_server():
    block1 = ModbusSequentialDataBlock(0x00, [717] * 0x0F)
    block2 = ModbusSequentialDataBlock(0x10, [323] * 0x1F)
    store2 = ModbusSlaveContext(hr=block1, ir=block2)
    slaves = {
        0x01: store2,
    }
    context = ModbusServerContext(slaves=slaves, single=False)
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '1.0'
    interval = 2
    reuse_server = ReusableModbusTcpServer(context,
                                           identity=identity,
                                           address=('0.0.0.0', 5021))
    Thread(target=reuse_server.serve_forever, daemon=True).start()
    loop = LoopingCall(f=update_values, a=reuse_server)
    loop.start(interval, now=True)
    reactor.run()
    reuse_server.shutdown()
    reuse_server.server_close()
    reuse_server.__shutdown_request = True

def update_values(a):
    print("-----------START-----------")
    rfuncode = 3
    wfuncode = 16
    slave_id = 0x01
    address = 0x00
    context_ = a.context[slave_id]
    values = context_.getValues(rfuncode, address, count=32)
    print(values)
    values = [val+1 for val in values]
    context_.setValues(wfuncode, address, values)
    print("------------END------------")

if __name__ == "__main__":
    run_server()

```

I also use from  `reuse_server.shutdown()` ,`reuse_server.server_close()` ,`reuse_server.__shutdown_request = True` and `self.allow_reuse_address = True` but I can't overcome to that.

dhoomakethu, 2018-08-31:
You are mixing threads and reactor . And twisted has its own signal handlers which could be the reason why the server is not quitting as expected. BTW, have you checked the [updating_server.py](https://github.com/riptideio/pymodbus/blob/master/examples/common/updating_server.py) example ? That looks similar to what you are doing ,except it uses Async server. It comes with reusing address by default and handles graceful terminations.

But in case you still want to go with your code. Here is a ugly hack to deal with blocking program. Note, in some case you will have to presee `CTRL+C` twice and you will see some ugly tracebacks from the threading module.

```
from pymodbus.server.sync import StartTcpServer, ModbusTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from twisted.internet.task import LoopingCall
from twisted.internet import reactor
import threading
import logging
import signal
import time

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

SERVER = None
THREADED_SERVER = None
LOOP = None

class ReusableModbusTcpServer(ModbusTcpServer):

    def __init__(self, context, framer=None, identity=None,
                 address=None, handler=None, **kwargs):
        self.allow_reuse_address = True
        ModbusTcpServer.__init__(self, context, framer, identity, address, handler, **kwargs)

class ThreadedModbusServer(threading.Thread):

    def __init__(self, server):
        super(ThreadedModbusServer, self).__init__(name="ModbusServerThread")
        self._server = server
        self.daemon = True

    def run(self):
        self._server.serve_forever()

    def stop(self):
        if isinstance(self._server, ModbusTcpServer):
            self._server.shutdown()
        else:
            if self._server.socket:
                self._server.server_close()

def update_values(a):
    print("-----------START-----------")
    rfuncode = 3
    wfuncode = 16
    slave_id = 0x01
    address = 0x00
    context_ = a.context[slave_id]
    values = context_.getValues(rfuncode, address, count=32)
    print(values)
    values = [val+1 for val in values]
    context_.setValues(wfuncode, address, values)
    print("------------END------------")
    time.sleep(0.1)

def run_server():
    global SERVER, THREADED_SERVER, LOOP
    block1 = ModbusSequentialDataBlock(0x00, [717] * 0x0F)
    block2 = ModbusSequentialDataBlock(0x10, [323] * 0x1F)
    store2 = ModbusSlaveContext(hr=block1, ir=block2)

    slaves = {
        0x01: store2,
    }

    context = ModbusServerContext(slaves=slaves, single=False)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '1.0'

    interval = 2
    SERVER = ReusableModbusTcpServer(context,
                             identity=identity,
                             address=('0.0.0.0', 5021))  # Problem cause.
    THREADED_SERVER = ThreadedModbusServer(SERVER)
    THREADED_SERVER.start()
    LOOP = LoopingCall(f=update_values, a=SERVER)
    LOOP.start(interval, now=True)
    reactor.run()

def signal_handler(signal, frame):
    global THREADED_SERVER, LOOP
    log.warning("You pressed Ctrl+C! ."
              "If the program does not quit, Try pressing the CTRL+C again!!!")
    if THREADED_SERVER:
        THREADED_SERVER.stop()
        THREADED_SERVER = None
    if LOOP:
        LOOP.stop()
        LOOP = None
    if reactor.running:
        reactor.stop()
    else:
        exit(1)

if __name__ == "__main__":
    signal.signal(signal.SIGINT, signal_handler)
    run_server()

```

agn-7, 2018-08-31:
Thank you, my problem resolved, this code works, and I think the mentioned updating_server.py (async server) is a better approach.

closed 2018-08-31 by dhoomakethu