pymodbus-dev/pymodbus #328: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

marcoprestes, 2018-09-16:
### Versions

* Python: 3.7 32bits
* OS: Windows 10 64bits
* Pymodbus: 1.5.2
* Modbus Hardware (if used): 

### Pymodbus Specific
* Client: rtu - sync

### Description

Hi,
I`m trying to perform some basic holding registers readings, but I`m always getting the same error: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
I`ve trying to find solutions for quite some time, but nothing seems to work.

### Code and Logs

Code:
import logging
from pymodbus.client.sync import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance

client= ModbusClient(method = "RTU", port = "COM6", stopbits = 1, bytesize = 8, parity = 'N', baudrate = 9600)
connection = client.connect()

FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def processResponse(result):
    log.debug(result)

rr = client.read_holding_registers(11, 2, unit=0x01)
log.debug(rr)
rr.addCallback(processResponse)

client.close()

Logs:
2018-09-16 10:55:23,624 MainThread      DEBUG    transaction    :107      Current transaction state - IDLE
2018-09-16 10:55:23,625 MainThread      DEBUG    transaction    :111      Running transaction 1
2018-09-16 10:55:23,625 MainThread      DEBUG    transaction    :201      SEND: 0x1 0x3 0x0 0xb 0x0 0x2 0xb5 0xc9
2018-09-16 10:55:23,625 MainThread      DEBUG    sync           :74       New Transaction state 'SENDING'
2018-09-16 10:55:23,626 MainThread      DEBUG    transaction    :204      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-09-16 10:55:23,655 MainThread      DEBUG    transaction    :279      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2018-09-16 10:55:23,655 MainThread      DEBUG    transaction    :209      RECV: 0x0 0x1 0x3 0x4 0x0 0xed 0x1 0x29 0xaa
2018-09-16 10:55:23,655 MainThread      DEBUG    transaction    :420      Getting transaction 1
2018-09-16 10:55:23,655 MainThread      DEBUG    transaction    :175      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2018-09-16 10:55:23,656 MainThread      DEBUG    Teste          :25       Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

```

dhoomakethu, 2018-09-16:
Looks like your response is having an extra byte (0x0) at the beginning .May be you could try different baudrate if possible?

marcoprestes, 2018-09-19:
I`ve tried to decrease and increase the baud rate, the extra byte does not shows up again, but I got another error:
2018-09-18 21:16:19,285 MainThread      DEBUG    transaction    :107      Current transaction state - IDLE
2018-09-18 21:16:19,285 MainThread      DEBUG    transaction    :111      Running transaction 1
2018-09-18 21:16:19,285 MainThread      DEBUG    transaction    :201      SEND: 0x1 0x3 0x0 0x7 0x0 0x1 0x35 0xcb
2018-09-18 21:16:19,285 MainThread      DEBUG    sync           :74       New Transaction state 'SENDING'
2018-09-18 21:16:19,286 MainThread      DEBUG    transaction    :204      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-09-18 21:16:22,399 MainThread      DEBUG    transaction    :213      Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (1 received)) 
2018-09-18 21:16:22,399 MainThread      DEBUG    rtu_framer     :225      Frame - [b''] not ready
2018-09-18 21:16:22,399 MainThread      DEBUG    transaction    :420      Getting transaction 1
2018-09-18 21:16:22,399 MainThread      DEBUG    transaction    :175      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2018-09-18 21:16:22,400 MainThread      DEBUG    Teste          :19       reading register:
2018-09-18 21:16:22,400 MainThread      DEBUG    Teste          :20       Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (1 received)

dhoomakethu, 2018-09-19:
There is no response being received as per your logs. Are you sure you are using the correct baudrate/stop bit/parity etc ?

closed 2019-02-12 by dhoomakethu

cheryjose, 2020-03-26:
@marcoprestes did you find a solution for this problem

vabhasin, 2021-07-23:
I think @mpf82 's comment above mentions that this was an issue in version 2.1.0 and was fixed in 2.2.0. Haven't tried it myself but its confirmed in that issue's thread 
[Issue with read_input_registers in version 2.1.0 #353](https://github.com/riptideio/pymodbus/issues/353)