pymodbus-dev/pymodbus #330: bug in utilities.py BinaryPayloadDecoder

gbrucepayne, 2018-09-18:
### Versions

* Python: 2.7.13
* OS: Windows 7 Enterprise (64-bit)
* Pymodbus: 1.5.2
* Modbus Hardware (if used): n/a (any)

### Pymodbus Specific
* Server: tcp - async

### Description

I'm trying to use a BinaryPayloadDecoder to read a value that could be spread across multiple registers. Stepping through the code **decode_16bit_int** passes a list to the utilities function **make_byte_string** which is expecting a string but only returns a string presumably if IS_PYTHON3 (I'm testing on Python 2.7.13 but reads like it would still break since the handle is a list not a string type).

Also encoding/decoding for 8-bit values is broken since it encodes as a 16-bit value with **build()**, the decode crashes.  Workaround using 16-bit or bits.

I was able to come up with the following fix:

### Code and Logs

```
def make_byte_string(s):
   if isinstance(s, list):
      byte_string = ''
      for i in s:
         byte_string += i
   return byte_string
```

dhoomakethu, 2018-09-18:
Could you provide some more information on how you are using `BinaryPayloadDecoder` also some test values for the registers you are dealing and expected result ?

dhoomakethu, 2018-09-18:
I think you are using the library wrong, Have you looked in to [modbus_payload_server](https://github.com/riptideio/pymodbus/blob/master/examples/common/modbus_payload_server.py) example ? Basically when you are passing your payload to decoder , you will have to pass it as byte string if you directly instantiating the object like `BinaryPayloadDecoder(payload)` . The value returned from `context.getValues` is a list and hence the decoder fails when decoding to required formatters. The better way would be to use the class method `BinaryPayloadDecoder.fromRegisters(payload)` where in the payload needs to be a list of integers not bytestring. To make slave context store the data as `int` instead of `bytestring` you will have to use the method `to_registers()` on the builder object.

Here is how you can handle the data in both cases
```
from pymodbus.datastore import ModbusSlaveContext
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.payload import BinaryPayloadBuilder

builder = BinaryPayloadBuilder()
builder.add_16bit_int(100)
payload = builder.build()

context = ModbusSlaveContext()
context.setValues(3, 0, payload)

value = context.getValues(3, 0)

decoder = BinaryPayloadDecoder(value[0])  # Pass as bytestring instead of list
actual = decoder.decode_16bit_int() 
print(actual)

# The better way use to_registers and fromRegisters
builder = BinaryPayloadBuilder()
builder.add_16bit_int(100)
payload = builder.to_registers()

context = ModbusSlaveContext()
context.setValues(3, 0, payload)

value = context.getValues(3, 0)

decoder = BinaryPayloadDecoder.fromRegisters(value)
actual = decoder.decode_16bit_int()
print(actual)
```

closed 2018-09-18 by dhoomakethu

gbrucepayne, 2018-09-18:
Thanks I will try that and let you know.

I did look at the examples and it was not clear to me from those, I had to do a lot of trial and error with different methods.

gbrucepayne, 2018-09-18:
Thanks for clarifying with examples.  Your suggestion fixed my problem and misunderstanding of the operations.