pymodbus-dev/pymodbus #331: Hi All,

deverakondabalakrishna, 2018-09-19:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```

deverakondabalakrishna, 2018-09-19:
Hi All, I am able to execute pymodbus examples able to do master and slave communication successfully. In my project, I need to map Modbus slave to a MySQL database and read this data from Modbus master. I am new to Modbus and python programming. Please suggest me any python based Modbus examples

dhoomakethu, 2018-09-19:
Refer https://github.com/riptideio/pymodbus/blob/master/examples/common/dbstore_update_server.py. Let us know if you need any further help.

closed 2018-09-19 by dhoomakethu

Makimax35, 2021-12-23:
Hi, I need to read ModbusTCP IP addresses and registers listed into a MySQL table and read those datas registers into corresponding Modbus devices and log values into another MySQL table. I am new to Modbus and python programming.

1. Get devices modbus informations from mysql
2. Read modbus corredponding input registeres
3. Print results and see possible errors
4. Every 15 minutes Insert last values into mysql