pymodbus-dev/pymodbus #333: RTU framer stuck in infinite loop

jjrobsontime, 2018-09-21:
### Versions

* Python: 2.7.13
* OS: Debian 9.4
* Pymodbus: 1.5.2

### Pymodbus Specific
* Client: rtu - sync

### Description

What were you trying:
Polling a device for 30 registers at a time (READ_HOLDING_REGISTERS). Polling over RS485 using an FTDI USB-RS485 adapter. The polling is 19200 baud, parity N, stopbits 1. Making around 5 requests per second. I was testing what happens when the USB converter is removed while the program is running in order to write correct error handling. 

What went wrong,
The RTU framer sendPacket() method got stuck in an infinite loop. It printed a stream of 'Sleeping' messages to the log. This causes the whole client to hang.

And what did you expect?
I would expect the framer to wait for a time 'timeout' before giving up and raising an ConnectionException or returning an exception response.

### Code and Logs
```python
import logging
import time
from pymodbus.client.sync import  ModbusSerialClient

master = ModbusSerialClient(method='rtu', port='/dev/ttyUSB0', retry_on_empty=False, timeout=0.08)

while True:
    time.sleep(0.15)
    try:
        response = master.read_holding_registers(address=140, unit=4, count=30)
    except KeyboardInterrupt:
        exit()
    else:
        print response.registers
```

DEBUG    2018-09-20 17:01:46,344     sync                           New Transaction state 'SENDING'
DEBUG    2018-09-20 17:01:46,346     transaction                Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG    2018-09-20 17:01:46,418     transaction                Current transaction state - WAITING_FOR_REPLY
DEBUG    2018-09-20 17:01:46,420     transaction                Running transaction 358
DEBUG    2018-09-20 17:01:46,426     transaction                SEND: 0x4 0x3 0x0 0x8c 0x0 0x1e 0x4 0x7c
DEBUG    2018-09-20 17:01:46,427     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,437     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,441     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,452     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,458     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,464     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,469     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,473     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,477     rtu_framer                 Sleeping
DEBUG    2018-09-20 17:01:46,480     rtu_framer                 Sleeping

dhoomakethu, 2018-09-25:
@jjrobsontime Could you please give a try with v2.0.0 and let me know if the problem exists there as well. I am trying to reproduce this issue with out any success so far. What kind of USB-Serial converter are you using ? Also if possible please provide the logs before the USB cable is removed .

jjrobsontime, 2018-09-26:
Thanks @dhoomakethu. I'll give it a go on v2.0.0 when I get some time in the next day or two. I didn't realise v2.0.0 was out the very day I decided to test this! I'm using an FTDI USB/rs485 converter :https://www.ftdichip.com/Products/Cables/USBRS485.html

I acknowledge that it is unlikely to be an easily reproduce able bug as it involves removing the USB converter at the exact moment the transaction hangs but I managed it pretty consistently.

Some additional logs showing a successful transaction before the failure:
DEBUG    2018-09-20 17:07:35,801  transaction                Running transaction 56
DEBUG    2018-09-20 17:07:35,803  transaction                SEND: 0x4 0x3 0x0 0x8c 0x0 0x1e 0x4 0x7c
DEBUG    2018-09-20 17:07:35,804  rtu_framer                 Changing state to IDLE - Last Frame End -
DEBUG    2018-09-20 17:07:35,805  sync                           New Transaction state 'SENDING'
DEBUG    2018-09-20 17:07:35,807  transaction                Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG    2018-09-20 17:07:35,865  transaction                Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG    2018-09-20 17:07:35,867  transaction                 RECV: 0x4 0x3 0x3c 0x0 0x6 0x0 0x2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x6b 0x0 0x0 0x0 0x34 0x13 0xf1 0x0 0x9 0x0 0x1a 0x19 0x64 0x0 0x0 0x8 0xfd 0x13 0x8a 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xff 0x0 0x0 0x0 0x0 0xff 0xff 0xff 0x4b 0x12 0x78
DEBUG    2018-09-20 17:07:35,869  rtu_framer                 Getting Frame - 0x3 0x3c 0x0 0x6 0x0 0x2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x6b 0x0 0x0 0x0 0x34 0x13 0xf1 0x0 0x9 0x0 0x1a 0x19 0x64 0x0 0x0 0x8 0xfd 0x13 0x8a 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xff 0x0 0x0 0x0 0x0 0xff 0xff 0xff 0x4b
DEBUG    2018-09-20 17:07:35,870  factory                       Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG    2018-09-20 17:07:35,872  rtu_framer                 Frame advanced, resetting header!!
DEBUG    2018-09-20 17:07:35,873  transaction                Adding transaction 56
DEBUG    2018-09-20 17:07:35,874  transaction                Getting transaction 56
DEBUG    2018-09-20 17:07:35,875  transaction                Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
DEBUG    2018-09-20 17:07:35,891  transaction                Current transaction state - TRANSCATION_COMPLETE
DEBUG    2018-09-20 17:07:35,892  transaction                Running transaction 57
DEBUG    2018-09-20 17:07:35,896  transaction                SEND: 0x2 0x3 0x0 0x0 0x0 0x2 0xc4 0x38
DEBUG    2018-09-20 17:07:35,898  rtu_framer                 Changing state to IDLE - Last Frame End - 
DEBUG    2018-09-20 17:07:35,899  sync                           New Transaction state 'SENDING'
DEBUG    2018-09-20 17:07:35,906  transaction                Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG    2018-09-20 17:07:35,932  transaction                Current transaction state - WAITING_FOR_REPLY
DEBUG    2018-09-20 17:07:35,934  transaction                Running transaction 58
DEBUG    2018-09-20 17:07:35,935  transaction                SEND: 0x4 0x10 0x0 0x80 0x0 0x2 0x4 0x0 0x2 0x0 0x2a 0xca 0x1c
DEBUG    2018-09-20 17:07:35,936  rtu_framer                 Sleeping
DEBUG    2018-09-20 17:07:35,951  rtu_framer                 Sleeping
DEBUG    2018-09-20 17:07:35,955  rtu_framer                 Sleeping
DEBUG    2018-09-20 17:07:35,964  rtu_framer                 Sleeping
DEBUG    2018-09-20 17:07:35,970  rtu_framer                 Sleeping
DEBUG    2018-09-20 17:07:35,977  rtu_framer                 Sleeping
DEBUG    2018-09-20 17:07:35,981  rtu_framer                 Sleeping
DEBUG    2018-09-20 17:07:35,985  rtu_framer                 Sleeping

dhoomakethu, 2018-09-26:
Thanks @jjrobsontime for the detailed logs. I will take a look.

dhoomakethu, 2018-10-01:
I was able to reproduce this issue but in a different way, If there is an exception raise when receiving the message, the framer gets in to a infinite loop. I do not see any such exceptions being raised in your case. Are you using multiple threads to talk to the same slave ?

dhoomakethu, 2018-10-01:
@jjrobsontime Please give a try with the changes in this [branch](https://github.com/riptideio/pymodbus/tree/%23333-Infinite-Loop) and let me know your findings. Thanks.

dhoomakethu, 2018-10-03:
Thanks for the confirmation. This should be available with the next release some time this week.

dhoomakethu, 2018-10-03:
Fixed with 2.1.0

closed 2018-10-03 by dhoomakethu

rag2180, 2020-05-27:
@dhoomakethu - I am getting this same error in v.2.3.0.
**OS**: Raspbian GNU/Linux 8 (jessie)
**Hardware**: Raspberry Pi
**Connector**: RS485

Following are the logs:

2020-05-27 12:00:20,211 - pymodbus.transaction - DEBUG - Current transaction state - TRANSACTION_COMPLETE
2020-05-27 12:00:20,212 - pymodbus.transaction - DEBUG - Running transaction 1017
2020-05-27 12:00:20,214 - pymodbus.transaction - DEBUG - SEND: 0xe 0x3 0x0 0x0 0x0 0x6 0xc5 0x37
2020-05-27 12:00:20,215 - twisted - INFO - Timesync read from file - 1
2020-05-27 12:00:20,216 - pymodbus.framer.rtu_framer - DEBUG - Changing state to IDLE - Last Frame End - 1590560985.77, Current Time stamp - 1590561020.22
2020-05-27 12:00:20,219 - twisted - INFO - Timesync adding - 1
2020-05-27 12:00:20,220 - pymodbus.client.sync - DEBUG - New Transaction state 'SENDING'
2020-05-27 12:00:20,224 - twisted - INFO - **********POLLING ACTUATION.JSON**********
2020-05-27 12:00:20,228 - twisted - INFO - Exception Counter Value - 128
2020-05-27 12:00:20,231 - twisted - INFO - inside check_if_timesynced_now - 1 1
2020-05-27 12:00:20,234 - twisted - WARNING - Unable to read from Relay Board - 14 -- Error - Unable to hit Read Query - id - 14 - [Errno 5] Input/output error
2020-05-27 12:00:20,237 - pymodbus.transaction - DEBUG - Current transaction state - SENDING
2020-05-27 12:00:20,238 - pymodbus.transaction - DEBUG - Running transaction 1018
2020-05-27 12:00:20,239 - pymodbus.transaction - DEBUG - SEND: 0xe 0x3 0x0 0x0 0x0 0x6 0xc5 0x37
2020-05-27 12:00:20,240 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,245 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,250 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,255 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,260 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,265 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,271 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,279 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,286 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,292 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,297 - pymodbus.framer.rtu_framer - DEBUG - Sleeping
2020-05-27 12:00:20,303 - pymodbus.framer.rtu_framer - DEBUG - Sleeping