pymodbus-dev/pymodbus #340: Exception thrown by BinaryPayloadDecoder

dzimine, 2018-09-26:
### Versions

* Python: 2.7
* OS: Ubuntu 16.04 + AWS Greengrass
* Pymodbus: 2.0.0
* Modbus Hardware (if used): No

### Pymodbus Specific
* Server: tcp - sync/async
* Client: tcp - sync/async

### Description

Parsing payload sometimes throws an exception: 
```
Error: 'ascii' codec can't encode character u'\xbb' in position 87: ordinal not in range(128)
```
The same code works with exactly the same input data on OS-X and Ubuntu 16.04 but strangely fails when run as Lambda on AWS Greengrass core (deployed on the same Ubuntu 16.04 box). 

The code line causing the exception is https://github.com/riptideio/pymodbus/blob/master/pymodbus/payload.py#L344: 

```
        _logger.debug(unicode_string(handle))
```
Commenting it out or removing `unicode_string` conversation fixes the problem. 

### Code and Logs

```
[2018-09-25T13:26:09.47Z][DEBUG]-transaction.py:112,Current transaction state - TRANSACTION_COMPLETE
[2018-09-25T13:26:09.472Z][DEBUG]-transaction.py:116,Running transaction 24
[2018-09-25T13:26:09.474Z][DEBUG]-transaction.py:206,SEND: 0x0 0x18 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x8
[2018-09-25T13:26:09.475Z][DEBUG]-sync.py:75,New Transaction state 'SENDING'
[2018-09-25T13:26:09.477Z][DEBUG]-transaction.py:209,Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
[2018-09-25T13:26:09.479Z][DEBUG]-transaction.py:284,Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
[2018-09-25T13:26:09.48Z][DEBUG]-transaction.py:214,RECV: 0x0 0x18 0x0 0x0 0x0 0x13 0x1 0x3 0x10 0x18 0x3f 0xbb 0x48 0x0 0xd 0x0 0xd 0x0 0xd 0x0 0xd 0x0 0xd 0x0 0xd
[2018-09-25T13:26:09.481Z][DEBUG]-socket_framer.py:147,Processing: 0x0 0x18 0x0 0x0 0x0 0x13 0x1 0x3 0x10 0x18 0x3f 0xbb 0x48 0x0 0xd 0x0 0xd 0x0 0xd 0x0 0xd 0x0 0xd 0x0 0xd
[2018-09-25T13:26:09.482Z][DEBUG]-factory.py:246,Factory Response[ReadHoldingRegistersResponse: 3]
[2018-09-25T13:26:09.484Z][DEBUG]-transaction.py:363,Adding transaction 24
[2018-09-25T13:26:09.485Z][DEBUG]-transaction.py:374,Getting transaction 24
[2018-09-25T13:26:09.486Z][DEBUG]-transaction.py:180,Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'

[2018-09-25T13:26:09.487Z][DEBUG]-payload.py:299,[6207, 47944, 13, 13, 13, 13, 13, 13]
[2018-09-25T13:26:09.49Z][INFO]-function.py:67,Error: 'ascii' codec can't encode character u'\xbb' in position 87: ordinal not in range(128)
```

dhoomakethu, 2018-09-26:
@dzimine, this error is hard to fix on multiple platforms. I will accept your PR as a fix and will try to find a more reliable and foolproof solution later.

dhoomakethu, 2018-09-27:
New version with the fix is now available on pypu (v2.0.1)

closed 2018-09-27 by dhoomakethu