pymodbus-dev/pymodbus #353: Issue with read_input_registers in version 2.1.0

mpf82, 2018-10-16:
### Versions

* Python: 3.4, 64
* OS: Windows 10, 64
* Pymodbus: 2.1.0
* Modbus Hardware (if used): PM-3133

### Pymodbus Specific
* Client: rtu - sync

### Description

Since upgrading to version 2.1.0 I cannot read the whole input registers as I could with other versions.
Instead I receive an error:

> {'message': '[Input/Output] No Response received from the remote unit/Unable to decode response', 'fcode': 4, 'string': '[Input/Output] No Response received from the remote unit/Unable to decode response'}

### Code and Logs

```python
import logging
from pymodbus.constants import Defaults as ModbusDefaults
from pymodbus.client.sync import ModbusTcpClient, ModbusUdpClient, ModbusSerialClient

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def pymodbus_issue():
    params  = dict(
        method      = "rtu",
        port        = "COM4",
        stopbits    = ModbusDefaults.Stopbits,
        bytesize    = ModbusDefaults.Bytesize,
        parity      = ModbusDefaults.Parity,
        baudrate    = ModbusDefaults.Baudrate,
        timeout     = ModbusDefaults.Timeout,
    )
    client = ModbusSerialClient( **params )
    client.connect()
    rr = client.read_input_registers(0x1100, 18, unit=1)
    print(rr.__dict__)
    client.close()

pymodbus_issue()

```

### Log with pymodbus version 2.0.1
```
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x4 0x11 0x0 0x0 0x12 0x75 0x3b
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x4 0x24 0x0 0x0 0x0 0x0 0x57 0xa 0x3d 0x1c 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe8 0x69 0x3d 0x47 0x9b 0x40 0x3a 0xa9 0xf4 0x3d 0x3d 0x48 0xff 0xd0
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x4 0x24 0x0 0x0 0x0 0x0 0x57 0xa 0x3d 0x1c 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe8 0x69 0x3d 0x47 0x9b 0x40 0x3a 0xa9 0xf4 0x3d 0x3d 0x48
DEBUG:pymodbus.factory:Factory Response[ReadInputRegistersResponse: 4]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
{'transaction_id': 0, 'protocol_id': 0, 'registers': [0, 0, 22282, 15644, 0, 0, 0, 0, 0, 0, 0, 0, 59497, 15687, 39744, 15017, 62525, 15688], 'check': 0, 'unit_id': 1, 'skip_encode': False}
```

### Log with pymodbus version 2.1.0
```
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x4 0x11 0x0 0x0 0x12 0x75 0x3b
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Incomplete message received, Expected 41 bytes Recieved 16 bytes !!!!
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x4 0x24 0x0 0x0 0x0 0x0 0xaf 0x8c 0x3d 0x1c 0x0 0x0 0x0 0x0 0x0
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
{'message': '[Input/Output] No Response received from the remote unit/Unable to decode response', 'fcode': 4, 'string': '[Input/Output] No Response received from the remote unit/Unable to decode response'}

```

This *might* be related to issue #328 - however my code runs fine with pymodbus version 1.5.2 - that's why I have created a new issue.

Actually, I tested pretty much all pymodbus versions from 1.2.0 onwards, but the code "breaks" with version 2.1.0

If I change the numer of registers to read (currently 18) to something smaller it works. To be more specific, everything from 1 to 6 works, for everything else I get the mentioned error.

I have also tried with a third party Modbus viewer - also no problems there when trying to read 18 registers starting at 0x1100.

dhoomakethu, 2018-10-16:
@mpf82 Thanks for the report, I tried the latest (v2.1.0) from pypi , as well as on dev and master branches. I was not able to reproduce this issue (against simulator). I was able to read the maximum permissible register count (125) with out any issues. From the logs it seems that the response received is partial. If you could add the below log format and paste the logs again (both with 2.1.0 and 2.0.1) we can compare the timings and see if something is going wrong there.

```
FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)
```

mpf82, 2018-10-16:
Thanks for the fast reply.

### 2.0.1
```
[pymodbus, version 2.0.1]
2018-10-16 14:54:21,005 MainThread      DEBUG    transaction    :112      Current transaction state - IDLE
2018-10-16 14:54:21,005 MainThread      DEBUG    transaction    :116      Running transaction 1
2018-10-16 14:54:21,005 MainThread      DEBUG    transaction    :206      SEND: 0x1 0x4 0x11 0x0 0x0 0x12 0x75 0x3b
2018-10-16 14:54:21,006 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-10-16 14:54:21,006 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-10-16 14:54:21,036 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2018-10-16 14:54:21,036 MainThread      DEBUG    transaction    :214      RECV: 0x1 0x4 0x24 0x0 0x0 0x0 0x0 0x8 0xe 0x3d 0x1d 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe8 0x69 0x3d 0x47 0x9b 0x40 0x3a 0xa9 0xf4 0x3d 0x3d 0x48 0x55 0xf7
2018-10-16 14:54:21,036 MainThread      DEBUG    rtu_framer     :175      Getting Frame - 0x4 0x24 0x0 0x0 0x0 0x0 0x8 0xe 0x3d 0x1d 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe8 0x69 0x3d 0x47 0x9b 0x40 0x3a 0xa9 0xf4 0x3d 0x3d 0x48
2018-10-16 14:54:21,036 MainThread      DEBUG    factory        :246      Factory Response[ReadInputRegistersResponse: 4]
2018-10-16 14:54:21,036 MainThread      DEBUG    rtu_framer     :110      Frame advanced, resetting header!!
2018-10-16 14:54:21,036 MainThread      DEBUG    transaction    :418      Adding transaction 1
2018-10-16 14:54:21,037 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
{'transaction_id': 0, 'check': 0, 'protocol_id': 0, 'unit_id': 1, 'skip_encode': False, 'registers': [0, 0, 2062, 15645, 0, 0, 0, 0, 0, 0, 0, 0, 59497, 15687, 39744, 15017, 62525, 15688]}
```

### 2.1.0
```
[pymodbus, version 2.1.0]
2018-10-16 14:56:13,637 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2018-10-16 14:56:13,638 MainThread      DEBUG    transaction    :115      Running transaction 1
2018-10-16 14:56:13,638 MainThread      DEBUG    transaction    :206      SEND: 0x1 0x4 0x11 0x0 0x0 0x12 0x75 0x3b
2018-10-16 14:56:13,638 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-10-16 14:56:13,638 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-10-16 14:56:13,653 MainThread      DEBUG    transaction    :282      Incomplete message received, Expected 41 bytes Recieved 16 bytes !!!!
2018-10-16 14:56:13,653 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2018-10-16 14:56:13,653 MainThread      DEBUG    transaction    :214      RECV: 0x1 0x4 0x24 0x0 0x0 0x0 0x0 0x20 0x98 0x3d 0x1a 0x0 0x0 0x0 0x0 0x0
2018-10-16 14:56:13,653 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
{'message': '[Input/Output] No Response received from the remote unit/Unable to decode response', 'string': '[Input/Output] No Response received from the remote unit/Unable to decode response', 'fcode': 4}
```

dhoomakethu, 2018-10-16:
@mpf82 Thanks for the quick response, Could you please execute your test against this [branch](https://github.com/riptideio/pymodbus/tree/%23353-Error-Reading-Registers) ? I have just added the debug logs nothing else, just to see whats happening. You could also edit pymodbus/client/sync.py and `_logger.debug("Available in read buffer - {}".format(avaialble))` after line 536.

dhoomakethu, 2018-10-16:
My bad, thats correct it's not called for register reads, sorry for that. The socket read is for some reason not reading the full data , If you print `len(result)` after `result = self.socket.read(size)`. You would see that the read is incomplete (13 bytes). Interesting part is the read timeout is 3 secs but the actual socket read is timing with in 30ms. 

Are you running these tests in separate virtualenv and is the `pyserial` version same for both 2.0.1 and 2.1.0 ? Please share the pyserial version.

mpf82, 2018-10-16:
Yes, the tests are in a separate venv.

``print(serial.__version__)`` outputs 3.4 for both 2.01 and 2.1.0

which would match the output of pip:

> pip.exe install -I pymodbus==2.0.1
``Collecting pymodbus==2.0.1``
``Successfully installed pymodbus-2.1.0 pyserial-3.4 six-1.11.0`` 

(Yes, it really print "pymodbus-2.1.0" when installing 2.0.1, but I can tell by version.py and the behaviour that the version really changed.)

> pip.exe install -I pymodbus==2.1.0
``Collecting pymodbus==2.1.0``
``Successfully installed pymodbus-2.1.0 pyserial-3.4 six-1.11.0``

dhoomakethu, 2018-10-16:
With this , Could you please share the logs with timing info ?  Ideally , the response should be ready within in the read timeout. If possible and time permits, could you please also run the tests with out this fix but with bigger timeout (say 6 sec) ?. Thanks.

mpf82, 2018-10-16:
I'll look into int tomorrow.

mpf82, 2018-10-17:
Increasing the timeout to e.g. 10 or even more seems to have no effect at all. I also tested with a timeout of 1000, but it really makes no difference and I always the the afore mentioned error when I run the sample program w/o my fix.

The response of the ``self.socket.read(...)`` is almost immediate, no matter the timeout value.

Maybe it's worth noting that I'm connected to the device using an **USB adapter**, similar to [this one](https://images-na.ssl-images-amazon.com/images/I/7149KI4UoKL._SX466_.jpg).

But still, everything works out-of-the-box with version 2.0.1

I have added debugging information to my quick fix:

````python
    def _recv(self, size):
        if not self.socket:
            raise ConnectionException(self.__str__())
        if size is None:
            size = self._wait_for_data()
        _logger.debug("Before first read. Expected size: {}".format( size ))
        result = self.socket.read(size)
        _logger.debug("After first read. Actual result size: {}".format( len(result) ))
        while len(result)<size:
            remaining_size = size-len(result)
            _logger.debug("Keep reading: {}".format(remaining_size))
            result += self.socket.read(remaining_size)
            _logger.debug("New result size: {}".format( len(result) ))
        return result
````

Here's the log:

````
[pymodbus, version 2.1.0]
2018-10-17 09:05:53,688 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2018-10-17 09:05:53,688 MainThread      DEBUG    transaction    :115      Running transaction 1
2018-10-17 09:05:53,688 MainThread      DEBUG    transaction    :206      SEND: 0x1 0x4 0x11 0x0 0x0 0x12 0x75 0x3b
2018-10-17 09:05:53,688 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-10-17 09:05:53,688 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-10-17 09:05:53,689 MainThread      DEBUG    sync           :555      Before first read. Expected size: 2
2018-10-17 09:05:53,700 MainThread      DEBUG    sync           :557      After first read. Actual result size: 2
2018-10-17 09:05:53,701 MainThread      DEBUG    sync           :555      Before first read. Expected size: 39
2018-10-17 09:05:53,702 MainThread      DEBUG    sync           :557      After first read. Actual result size: 12
2018-10-17 09:05:53,702 MainThread      DEBUG    sync           :560      Keep reading: 27
2018-10-17 09:05:53,716 MainThread      DEBUG    sync           :562      New result size: 39
2018-10-17 09:05:53,717 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2018-10-17 09:05:53,717 MainThread      DEBUG    transaction    :214      RECV: 0x1 0x4 0x24 0x0 0x0 0x0 0x0 0x24 0xde 0x3d 0x86 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe8 0x69 0x3d 0x47 0x9b 0x40 0x3a 0xa9 0xf4 0x3d 0x3d 0x48 0x1 0x32
2018-10-17 09:05:53,717 MainThread      DEBUG    rtu_framer     :175      Getting Frame - 0x4 0x24 0x0 0x0 0x0 0x0 0x24 0xde 0x3d 0x86 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe8 0x69 0x3d 0x47 0x9b 0x40 0x3a 0xa9 0xf4 0x3d 0x3d 0x48
2018-10-17 09:05:53,717 MainThread      DEBUG    factory        :246      Factory Response[ReadInputRegistersResponse: 4]
2018-10-17 09:05:53,717 MainThread      DEBUG    rtu_framer     :110      Frame advanced, resetting header!!
2018-10-17 09:05:53,717 MainThread      DEBUG    transaction    :418      Adding transaction 1
2018-10-17 09:05:53,717 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
{'skip_encode': False, 'protocol_id': 0, 'check': 0, 'unit_id': 1, 'registers': [0, 0, 9438, 15750, 0, 0, 0, 0, 0, 0, 0, 0, 59497, 15687, 39744, 15017, 62525, 15688], 'transaction_id': 0}
````

I've run the sample like 20 times, and the first time the function attempts to read the 39 bytes, the actual size is something between 12 and 15 bytes, most of the time 13.

After entering the while loop, the other bytes are read in one go. So far I didn't encounter a case where the loop was executed more then once - at least not when reading **18 registers**.

If I attempt to read **all 125 register at once**, the log looks like this:

````
2018-10-17 09:11:52,521 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2018-10-17 09:11:52,521 MainThread      DEBUG    transaction    :115      Running transaction 1
2018-10-17 09:11:52,522 MainThread      DEBUG    transaction    :206      SEND: 0x1 0x4 0x11 0x0 0x0 0x7d 0x35 0x17
2018-10-17 09:11:52,522 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-10-17 09:11:52,522 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-10-17 09:11:52,522 MainThread      DEBUG    sync           :555      Before first read. Expected size: 2
2018-10-17 09:11:52,535 MainThread      DEBUG    sync           :557      After first read. Actual result size: 2
2018-10-17 09:11:52,535 MainThread      DEBUG    sync           :555      Before first read. Expected size: 253
2018-10-17 09:11:52,537 MainThread      DEBUG    sync           :557      After first read. Actual result size: 13
2018-10-17 09:11:52,537 MainThread      DEBUG    sync           :560      Keep reading: 240
[... cut ...] xb8 0x66 0x5e 0x69
2018-10-17 09:11:52,664 MainThread      DEBUG    factory        :246      Factory Response[ReadInputRegistersResponse: 4]
2018-10-17 09:11:52,664 MainThread      DEBUG    rtu_framer     :110      Frame advanced, resetting header!!
2018-10-17 09:11:52,664 MainThread      DEBUG    transaction    :418      Adding transaction 1
2018-10-17 09:11:52,664 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
````

dhoomakethu, 2018-10-17:
Thanks for the logs, interesting to know that socket.read() is not waiting for the complete data till read timeout. I have tried with almost all the modbus devices we have here (on linux) and the issue is not reproducible. The only major diff between 2.0.1 and 2.1.0 is introduction of `interCharTimeout` (refer [line](https://github.com/riptideio/pymodbus/blob/master/pymodbus/client/sync.py#L476) ). Could you please comment out that line and run tests one more time (with and with out your patch)?

mpf82, 2018-10-17:
That line seems to be the culprit.

If I comment it out, the whole data gets read in one go:

````
2018-10-17 09:45:12,895 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-10-17 09:45:12,895 MainThread      DEBUG    sync           :555      Before first read. Expected size: 2
2018-10-17 09:45:12,909 MainThread      DEBUG    sync           :557      After first read. Actual result size: 2
2018-10-17 09:45:12,909 MainThread      DEBUG    sync           :555      Before first read. Expected size: 253
2018-10-17 09:45:13,037 MainThread      DEBUG    sync           :557      After first read. Actual result size: 253
2018-10-17 09:45:13,037 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
````

dhoomakethu, 2018-10-17:
So my guess was correct, I went through the serial implementation for windows and found this line in `serialwin32.py`
```
        if self._timeout != 0 and self._inter_byte_timeout is not None:
            timeouts.ReadIntervalTimeout = max(int(self._inter_byte_timeout * 1000), 1)
```
The windows documentation for [COMTIMEOUT](https://docs.microsoft.com/en-us/windows/desktop/api/winbase/ns-winbase-_commtimeouts) has this description for `ReadIntervalTimeout`

> ReadIntervalTimeout
> 
> The maximum time allowed to elapse before the arrival of the next byte on the communications line, in milliseconds. If the interval between the arrival of any two bytes exceeds this amount, the ReadFile operation is completed and any buffered data is returned. A value of zero indicates that interval time-outs are not used.

So it looks like the data is being received slower than expected (which basically does not adhere with modbus specs, refer image below ). 
[image]

mpf82, 2018-10-17:
Like I said, I'm connected via an USB adapter, maybe that's slowing down the communication?

And a timeout of ``0.000859375`` still seems pretty low to me - is that an "OK" value?

dhoomakethu, 2018-10-17:
Hmm, that 859.375 micro seconds is still ok as per modbus spec per-say  but yes, using an external USB-RS485 can impose its own limitations. How about providing an option to disable this timeout during client creation ? By default it is enabled but for cases like yours , one could choose to turnoff it .

mpf82, 2018-10-17:
I tested around, and the "sweet spot" in my case seems to be 

    self.socket.interCharTimeout = 0.016 # self.inter_char_timeout

where the communication would sometime fail and sometime succeed.

As in ``0.015`` -> always fail 
and ``0.017`` -> always succeed

But yeah, an option to disable it completely would be nice.

dhoomakethu, 2018-10-17:
I am planning to introduce another kwarg `strict` which is `True` by default.
```
 :param strict:  Use Inter char timeout for baudrates <= 19200 (adhere
        to modbus standards)
....
self._strict = kwargs.get("strict", True)

....
        if self.method == "rtu":
            if self._strict:
                self.socket.interCharTimeout = self.inter_char_timeout
```

dhoomakethu, 2019-01-14:
Fix Now part of branch [pymodbus-2.2.0](https://github.com/riptideio/pymodbus/tree/pymodbus-2.2.0), please give it a try and feedback

mpf82, 2019-01-28:
Sorry for the late reply.

Yes, I can confirm that with version ``2.2.0rc1`` setting ``strict      = False,`` will fix the issue.

closed 2019-01-28 by dhoomakethu

f-steff, 2020-11-19:
Two year later, "client.inter_char_timeout = 0.05" just saved my ass.
It seems like not everything is well in this code.