pymodbus-dev/pymodbus #357: Extension hook to support messages with custom function codes

laberning, 2018-11-02:
### Versions

* Python: 3.6.5
* OS: Mac OS
* Pymodbus: 2.1.0
* Modbus Hardware (if used):  Danfoss ECL 310

### Pymodbus Specific
* Client: tcp - sync/async

### Description

Add an extension hook to pymodbus to support messages with custom function codes.

[custom_message.py](https://github.com/riptideio/pymodbus/blob/master/examples/common/custom_message.py) shows an example for a simple custom message. It is possible to use this approach to extend the supported request messages with messages that use custom function codes.

However so far it is not possible to parse the corresponding response messages, since there is no way to extend the `ClientDecoder. __function_table` without patching the library.

See [here](https://stackoverflow.com/questions/52919426/how-to-implement-a-message-with-a-custom-function-code-in-pymodbus) for an example of the problem.

dhoomakethu, 2018-12-03:
@laberning Please give a try on [this](https://github.com/riptideio/pymodbus/tree/%23357-Custom-Function) branch. Refer examples 
https://github.com/riptideio/pymodbus/blob/%23357-Custom-Function/examples/common/custom_message.py
and 
https://github.com/riptideio/pymodbus/blob/%23357-Custom-Function/examples/common/custom_synchronous_server.py
on usage.

dhoomakethu, 2019-01-14:
Fix Now part of branch [pymodbus-2.2.0](https://github.com/riptideio/pymodbus/tree/pymodbus-2.2.0), please give it a try and feedback

dhoomakethu, 2019-01-16:
Fixed and available as part of [v2.2.0rc1](https://github.com/riptideio/pymodbus/releases/tag/v2.2.0rc1).
`pip install pymodbus==2.2.0rc1`

closed 2019-01-16 by dhoomakethu