pymodbus-dev/pymodbus #359: Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)

Yaser-Motiee, 2018-11-20:
### Versions

* Python: 3.7.0
* OS: 10-64bit
* Pymodbus: 2.1.1
* Modbus Hardware (if used): 

### Pymodbus Specific
* Client: rtu - sync

### Description

Hello .
When I want to read information from two clients, I read the value of the first client, but it will make a mistake for the second client!

### Code
```python
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
import logging

client = ModbusClient(method='rtu', port='COM13', parity='N', stopbits=2, bytesize=8, baudrate=19200, timeout=3)

FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def read_sens_ds(address, count, unit):
	result = client.read_holding_registers(address=address, count=count, unit=unit)
	temp = result.registers[0] / 100
	return temp

def read_sens(address, count, unit):
	result = client.read_holding_registers(address=address, count=count, unit=unit)
	temp = result.registers[0] / 100
	humid = result.registers[1] / 100
	return temp, humid

if __name__ == "__main__":
	print(read_sens_ds(0, 1, 1))
	print(read_sens(0, 2, 2))
```
### Logs
```
2018-11-20 04:53:47,281 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2018-11-20 04:53:47,281 MainThread      DEBUG    transaction    :115      Running transaction 1
2018-11-20 04:53:47,281 MainThread      DEBUG    transaction    :206      SEND: 0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
2018-11-20 04:53:47,281 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-11-20 04:53:47,281 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
23.12
2018-11-20 04:53:47,759 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2018-11-20 04:53:47,760 MainThread      DEBUG    transaction    :214      RECV: 0x1 0x3 0x2 0x9 0x8 0xbf 0xd2
2018-11-20 04:53:47,760 MainThread      DEBUG    rtu_framer     :175      Getting Frame - 0x3 0x2 0x9 0x8
2018-11-20 04:53:47,760 MainThread      DEBUG    factory        :246      Factory Response[ReadHoldingRegistersResponse: 3]
2018-11-20 04:53:47,761 MainThread      DEBUG    rtu_framer     :110      Frame advanced, resetting header!!
2018-11-20 04:53:47,761 MainThread      DEBUG    transaction    :418      Adding transaction 1
2018-11-20 04:53:47,761 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2018-11-20 04:53:47,762 MainThread      DEBUG    transaction    :111      Current transaction state - TRANSACTION_COMPLETE
2018-11-20 04:53:47,762 MainThread      DEBUG    transaction    :115      Running transaction 2
2018-11-20 04:53:47,762 MainThread      DEBUG    transaction    :206      SEND: 0x2 0x3 0x0 0x0 0x0 0x2 0xc4 0x38
2018-11-20 04:53:47,762 MainThread      DEBUG    rtu_framer     :253      Changing state to IDLE - Last Frame End - 1542677027.759198, Current Time stamp - 1542677027.762195
2018-11-20 04:53:47,763 MainThread      DEBUG    rtu_framer     :261      Waiting for 3.5 char before next send - 2.005 ms
2018-11-20 04:53:47,767 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-11-20 04:53:47,768 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-11-20 04:53:50,770 MainThread      DEBUG    transaction    :218      Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
2018-11-20 04:53:50,771 MainThread      DEBUG    rtu_framer     :225      Frame - [b''] not ready
2018-11-20 04:53:50,771 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

dhoomakethu, 2018-11-20:
are both the slaves connected on the same interface ?

Yaser-Motiee, 2018-11-20:
Yes, both of them connected. When each of the slaves is connected alone, they get the right answer.
It's just right when they are connected and only a slave is called.

dhoomakethu, 2018-11-20:
Can you share the logs of them individually connected ?

Yaser-Motiee, 2018-11-20:
Yes, Sure

Yaser-Motiee, 2018-11-21:
In this case, the loop is complete, but this is not desirable to me.

> I run the loop five times so that the logs do not go up.

**Code**
```
if __name__ == "__main__":
	for _ in range(5):
		print(read_sens_ds(0, 1, 1))
		time.sleep(1)
		print(read_sens(0, 2, 2))
		time.sleep(1)
```
**Logs**
```
2018-11-21 11:55:48,204 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2018-11-21 11:55:48,204 MainThread      DEBUG    transaction    :115      Running transaction 1
2018-11-21 11:55:48,204 MainThread      DEBUG    transaction    :206      SEND: 0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
2018-11-21 11:55:48,204 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-11-21 11:55:48,205 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
24.37
2018-11-21 11:55:48,479 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2018-11-21 11:55:48,479 MainThread      DEBUG    transaction    :214      RECV: 0x1 0x3 0x2 0x9 0x85 0x7f 0xb7
2018-11-21 11:55:48,479 MainThread      DEBUG    rtu_framer     :175      Getting Frame - 0x3 0x2 0x9 0x85
2018-11-21 11:55:48,480 MainThread      DEBUG    factory        :246      Factory Response[ReadHoldingRegistersResponse: 3]
2018-11-21 11:55:48,480 MainThread      DEBUG    rtu_framer     :110      Frame advanced, resetting header!!
2018-11-21 11:55:48,480 MainThread      DEBUG    transaction    :418      Adding transaction 1
2018-11-21 11:55:48,481 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2018-11-21 11:55:49,482 MainThread      DEBUG    transaction    :111      Current transaction state - TRANSACTION_COMPLETE
2018-11-21 11:55:49,482 MainThread      DEBUG    transaction    :115      Running transaction 2
2018-11-21 11:55:49,483 MainThread      DEBUG    transaction    :206      SEND: 0x2 0x3 0x0 0x0 0x0 0x2 0xc4 0x38
2018-11-21 11:55:49,483 MainThread      DEBUG    rtu_framer     :253      Changing state to IDLE - Last Frame End - 1542788748.478566, Current Time stamp - 1542788749.483172
2018-11-21 11:55:49,483 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-11-21 11:55:49,484 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-11-21 11:55:50,396 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2018-11-21 11:55:50,397 MainThread      DEBUG    transaction    :214      RECV: 0x2 0x3 0x4 0x9 0xb6 0xf 0x16 0xaf 0x77
2018-11-21 11:55:50,397 MainThread      DEBUG    rtu_framer     :175      Getting Frame - 0x3 0x4 0x9 0xb6 0xf 0x16
2018-11-21 11:55:50,397 MainThread      DEBUG    factory        :246      Factory Response[ReadHoldingRegistersResponse: 3]
2018-11-21 11:55:50,398 MainThread      DEBUG    rtu_framer     :110      Frame advanced, resetting header!!
2018-11-21 11:55:50,398 MainThread      DEBUG    transaction    :418      Adding transaction 2
2018-11-21 11:55:50,398 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
(24.86, 38.62)
2018-11-21 11:55:51,398 MainThread      DEBUG    transaction    :111      Current transaction state - TRANSACTION_COMPLETE
2018-11-21 11:55:51,398 MainThread      DEBUG    transaction    :115      Running transaction 3
2018-11-21 11:55:51,399 MainThread      DEBUG    transaction    :206      SEND: 0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
2018-11-21 11:55:51,399 MainThread      DEBUG    rtu_framer     :253      Changing state to IDLE - Last Frame End - 1542788750.396194, Current Time stamp - 1542788751.399827
2018-11-21 11:55:51,399 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2018-11-21 11:55:51,400 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
24.37
2018-11-21 11:55:51,921 MainThread      DEBUG    transaction    :284      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2018-11-21 11:55:51,921 MainThread      DEBUG    transaction    :214      RECV: 0x1 0x3 0x2 0x9 0x85 0x7f 0xb7
2018-11-21 11:55:51,922 MainThread      DEBUG    rtu_framer     :175      Getting Frame - 0x3 0x2 0x9 0x85
2018-11-21 11:55:51,922 MainThread      DEBUG    factory        :246      Factory Response[ReadHoldingRegistersResponse: 3]
2018-11-21 11:55:51,922 MainThread      DEBUG    rtu_framer     :110      Frame advanced, resetting header!!
2018-11-21 11:55:51,923 MainThread      DEBUG    transaction    :418      Adding transaction 3
2018-11-21 11:55:51,923 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2018-11-21 11:55:52,923 MainThread      DEBUG    transaction    :111      Current transaction state - TRANSACTION_COMPLETE
2018-11-21 11:55:52,923 MainThread      DEBUG    transaction    :115      Running transaction 4
2018-11-21 11:55:52,923 MainThread      DEBUG    transaction    :206      SEND: 0x2 0x3 0x0 0x0 0x0 0x2 0xc4 0x38
2018-11-21 11:55:52,924 MainThread      DEBUG    rtu_framer     :253      Changing state to IDLE - Last Frame End - 1542788751.921097, Current Time stamp - 1542788752.923109
[... cut ...]
2018-11-21 11:56:01,885 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2018-11-21 11:56:04,888 MainThread      DEBUG    transaction    :218      Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
2018-11-21 11:56:04,889 MainThread      DEBUG    rtu_framer     :225      Frame - [b''] not ready
2018-11-21 11:56:04,889 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

dhoomakethu, 2018-11-21:
Thanks for the logs, I was more interested in the complete logs for 10 iteration in the first case or the second (with 1 sec sleep in between). In any case ,I would definitely see this as an issue with the serial line rather than pymodbus.  we use [this](https://www.startech.com/Cards-Adapters/Serial-Cards-Adapters/USB-to-RS-422-485-Converter-with-Isolation~ICUSB422IS) adapter in our test environment to connect multiple devices on the same port and it works fine as expected. Please share the specs of the usb-serial adapter you are using , I will give a try if I find it locally. I will be labelling this as not an issue.

Yaser-Motiee, 2018-11-21:
Thanks a lot, I use the usb to rs-485 converter, which is the CH340 chip. 
[USB to RS485 Converter Adapter ch340T ](https://www.amazon.in/Converter-Adapter-ch340T-64-bit-Supported/dp/B00NKAJGZM)

Yaser-Motiee, 2018-11-21:
> Thanks for the logs, I was more interested in the complete logs for 10 iteration in the first case or the second (with 1 sec sleep in between). In any case ,I would definitely see this as an issue with the serial line rather than pymodbus. we use [this](https://www.startech.com/Cards-Adapters/Serial-Cards-Adapters/USB-to-RS-422-485-Converter-with-Isolation~ICUSB422IS) adapter in our test environment to connect multiple devices on the same port and it works fine as expected. Please share the specs of the usb-serial adapter you are using , I will give a try if I find it locally. I will be labelling this as not an issue.
> 
> P.S. If you are still unsure, please give it a try with other modbus packages such as modbus-tk/minimal-modbus and see if the issue could be reproduced there as well. If you do not see these issues with those packages. Do share your code sample and logs. Would definitely help to capture the corner cases.

Yes, that's right. I tried the mimimal modbus and it was the same problem.

dhoomakethu, 2018-11-22:
Thanks for the confirmation, I shall close this issue as of now.

closed 2018-11-22 by dhoomakethu