pymodbus-dev/pymodbus #360: Async Serial Client Can't Recover from task cancellations

pazzarpj, 2018-11-30:
### Versions

* Python: 3.5
* OS: Windows/Linux
* Pymodbus: 2.1
* Modbus Hardware (if used): Serial loopback

### Pymodbus Specific
* Client: rtu - async

### Description
I encountered this issue when trying to read asynchronously after performing a task.cancel on an awaited read.

```python
import async_timeout
with async_timeout.timeout(1) as cm:
    await client.protocol.read_holding_registers(0, 1)
```
If the timeout is allowed to kick in (ie. the read takes longer than 1 second, by not having server running or something) , you have a cancelled task that isn't cleaned up by the transaction handler.
In the socket based systems this is fine as the transaction ID is explicitly retrieved via dictionary call but in the FifoTransactionManager, you now have a lurking cancelled task that is returned next time we call to the client. Eg.
[image]

So the next time we 
```python
await client.protocol.read_holding_registers(0,1)
```
We are going to be getting the cancelled future from the transaction manager when we call getTransaction(). This means that the await task is never woken up and you have crashed it until task.cancel() is called, which will mean that next time it is called there is another cancelled future waiting to be returned.

dhoomakethu, 2018-12-01:
@pazzarpj Thanks for the report and the PR, will take a look in to it soon.

janiversen, 2022-05-25:
you miss an await for the timeout:
```python
import async_timeout
await with async_timeout.timeout(1) as cm:
    await client.protocol.read_holding_registers(0, 1)
```
that should solve your problem.

janiversen, 2022-06-30:
Cannot be reproduced in 3.0dev4.

closed 2022-06-30 by janiversen