pymodbus-dev/pymodbus #364: SqlSlaveContext seems not work

zeinsteinz, 2018-12-07:
### Versions

* Python: 3.5.2
* OS: ubuntu 16.04
* Pymodbus: 2.1.0
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp - sync/async
* Client: tcp - sync/async

### Description

I am running the "examples/common/dbstore_update_server.py" without any modification, and it works fine when updating related datastores. However, when I try to use a client to request holding registers, I always get illegal address response. Could you please help me to figure the problem out? 

### Code and Logs
The server code is just exactly as same as "dbstore_update_server.py".
My client code is as follows:

```python
 # code and logs here.
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.mei_message import ReadDeviceInformationRequest
from pymodbus.device import ModbusDeviceIdentification

UNIT = 0x01

client = ModbusClient('localhost', port=5020)
res = client.read_holding_registers(0, 2, unit=UNIT)
print(res)
```

dhoomakethu, 2018-12-07:
You will have to access the correct register, the example generates rando, register address and the register you are trying to read might not actuall exists.
Check the db for the correct register to poll or change this [line](https://github.com/riptideio/pymodbus/blob/master/examples/common/dbstore_update_server.py#L61) in the code .

zeinsteinz, 2018-12-07:
sorry, I did not make it clear. I am sure that the registers I try to access are available. For example, I query "pymodbus.db", and the results are as follows.

sqlite> select * from pymodbus;
h|40001|111
h|40002|111
h|40003|111
h|40004|111
h|40005|111
h|40006|111
h|40007|111
h|40008|111
h|29549|538
h|60700|3120
h|49585|3983
h|58959|820
h|32452|368
h|4557|52

Then I use my client to make a request like this:
res = client.read_holding_registers(0, 2, unit=UNIT)
This is supposed to request holding registers 40001 and 40002, but I still get illegal address.

dhoomakethu, 2018-12-07:
There is a fix already in place, please give it a try  with `dev` branch. Refer #351

zeinsteinz, 2018-12-10:
I tried, but it seems not work yet.

dhoomakethu, 2018-12-10:
Please note, you will have to use the offset ( `offset = address-1`) when executing the read transaction. Refer the image below. If you check the db , the address in the db would `offset + 1`

```
sqlite> select * from pymodbus where "index"=1365 or "index"=51804;
type|index|value
h|1365|5682
h|51804|7427
```

[image]

closed 2018-12-11 by dhoomakethu