pymodbus-dev/pymodbus #365: Question about write register when using SqlSlaveContext

zeinsteinz, 2018-12-11:
### Versions

* Python: 3.5.2
* OS: Ubuntu 16.04
* Pymodbus: 2.1.1-dev
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp - sync/async
* Client: tcp - sync/async

### Description

I failed to write register when using SqlSlaveContext, since nothing is written to the database. I find that the problem is related to "_check" function in "sql_datastore.py". Every time when we try to write registers, "_check" is called to check whether these registers already have values. According to the codes, if there are already values, it will return False, which means it will not overwrite these values. I would like to konw whether this is a bug or by design?

### Code and Logs
start from line 134 in sql_datastore.py
```python
 # code and logs here.
    def _check(self, type, offset, values):
        result = self._get(type, offset, count=1)
        return False if len(result) > 0 else True
```

dhoomakethu, 2018-12-11:
looks like a bug. Will check and revert.

dhoomakethu, 2019-01-14:
Fix Now part of branch [pymodbus-2.2.0](https://github.com/riptideio/pymodbus/tree/pymodbus-2.2.0), please give it a try and feedback

dhoomakethu, 2019-01-16:
Fixed and available as part of [v2.2.0rc1](https://github.com/riptideio/pymodbus/releases/tag/v2.2.0rc1).
`pip install pymodbus==2.2.0rc1`

closed 2019-01-16 by dhoomakethu