pymodbus-dev/pymodbus #368: RTU broadcast doesn't function as expected using client.write_registers(add, val, unit=0)

muhlbaier, 2018-12-18:
### Versions

* Python: 2.7.15
* OS: Win 7
* Pymodbus: 2.1.0
* Modbus Hardware (if used): pymodbus on both sides

### Pymodbus Specific
* Server: rtu - sync
* Client: rtu - sync

### Description

When there are multiple slaves on a pymodbus server I would expect them all to receive a broadcast message but none receive the message. There doesn't seem to be any special handling for the broadcast address. If I were to create a slave context at address 0 it alone would respond to the broadcast address. The below example creates slave 1 and 2 and writes to slave 1 then slave 2 then writes to 0 to broadcast to both of them but this doesn't work as expected.

### Code and Logs

```python
import threading
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.server.sync import StartSerialServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.transaction import ModbusRtuFramer

# custom datablock to print what is going on
class MyDataBlock(ModbusSequentialDataBlock):
    def __init__(self, address, values, slave_id):
        self.slave_id = slave_id
        super(MyDataBlock, self).__init__(address, values)

    def setValues(self, address, values):
        print "slave " + str(self.slave_id) + " write address: " + str(address) + " values: " + str(values)
        super(MyDataBlock, self).setValues(address, values)

# create two slaves with unit ids 1 and 2
store = dict()
store[1] = ModbusSlaveContext(di=None, co=None, hr=MyDataBlock(0, [0] * 100, 1), ir=None)
store[2] = ModbusSlaveContext(di=None, co=None, hr=MyDataBlock(0, [0] * 100, 2), ir=None)

context = ModbusServerContext(slaves=store, single=False)
# initialize the server information
identity = ModbusDeviceIdentification()
identity.VendorName = 'Pymodbus'
identity.ProductCode = 'PM'
identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
identity.ProductName = 'Pymodbus Server'
identity.ModelName = 'Pymodbus Server'
identity.MajorMinorRevision = '1.0'

def server_thread():
    StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port="COM83", timeout=0.1, baudrate=115200)

server = threading.Thread(target=server_thread)
server.daemon = True
server.start()

client = ModbusClient(method='rtu', port="COM82", baudrate=115200, timeout=1)
client.connect()
# write to slave 1
registers = [2, 3, 4, 5]
request_result = client.write_registers(1, registers, unit=1)
print request_result
# write to slave 2
registers = [6, 7, 8, 9]
request_result = client.write_registers(1, registers, unit=2)
print request_result
# broadcast to both does not work
registers = [1, 1, 1, 1]
request_result = client.write_registers(1, registers, unit=0)
print request_result

```
Actual Console Output:
slave 1 write address: 2 values: [2, 3, 4, 5]
WriteMultipleRegisterResponse (1,4)
slave 2 write address: 2 values: [6, 7, 8, 9]
WriteMultipleRegisterResponse (1,4)
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)

Expected Console Output:
slave 1 write address: 2 values: [2, 3, 4, 5]
WriteMultipleRegisterResponse (1,4)
slave 2 write address: 2 values: [6, 7, 8, 9]
WriteMultipleRegisterResponse (1,4)
slave 1 write address: 2 values: [1, 1, 1, 1]
slave 2 write address: 2 values: [1, 1, 1, 1]

muhlbaier, 2018-12-19:
If there is anything I can do to assist in implementing this feature please let me know.

dhoomakethu, 2018-12-19:
@muhlbaier thanks for asking, If you have a fix handy, please raise a PR.

muhlbaier, 2019-01-09:
There is another related issue where when writing a broadcast message (sending to unit_id 0) the transaction will always timeout as current implementation tries to read bytes but there is no response when writing to unit_id 0 according to the modbus specs.

@dhoomakethu I have a fix already for the above issue but I get authentication errors when trying to publish my new branch (based on "dev") and push the related commit. Do you know how to fix that? I could also fork and raise the PR from there.

dhoomakethu, 2019-01-09:
You will have to fork the repo and raise a PR . You will not be able to push directly to the dev Branch

closed 2021-03-08 by github-actions[bot]