pymodbus-dev/pymodbus #369: timeout in client.sync._recv

MarekLew, 2019-01-04:
I find maby ridiculus problem in https://github.com/riptideio/pymodbus/blame/master/pymodbus/client/sync.py#L271

timeout is badly calculated

### Versions
* Pymodbus: source < f0edef3071d500b

dhoomakethu, 2019-01-04:
Thankyou @MarekLew I will take a look.

MarekLew, 2019-01-06:
this is a standard error (inicialize `begin` variable over loop)

move line 254 to while loop
before:
```Python
begin = time.time()
while recive_size > 0:
```
after:
```Python
while recive_size > 0:
    begin = time.time()
```
is fix this.

MarekLew, 2019-01-12:
Problem is in ModbusTcpClient not in ModbusSerialClient. Timout in ModbusSerialClient is calculate correctly(I don't see any problem). I not have scenario when transmission for TCP have 2 or more time breaks for next package, when `self.socket.recv(recv_size)` not receive all data. 

After a year, I wanted to update the library to a newer version and this was the first function I reviewed. I saw and wrote identical mistakes so I write about it. This is not a good way to calculate the remaining time.

dhoomakethu, 2019-01-14:
Oops, sorry I overlooked it. Anyhow I ran the same program with `ModbusTcpClient` and could not reproduce the issue (as per your comments on the commit ). Please share any logs you have for further analysis.

```
2019-01-14 08:28:47,564 MainThread      DEBUG    scratch_114    :26       Timeout - 1 seconds
2019-01-14 08:28:47,582 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2019-01-14 08:28:47,583 MainThread      DEBUG    transaction    :115      Running transaction 1
2019-01-14 08:28:47,583 MainThread      DEBUG    transaction    :206      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
2019-01-14 08:28:47,583 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2019-01-14 08:28:47,583 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-01-14 08:28:48,587 MainThread      DEBUG    transaction    :218      Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
2019-01-14 08:28:48,587 MainThread      DEBUG    socket_framer  :147      Processing: 
2019-01-14 08:28:48,588 MainThread      DEBUG    transaction    :374      Getting transaction 1
2019-01-14 08:28:48,588 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2019-01-14 08:28:48,588 MainThread      DEBUG    scratch_114    :30       Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
2019-01-14 08:28:48,590 MainThread      DEBUG    scratch_114    :31       1.0238959789276123 second
2019-01-14 08:28:48,590 MainThread      DEBUG    scratch_114    :33       
2019-01-14 08:28:48,590 MainThread      DEBUG    scratch_114    :26       Timeout - 2 seconds
2019-01-14 08:28:48,591 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2019-01-14 08:28:48,591 MainThread      DEBUG    transaction    :115      Running transaction 1
2019-01-14 08:28:48,592 MainThread      DEBUG    transaction    :206      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
2019-01-14 08:28:48,592 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2019-01-14 08:28:48,592 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-01-14 08:28:50,598 MainThread      DEBUG    transaction    :218      Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
2019-01-14 08:28:50,598 MainThread      DEBUG    socket_framer  :147      Processing: 
2019-01-14 08:28:50,598 MainThread      DEBUG    transaction    :374      Getting transaction 1
2019-01-14 08:28:50,598 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2019-01-14 08:28:50,599 MainThread      DEBUG    scratch_114    :30       Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
2019-01-14 08:28:50,599 MainThread      DEBUG    scratch_114    :31       2.0080931186676025 second
2019-01-14 08:28:50,599 MainThread      DEBUG    scratch_114    :33       
2019-01-14 08:28:50,599 MainThread      DEBUG    scratch_114    :26       Timeout - 3 seconds
2019-01-14 08:28:50,600 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2019-01-14 08:28:50,600 MainThread      DEBUG    transaction    :115      Running transaction 1
2019-01-14 08:28:50,600 MainThread      DEBUG    transaction    :206      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
2019-01-14 08:28:50,600 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2019-01-14 08:28:50,601 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-01-14 08:28:53,605 MainThread      DEBUG    transaction    :218      Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
2019-01-14 08:28:53,605 MainThread      DEBUG    socket_framer  :147      Processing: 
2019-01-14 08:28:53,606 MainThread      DEBUG    transaction    :374      Getting transaction 1
2019-01-14 08:28:53,606 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2019-01-14 08:28:53,606 MainThread      DEBUG    scratch_114    :30       Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
2019-01-14 08:28:53,606 MainThread      DEBUG    scratch_114    :31       3.0067830085754395 second
2019-01-14 08:28:53,606 MainThread      DEBUG    scratch_114    :33       
2019-01-14 08:28:53,606 MainThread      DEBUG    scratch_114    :26       Timeout - 4 seconds
2019-01-14 08:28:53,607 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2019-01-14 08:28:53,607 MainThread      DEBUG    transaction    :115      Running transaction 1
2019-01-14 08:28:53,608 MainThread      DEBUG    transaction    :206      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
2019-01-14 08:28:53,608 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
[... cut ...]
2019-01-14 08:29:42,655 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2019-01-14 08:29:42,656 MainThread      DEBUG    scratch_114    :30       Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
2019-01-14 08:29:42,656 MainThread      DEBUG    scratch_114    :31       10.00805401802063 second
2019-01-14 08:29:42,656 MainThread      DEBUG    scratch_114    :33    
```

MarekLew, 2019-01-14:
I have test for this. Client disconect after `4.02500009537` `timeout=10`
```
$ python serwer.py
DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 60024)
DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:60024]
DEBUG:pymodbus.server.sync:Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:1
DEBUG:pymodbus.datastore.context:getValues[3] 1:1
DEBUG:pymodbus.server.sync:send: 0001000000050103020011
ERROR:pymodbus.server.sync:Socket error occurred [Errno 10053] Nawi╣zane po│╣czenie zosta│o przerwane przez oprogramowanie zainstalowane w komputerze-hoťcie
DEBUG:pymodbus.server.sync:Client Disconnected [127.0.0.1:60024]
```
```
$ python client.py
2019-01-14 10:46:56,611 MainThread      DEBUG    client         :10       Timeout - 10 seconds
2019-01-14 10:46:56,615 MainThread      DEBUG    transaction    :111      Current transaction state - IDLE
2019-01-14 10:46:56,615 MainThread      DEBUG    transaction    :115      Running transaction 1
2019-01-14 10:46:56,615 MainThread      DEBUG    transaction    :206      SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
2019-01-14 10:46:56,617 MainThread      DEBUG    sync           :75       New Transaction state 'SENDING'
2019-01-14 10:46:56,617 MainThread      DEBUG    transaction    :209      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2019-01-14 10:47:00,634 MainThread      DEBUG    transaction    :218      Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (5 received))
2019-01-14 10:47:00,634 MainThread      DEBUG    socket_framer  :147      Processing:
2019-01-14 10:47:00,634 MainThread      DEBUG    transaction    :374      Getting transaction 1
2019-01-14 10:47:00,635 MainThread      DEBUG    transaction    :180      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2019-01-14 10:47:00,635 MainThread      DEBUG    client         :13       Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (5 received)
2019-01-14 10:47:00,637 MainThread      DEBUG    client         :14       4.02500009537 second
2019-01-14 10:47:00,648 MainThread      DEBUG    client         :16
```

[client.py](https://github.com/riptideio/pymodbus/files/2754651/client.py.txt)
[serwer.py](https://github.com/riptideio/pymodbus/files/2754635/serwer.py.txt)

MarekLew, 2019-01-14:
Other test:
[client.py](https://github.com/riptideio/pymodbus/files/2754679/client.py.txt)
[log.txt](https://github.com/riptideio/pymodbus/files/2754680/log.txt)
[serwer.py](https://github.com/riptideio/pymodbus/files/2754681/serwer.py.txt)

dhoomakethu, 2019-01-14:
Thanks @MarekLew . I see what's happening and you are right, the timeout is calculated in-correctly when partial data is received .

MarekLew, 2019-01-16:
Sorry for not rebranching.  Not have t
time in this weak.