pymodbus-dev/pymodbus #371: Pymodbus REPL on python3 fails during auto completion for `result`

dhoomakethu, 2019-01-10:
### Versions

* Python: python 3.6.5
* OS: Mac OSX
* Pymodbus: 2.1.0
* Modbus Hardware (if used): NA

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync
* Client: tcp/rtu/ascii - sync

### Description

Trying to use `Pymodbus repl` to read and decode holding registers, the read goes fine but while trying to decode the data with `result` command, no auto-completion is provided upon typing `result`. A tab after typing `result` produces error traceback.

**NOTE** : Help also does not show commands related to result decode

### Code and Logs

```python
 > result

Unhandled exception in event loop:
  File "/Users/sanjay/.virtualenvs/testpym/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 90, in step_next
    new_f = coroutine.throw(exc)
  File "/Users/sanjay/.virtualenvs/testpym/lib/python3.6/site-packages/prompt_toolkit/buffer.py", line 1637, in new_coroutine
    yield From(coroutine(*a, **kw))
  File "/Users/sanjay/.virtualenvs/testpym/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 90, in step_next
    new_f = coroutine.throw(exc)
  File "/Users/sanjay/.virtualenvs/testpym/lib/python3.6/site-packages/prompt_toolkit/buffer.py", line 1495, in async_completer
    cancel=lambda: not proceed()))
  File "/Users/sanjay/.virtualenvs/testpym/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 86, in step_next
    new_f = coroutine.send(None)
  File "/Users/sanjay/.virtualenvs/testpym/lib/python3.6/site-packages/prompt_toolkit/eventloop/async_generator.py", line 114, in consume_async_generator
    item = iterator.send(send)
  File "/Users/sanjay/.virtualenvs/testpym/lib/python3.6/site-packages/prompt_toolkit/completion/base.py", line 166, in get_completions_async
    for item in self.get_completions(document, complete_event):
  File "/Users/sanjay/.virtualenvs/testpym/lib/python3.6/site-packages/pymodbus/repl/completer.py", line 149, in get_completions
    commands = list(command.get_completion())

Exception 'NoneType' object has no attribute 'get_completion'

```

dhoomakethu, 2019-01-16:
Fixed and available as part of [v2.2.0rc1](https://github.com/riptideio/pymodbus/releases/tag/v2.2.0rc1).
`pip install pymodbus==2.2.0rc1`

landon-humatics, 2019-02-13:
FYI - same type of problem occurred with 'help' command completion and v2.2.0rc1 does not fix the issue.

```bash
$ pymodbus.console tcp

----------------------------------------------------------------------------
__________          _____             .___  __________              .__   
\______   \___.__. /     \   ____   __| _/  \______   \ ____ ______ |  |  
 |     ___<   |  |/  \ /  \ /  _ \ / __ |    |       _// __ \\____ \|  |  
 |    |    \___  /    Y    (  <_> ) /_/ |    |    |   \  ___/|  |_> >  |__
 |____|    / ____\____|__  /\____/\____ | /\ |____|_  /\___  >   __/|____/
           \/            \/            \/ \/        \/     \/|__|
                                        v1.0.0 - [pymodbus, version 2.1.0]         
----------------------------------------------------------------------------

Unhandled exception in event loop:
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 90, in step_next
    new_f = coroutine.throw(exc)
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/buffer.py", line 1648, in new_coroutine
    yield From(coroutine(*a, **kw))
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 90, in step_next
    new_f = coroutine.throw(exc)
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/buffer.py", line 1500, in async_completer
    cancel=lambda: not proceed()))
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 86, in step_next
    new_f = coroutine.send(None)
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/eventloop/async_generator.py", line 114, in consume_async_generator
    item = iterator.send(send)
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/completion/base.py", line 170, in get_completions_async
    for item in self.get_completions(document, complete_event):
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/pymodbus/repl/completer.py", line 149, in get_completions
    commands = list(command.get_completion())

Exception 'NoneType' object has no attribute 'get_completion'
```

Installed 2.2.0rc1....

```bash
$ pip install pymodbus==2.2.0rc1
Collecting pymodbus==2.2.0rc1
  Downloading https://files.pythonhosted.org/packages/19/9d/02c6e9d090feb7353e96174a866d18f1041941f57be3d21123cfa87d1421/pymodbus-2.2.0rc1-py2.py3-none-any.whl (128kB)
    100% |████████████████████████████████| 133kB 3.7MB/s 
Requirement already satisfied: pyserial>=3.4 in ./anaconda3/lib/python3.6/site-packages (from pymodbus==2.2.0rc1) (3.4)
Requirement already satisfied: six==1.11.0 in ./anaconda3/lib/python3.6/site-packages (from pymodbus==2.2.0rc1) (1.11.0)
Installing collected packages: pymodbus
  Found existing installation: pymodbus 2.1.0
    Uninstalling pymodbus-2.1.0:
      Successfully uninstalled pymodbus-2.1.0
Successfully installed pymodbus-2.2.0rc1
(base) rampart:~ lcox$ pymodbus.console tcp

----------------------------------------------------------------------------
__________          _____             .___  __________              .__   
\______   \___.__. /     \   ____   __| _/  \______   \ ____ ______ |  |  
 |     ___<   |  |/  \ /  \ /  _ \ / __ |    |       _// __ \\____ \|  |  
 |    |    \___  /    Y    (  <_> ) /_/ |    |    |   \  ___/|  |_> >  |__
 |____|    / ____\____|__  /\____/\____ | /\ |____|_  /\___  >   __/|____/
           \/            \/            \/ \/        \/     \/|__|
                                        v1.1.0 - [pymodbus, version 2.2.0.rc1]         
----------------------------------------------------------------------------

Unhandled exception in event loop:
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 90, in step_next
    new_f = coroutine.throw(exc)
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/buffer.py", line 1648, in new_coroutine
    yield From(coroutine(*a, **kw))
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 90, in step_next
    new_f = coroutine.throw(exc)
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/buffer.py", line 1500, in async_completer
    cancel=lambda: not proceed()))
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/eventloop/coroutine.py", line 86, in step_next
    new_f = coroutine.send(None)
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/eventloop/async_generator.py", line 114, in consume_async_generator
    item = iterator.send(send)
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/prompt_toolkit/completion/base.py", line 170, in get_completions_async
    for item in self.get_completions(document, complete_event):
  File "/Users/lcox/anaconda3/lib/python3.6/site-packages/pymodbus/repl/completer.py", line 149, in get_completions
    commands = list(command.get_completion())

Exception 'NoneType' object has no attribute 'get_completion'

```
```bash
$ python --version
Python 3.6.8 :: Anaconda, Inc.
```

dhoomakethu, 2019-02-16:
@landon-humatics thanks for reporting. Will take a look

dhoomakethu, 2019-02-16:
@landon-humatics please check v2.2.0rc3 and see if the issue is fixed

commit 826240b5 (2019-01-14): Fix #371 pymodbus repl on python3
commit 9ff42d16 (2019-04-18): Pymodbus 2.2.0 (#375)

* #357 Support registration of custom requests

* #368 Fixes write to broadcast address

When writing to broadcast address (unit_id=0) there should be no response according to the Modbus spec. This fix changes expected_response_length to 0 when writing to unit_id=0. This will break any existing code that is improperly using unit_id 0 for a slave address.

* Bump version to 2.2.0

Fix #366 Update failures in sql context

Update Changelog

Fix major minor version in example codes

* Fix #371 pymodbus repl on python3

* 1. Fix tornado async serial client `TypeError` while p [... cut]
commit e6da559e (2019-09-09): asyncio server implementation (#400)

* #357 Support registration of custom requests

* #368 Fixes write to broadcast address

When writing to broadcast address (unit_id=0) there should be no response according to the Modbus spec. This fix changes expected_response_length to 0 when writing to unit_id=0. This will break any existing code that is improperly using unit_id 0 for a slave address.

* Bump version to 2.2.0

Fix #366 Update failures in sql context

Update Changelog

Fix major minor version in example codes

* Fix #371 pymodbus repl on python3

* 1. Fix tornado async serial client `Typ [... cut]
closed 2019-04-18 by dhoomakethu