pymodbus-dev/pymodbus #373: AttributeError: 'NoneType' object has no attribute 'interCharTimeout'

neerajkumarjarvis, 2019-01-12:
https://github.com/riptideio/pymodbus/blob/f0edef3071d500b7e3a987ee196c38bb79f59518/pymodbus/client/sync.py#L476

Following error is coming when i am trying to make a serial modbus connection through ```rtu```
Please put a check None check here 

```
self.socket.interCharTimeout = self.inter_char_timeout
AttributeError: 'NoneType' object has no attribute 'interCharTimeout'
```

dhoomakethu, 2019-01-12:
what version of pyserial are you using ? Please use the template to report the issue (OS , Pymodbus etc)
Also, please refer pyserial documentation [here](https://pythonhosted.org/pyserial/pyserial_api.html#serial.Serial.interCharTimeout) for interCharTimeout

neerajkumarjarvis, 2019-01-12:
Name: pymodbus
Version: 2.1.0
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/riptideio/pymodbus/
Author: Galen Collins
Author-email: bashwork@gmail.com
License: BSD
Location: /home/user/.local/lib/python3.6/site-packages
Requires: six, pyserial

OS: 
Raspberry Pi 
Linux (uname -a) -> 4.15.0-43-generic #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

dhoomakethu, 2019-01-12:
Could you please paste the output of `pip list`.

neerajkumarjarvis, 2019-01-12:
alembic (1.0.5)
apturl (0.5.2)
asn1crypto (0.24.0)
Babel (2.4.0)
blinker (1.4)
Brlapi (0.6.6)
certifi (2018.11.29)
chardet (3.0.4)
Click (7.0)
colorama (0.3.7)
command-not-found (0.3)
cryptography (2.1.4)
cupshelpers (1.0)
defer (1.0.6)
distro-info (0.18)
Flask (1.0.2)
Flask-BabelEx (0.9.3)
Flask-Gravatar (0.4.2)
Flask-HTMLmin (1.3.2)
Flask-Login (0.4.0)
Flask-Mail (0.9.1)
Flask-Migrate (2.3.1)
Flask-Paranoid (0.2.0)
Flask-Principal (0.4.0)
Flask-Security (1.7.5)
Flask-SQLAlchemy (2.3.2)
Flask-WTF (0.14.2)
htmlmin (0.1.12)
httplib2 (0.11.3)
idna (2.8)
itsdangerous (1.1.0)
Jinja2 (2.10)
keyring (10.6.0)
keyrings.alt (3.0)
language-selector (0.1)
launchpadlib (1.10.6)
lazr.restfulclient (0.13.5)
lazr.uri (1.0.3)
louis (3.5.0)
macaroonbakery (1.1.3)
Mako (1.0.7)
MarkupSafe (1.1.0)
MinimalModbus (0.7)
mysqlclient (1.3.14)
netifaces (0.10.4)
oauth (1.0.1)
olefile (0.45.1)
paramiko (2.0.0)
passlib (1.7.1)
pexpect (4.2.1)
Pillow (5.1.0)
pip (9.0.1)
protobuf (3.0.0)
psutil (5.4.2)
psycopg2 (2.7.5)
pyasn1 (0.4.2)
pycairo (1.16.2)
pycrypto (2.6.1)
pycups (1.9.73)
pygobject (3.26.1)
pyinotify (0.9.6)
pymacaroons (0.13.0)
pymodbus (2.1.0)
PyMySQL (0.8.0)
PyNaCl (1.1.2)
pyOpenSSL (17.5.0)
pyRFC3339 (1.0)
pyserial (3.4)
python-apt (1.6.3)
python-dateutil (2.7.5)
python-debian (0.1.32)
python-editor (1.0.3)
pytz (2018.3)
pyxdg (0.25)
PyYAML (3.12)
reportlab (3.4.0)
requests (2.21.0)
requests-unixsocket (0.1.5)
SecretStorage (2.3.1)
setuptools (39.0.1)
simplejson (3.13.2)
six (1.11.0)
speedtest-cli (2.0.0)
SQLAlchemy (1.2.16)
sqlparse (0.2.4)
sshtunnel (0.1.4)
system-service (0.3)
systemd-python (234)
ubuntu-drivers-common (0.0.0)
ufw (0.35)
unattended-upgrades (0.1)
urllib3 (1.24.1)
usb-creator (0.3.3)
wadllib (1.3.2)
Werkzeug (0.14.1)
wheel (0.30.0)
WTForms (2.1)
xkit (0.0.0)
zope.interface (4.3.2)

dhoomakethu, 2019-01-12:
I see you are using `pyserial==3.4` . Which definitely has `interCharTimeout`. 

```
Python 2.7.14 (default, Jan  6 2018, 12:16:16)
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import serial
>>> serial.VERSION
'3.4'
>>> s = serial.Serial
>>> dir(s)
['BAUDRATES', 'BAUDRATE_CONSTANTS', 'BYTESIZES', 'PARITIES', 'STOPBITS', '_SAVED_SETTINGS', '__abstractmethods__', '__class__', '__delattr__', '__dict__', '__doc__', '__enter__', '__exit__', '__format__', '__getattribute__', '__hash__', '__init__', '__iter__', '__metaclass__', '__module__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_abc_cache', '_abc_negative_cache', '_abc_negative_cache_version', '_abc_registry', '_checkClosed', '_checkReadable', '_checkSeekable', '_checkWritable', '_reconfigure_port', '_set_rs485_mode', '_set_special_baudrate', '_update_break_state', '_update_dtr_state', '_update_rts_state', 'applySettingsDict', 'apply_settings', 'baudrate', 'break_condition', 'bytesize', 'cancel_read', 'cancel_write', 'cd', 'close', 'closed', 'cts', 'dsr', 'dsrdtr', 'dtr', 'exclusive', 'fileno', 'flush', 'flushInput', 'flushOutput', 'getCD', 'getCTS', 'getDSR', 'getRI', 'getSettingsDict', 'get_settings', 'inWaiting', 'in_waiting', 'interCharTimeout', 'inter_byte_timeout', 'iread_until', 'isOpen', 'isatty', 'next', 'nonblocking', 'open', 'osx_version', 'out_waiting', 'parity', 'port', 'read', 'read_all', 'read_until', 'readable', 'readall', 'readinto', 'readline', 'readlines', 'reset_input_buffer', 'reset_output_buffer', 'ri', 'rs485_mode', 'rts', 'rtscts', 'seek', 'seekable', 'sendBreak', 'send_break', 'setDTR', 'setPort', 'setRTS', 'set_input_flow_control', 'set_output_flow_control', 'stopbits', 'tell', 'timeout', 'truncate', 'writable', 'write', 'writeTimeout', 'write_timeout', 'writelines', 'xonxoff']
>>>
(pymodbus) $ workon pymodbus-py3
(pymodbus-py3) $ python
Python 3.5.2 (default, Oct 11 2016, 05:05:28)
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.38)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import serial
>>> serial.VERSION
'3.4'
>>> dir(serial.Serial)
['BAUDRATES', 'BAUDRATE_CONSTANTS', 'BYTESIZES', 'PARITIES', 'STOPBITS', '_SAVED_SETTINGS', '__abstractmethods__', '__class__', '__del__', '__delattr__', '__dict__', '__dir__', '__doc__', '__enter__', '__eq__', '__exit__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__iter__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__next__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_abc_cache', '_abc_negative_cache', '_abc_negative_cache_version', '_abc_registry', '_checkClosed', '_checkReadable', '_checkSeekable', '_checkWritable', '_reconfigure_port', '_set_rs485_mode', '_set_special_baudrate', '_update_break_state', '_update_dtr_state', '_update_rts_state', 'applySettingsDict', 'apply_settings', 'baudrate', 'break_condition', 'bytesize', 'cancel_read', 'cancel_write', 'cd', 'close', 'closed', 'cts', 'dsr', 'dsrdtr', 'dtr', 'exclusive', 'fileno', 'flush', 'flushInput', 'flushOutput', 'getCD', 'getCTS', 'getDSR', 'getRI', 'getSettingsDict', 'get_settings', 'inWaiting', 'in_waiting', 'interCharTimeout', 'inter_byte_timeout', 'iread_until', 'isOpen', 'isatty', 'nonblocking', 'open', 'osx_version', 'out_waiting', 'parity', 'port', 'read', 'read_all', 'read_until', 'readable', 'readall', 'readinto', 'readline', 'readlines', 'reset_input_buffer', 'reset_output_buffer', 'ri', 'rs485_mode', 'rts', 'rtscts', 'seek', 'seekable', 'sendBreak', 'send_break', 'setDTR', 'setPort', 'setRTS', 'set_input_flow_control', 'set_output_flow_control', 'stopbits', 'tell', 'timeout', 'truncate', 'writable', 'write', 'writeTimeout', 'write_timeout', 'writelines', 'xonxoff']
>>>
```

neerajkumarjarvis, 2019-01-12:
Yes it has attributes
>>> serial.Serial.interCharTimeout

only ```none``` check need before calling this attribute in case ```Serial``` object is not initialised 
```self.socket.interCharTimeout```

dhoomakethu, 2019-01-13:
A PR already is in place #354  for the same.

programatt, 2019-01-24:
@dhoomakethu I've installed 2.2.0rc1 and getting the same error as the OP.

    /Users/mathewphillips/code/venvs/TestModbusTk/bin/python         
    /Users/mathewphillips/code/TestModbusTk/client.py
    Traceback (most recent call last):
      File "/Users/mathewphillips/code/TestModbusTk/client.py", line 30, in <module>
        main()
      File "/Users/mathewphillips/code/TestModbusTk/client.py", line 26, in main
        client.connect()
      File "/Users/mathewphillips/code/venvs/TestModbusTk/lib/python3.7/site-packages/pymodbus/client/sync.py", line 489, in connect
        self.socket.interCharTimeout = self.inter_char_timeout
    AttributeError: 'NoneType' object has no attribute 'interCharTimeout'

`pip list`

    Package    Version 
    ---------- --------
    modbus-tk  0.5.9   
    pip        19.0.1  
    pymodbus   2.2.0rc1
    pyserial   3.4     
    setuptools 39.1.0  
    six        1.11.0 

and my client code is

    from pymodbus.client.sync import ModbusSerialClient
    PORT = '/dev/ttys1'
    def main():
        client = ModbusSerialClient(method='rtu', port=PORT, timeout=1)
        client.connect()
        print(client.read_holding_registers(1,10))
    
    if __name__ == "__main__":
        main()

dhoomakethu, 2019-01-25:
As a temp workaround , you can create your client s with kwarg `strict=False`>.
Client = ModdbusSerialClient(....,strict=False)`

bhtakti-aatma, 2019-01-31:
@programatt did you got solution. I'm struggling with same. So please help me in communicating with RS-485. I'm able to get value in modscan32 but not in Rpi.

dhoomakethu, 2019-01-31:
@bhtakti-aatma Please refer #377 for a fix.

bhtakti-aatma, 2019-01-31:
@dhoomakethu thanks for quick response. Now I'm getting result:
False
Connection lost, Try again
but when I'm operating with another code it's giving my connection as True.
please help me to get communicated with RS-485.

dhoomakethu, 2019-01-31:
Could you be more specific with what you mean by `another code` ? Please share your code which is working and also the one which doesn't work. Also Is both code running against the same modbus slave using same settings or different one. Please share any other info which might be useful (OS etc)

bhtakti-aatma, 2019-01-31:
@dhoomakethu here's my python script:

from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.client.sync import ModbusTcpClient
from pymodbus.client.sync import ModbusSerialClient #as ModbusClient

def validator(instance):
    if not instance.isError():
        '''.isError() implemented in pymodbus 1.4.0 and above.'''
        decoder = BinaryPayloadDecoder.fromRegisters(
            instance.registers,
            byteorder=Endian.Big, wordorder=Endian.Little
        )   
        return float(decoder.decode_32bit_float())

    else:
        # Error handling.
        print("There aren't the registers, Try again.")
        return None

client=ModbusSerialClient(method='rtu',port="/dev/ttyUSB1",
    baudrate=9600,timeout=2,parity='E',stopbits=1,strict=False)
#client = ModbusClient(method='rtu', port="dev/ttyUSB1", baudrate=9600, timeout=2, 
            #parity='E',stopbits=1,strict=False)
connection = client.connect()
print(connection)

if connection:
    request = client.read_holding_registers(3911, count=2, unit=1)
    data = validator(request)
    print(data)

    client.close()

else:
    print('Connection lost, Try again')

True
There aren't the registers, Try again.
None

bhtakti-aatma, 2019-01-31:
@dhoomakethu now it's coming : ========================== RESTART: /home/pi/rs3.py ==========================
>>> 
True
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0xf 0x57 0x0 0x2 0x76 0xcf
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x83 0x2 0xc0 0xf1
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x83 0x2
DEBUG:pymodbus.factory:Factory Response[131]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
There aren't the registers, Try again.
None
>>>

dhoomakethu, 2019-01-31:
You are passing the wrong data address (`DEBUG:pymodbus.transaction:RECV: 0x1 0x83 0x2 0xc0 0xf1`) . You will have to pass the exact offset to pymodbus for it to work.

bhtakti-aatma, 2019-01-31:
@dhoomakethu Your help is too much appreciable, I'm making progress step by step. Now it's coming like this:
True
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x4 0xf 0x57 0x0 0x2 0xc3 0xf
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x84 0x1 0x82 0xc0
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x84 0x1
DEBUG:pymodbus.factory:Factory Response[132]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Exception Response(132, 4, IllegalFunction)
Traceback (most recent call last):
  File "/home/pi/rs3.py", line 47, in <module>
    data =validator(result.registers)
AttributeError: 'ExceptionResponse' object has no attribute 'registers'

py2pi, 2019-01-31:
@dhoomakethu  if you need some more information for this, then please let me know ? I'm still  unable solve it.

py2pi, 2019-01-31:
@dhoomakethu thanks for letting me solve this at my own after guiding me.  You are a good instructor. Keep doing your good work.

willcharlton, 2019-02-03:
@py2pi can you summarize how this issue was resolved for you? I'm having a similar issue. Was it as simple as `strict=True` in client constructor kwargs?

py2pi, 2019-02-07:
@willcharlton Sorry for late reply. It can work on strict=False. You just have to read the correct registers.

py2pi, 2019-02-07:
@willcharlton I used this code to read the holding registers. Hope it'll help out you. 

import pymodbus
import serial
from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient #as ModbusClient #initialize a serial RTU client instance
from pymodbus.transaction import ModbusRtuFramer
import logging
import time
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def conn():
    client1 = ModbusSerialClient(method="rtu",
                    port="/dev/ttyUSB0", baudrate=9600)
    connection1 = client1.connect()
    print(connection1)

def read():
    client1 = ModbusSerialClient(method = "rtu", port="/dev/ttyUSB0",
                                 stopbits = 1, bytesize = 8, parity = 'E',
                                 baudrate= 9600,timeout=2,strict=False)
    connection1 = client1.connect()
    print(connection1)
    registers_set = 1

    address =3940

    count = registers_set * 2

    unit = 1

    result = client1.read_holding_registers(address, count, unit = unit)
    #time.sleep(5)
    print(result)
    print(result.registers)

    """for i in range(registers_set):
            j = i * 2
            k = (i + 1) * 2
            print (j, k, "register range")
            tmp=numpy.array(result.registers[j:k], numpy.int16)
            tmp.dtype = numpy.float32
            #time.sleep(1)
            print (tmp)"""

read()