pymodbus-dev/pymodbus #377: ModbusSerialClient throws AttributeError instead of returning False

stegm, 2019-01-18:
### Versions

* Python: 3.6
* OS: Ubuntu
* Pymodbus: 2.1
* Modbus Hardware (if used): None

### Pymodbus Specific
* Server: None
* Client: rtu - sync

### Description

If the `ModbusSerialClient` is created with method 'rtu' and a non-existing device, the `connect`-method throws an attribute error instead of returning `False`.

Example:

```python
>>> client = ModbusSerialClient(port='notexist', method='rtu')
>>> client.connect()
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "lib/python3.6/site-packages/pymodbus/client/sync.py", line 476, in connect
    self.socket.interCharTimeout = self.inter_char_timeout
AttributeError: 'NoneType' object has no attribute 'interCharTimeout'
```

### Code and Logs

This is because of this code (`ModbusSerialClient.connect()`):
```python
            self.socket = serial.Serial(port=self.port, ...)
        except serial.SerialException as msg:
            _logger.error(msg)
            self.close()
        if self.method == "rtu":
            self.socket.interCharTimeout = self.inter_char_timeout
            self.last_frame_end = None
```
If the serial constructor fails, the exception is catched and close is called, which sets the `socket` field to `None`. Afterwards if the code access the `socket` field, an `AttributeError` is thrown.

dhoomakethu, 2019-01-19:
There is sa fix available with v2.2.0rc1, please do `pip install pymodbus==2.2.0rc1` and update If that still does not solve the problem

stegm, 2019-01-21:
2.2.0rc1 gives the same error for me.

```python
>>> import pymodbus
>>> pymodbus.__version__
'2.2.0.rc1'
>>> from pymodbus.client.sync import ModbusSerialClient
>>> client = ModbusSerialClient(port='notexist', method='rtu')
>>> client.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\temp\pvenv\lib\site-packages\pymodbus\client\sync.py", line 489, in connect
    self.socket.interCharTimeout = self.inter_char_timeout
AttributeError: 'NoneType' object has no attribute 'interCharTimeout'
```

dhoomakethu, 2019-01-21:
Thanks for the confirmation. I will look in to this.

dhoomakethu, 2019-01-26:
Please give a try with the latest on [pymodbus-2.2.0](https://github.com/riptideio/pymodbus/tree/pymodbus-2.2.0) branch and feedback. Thanks

commit 6960d9c0 (2019-01-26): Fix #377 when invalid port is supplied and minor updates in logging
commit 9ff42d16 (2019-04-18): Pymodbus 2.2.0 (#375)

* #357 Support registration of custom requests

* #368 Fixes write to broadcast address

When writing to broadcast address (unit_id=0) there should be no response according to the Modbus spec. This fix changes expected_response_length to 0 when writing to unit_id=0. This will break any existing code that is improperly using unit_id 0 for a slave address.

* Bump version to 2.2.0

Fix #366 Update failures in sql context

Update Changelog

Fix major minor version in example codes

* Fix #371 pymodbus repl on python3

* 1. Fix tornado async serial client `TypeError` while p [... cut]
commit e6da559e (2019-09-09): asyncio server implementation (#400)

* #357 Support registration of custom requests

* #368 Fixes write to broadcast address

When writing to broadcast address (unit_id=0) there should be no response according to the Modbus spec. This fix changes expected_response_length to 0 when writing to unit_id=0. This will break any existing code that is improperly using unit_id 0 for a slave address.

* Bump version to 2.2.0

Fix #366 Update failures in sql context

Update Changelog

Fix major minor version in example codes

* Fix #371 pymodbus repl on python3

* 1. Fix tornado async serial client `Typ [... cut]