pymodbus-dev/pymodbus #379: Unable to stop an async ModbusTcpServer and restart it by a new context

agn-7, 2019-01-23:
### Versions

* Python: 3.6.5
* OS: Ubuntu-16.04
* Pymodbus: 2.1.0

### Pymodbus Specific
* Server: tcp - async

### Description

I want to stop a pymodbus async ModbusTcpServer then start a new server, so I tried with the following simplified snippet code, but I got an exception with `twisted.internet.error.ReactorNotRestartable` content.

### Code and Logs

```python
from pymodbus.server.async import StartTcpServer, StopServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

import threading
import logging

FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def stop():
    StopServer()

def run_async_server():
    store = ModbusSlaveContext(hr=ModbusSequentialDataBlock(0, [17] * 100))
    slaves = {
        0x01: store,
        0x02: store,
        0x03: store,
    }
    context = ModbusServerContext(slaves=slaves, single=False)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/bashwork/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '1.5'

    # TCP Server
    # StartTcpServer(context, identity=identity, address=("localhost", 5020))

    # TCP Server with deferred reactor run
    from twisted.internet import reactor
    StartTcpServer(context, identity=identity, address=("localhost", 5020),
                   defer_reactor_run=True)
    print('Start an async server.')
    t = threading.Timer(5, stop)
    t.daemon = True
    t.start()
    reactor.run()
    print('Server was stopped.')
    run_async_server()  # Recursive.

if __name__ == "__main__":
    run_async_server()

```
Out:

```
2019-01-23 11:35:51,067 MainThread      INFO     async          :254      Starting Modbus TCP Server on localhost:5020
Start an async server.
2019-01-23 11:35:56,069 Thread-1        DEBUG    async          :222      Running in spawned thread
2019-01-23 11:35:56,070 Thread-1        DEBUG    async          :332      Stopping Server from another thread
Server was stopped.
2019-01-23 11:35:56,077 MainThread      INFO     async          :254      Starting Modbus TCP Server on localhost:5020
Start an async server.
Traceback (most recent call last):
  File "async_server.py", line 87, in <module>
    run_async_server()
  File "async_server.py", line 79, in run_async_server
    run_async_server()  # Recursive.
  File "async_server.py", line 77, in run_async_server
    reactor.run()
  File "/usr/local/lib/python3.6/dist-packages/twisted/internet/base.py", line 1260, in run
    self.startRunning(installSignalHandlers=installSignalHandlers)
  File "/usr/local/lib/python3.6/dist-packages/twisted/internet/base.py", line 1240, in startRunning
    ReactorBase.startRunning(self)
  File "/usr/local/lib/python3.6/dist-packages/twisted/internet/base.py", line 748, in startRunning
    raise error.ReactorNotRestartable()
twisted.internet.error.ReactorNotRestartable

```

farzadpanahi, 2021-12-03:
Having the same problem. Any update on this issue?

janiversen, 2022-05-31:
This works if you test with asyncio, and we no longer support twisted, therefore closing issue.

closed 2022-05-31 by janiversen