pymodbus-dev/pymodbus #382: USRIOT TCP232-304 reading problem

snaptec, 2019-01-28:
### Versions

* Python: Python 2.7.13
* OS: Raspbian GNU/Linux 9 \n \l
* Pymodbus: pymodbus-2.1.0
* Modbus Hardware (if used): USRIOT TCP232-304

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
Server on Port 8899, 9600 / 8n1

### Description

I would like to read a Modbus Device connected to a USR TCP232-304.

What I would like to do:
Using the ModbusTCPClient to connect, read a register and disconnect.
The USR Device shows some received packets, but doesn't work as expected.

When I do the following:
```
sudo socat pty,link=/dev/virtualcom12,raw tcp:192.168.0.7:8899
```
and then use 
```
from pymodbus.client.sync import ModbusSerialClient
client = ModbusSerialClient(method = "rtu", port=/dev/virtualcom12, baudrate=9600,
                        stopbits=1, bytesize=8, timeout=1)
```
The reading of the device is working. Cables and devices are up and working

What I would like to do is:
```
from pymodbus.client.sync import ModbusTcpClient
client = ModbusTcpClient('192.168.0.7', method='rtu', port=8899, baudrate=9600, stopbits=1, bytesize=1)
```
But this fails.

I got another Modbus/Ethernet Konverter (Elfin EE11) where both variants work.
But I would like to use the USR one, but without initiate a socat before and link the USR to a local port.

Any hints what I'm doing wrong here?

snaptec, 2019-02-01:
Thanks @dhoomakethu 
More or less, that's it.
It is working, but about 2 out of 10 read outs returns:
AttributeError: 'ModbusIOException' object has no attribute 'registers'

I try to figure out why it is not working stable.

Grcutter, 2020-02-24:
@snaptec 
Did you eventually get this working without the occasional errors?
I would like to use the USR-TCP232-304 to communicate with an EPSOLAR Tracer 3210A  RS485/Modbus device.  Would this be possible?

closed 2021-03-08 by github-actions[bot]