pymodbus-dev/pymodbus #384: "Six" library should not be pinned to an exact version

sputt, 2019-02-02:
requirements.txt calls out six==1.11.0. This is over-constrained (and tests pass with 1.12.0)

dhoomakethu, 2019-02-04:
Is there any particular problem you have faced with `six==1.11.0` ? We try to keep things which are proved to be working fine and hence the requirements are not updated frequently. If you think having `1.12.0` against `1.11.0` is useful. please let us know.

sputt, 2019-02-05:
It's good practice to keep the requirements of a python library as unconstrained as possible.  For example, pymodbus may be used in a project that pulls in hundreds of libraries, and pymodbus alone may be the only reason they are unable to take a new version of six in their environment.

We can use CI to verify that new versions of reqs don't break this package, rather than relying on hard pins.  I'd be happy to help if you need it.  Luckily pymodbus doesn't have many third party requirements.

arunpersaud, 2019-02-21:
Just run into the same issue. My openSUSE distribution comes with 1.12.0 and I can't pip install pymodbus on it :( If I try to install it with --no-dependencies, then the install works, but when importing a module that depends on pymodbus, I still get errors:

ContextualVersionConflict: (six 1.12.0 (/usr/lib/python3.7/site-packages), Requirement.parse('six==1.11.0'), {'pymodbus'})

my current solution is to edit /usr/lib/python3.7/site-packages/pymodbus-2.1.0.dist-info/* files by hand.

tracernz, 2019-02-21:
Pinning so tightly means you can basically only use PyModbus in a virtualenv.

dhoomakethu, 2019-02-22:
If it helps, you can also use `pip install pymodbus` with `--user` option to install pymodbus to the Python user install directory for your platform.

jonemo, 2019-04-29:
Just ran into the same problem, trying to use `pymodbus` together with [`opcua`](https://pypi.org/project/opcua/) in a project and `pip install` shows this error in red (but still installs pymodbus):

```
pymodbus 2.2.0 has requirement six==1.11.0, but you'll have six 1.12.0 which is incompatible.
```

However, using any of the entrypoints of my project fails with this setuptools error message:

```
$ myproject_entrypoint
Traceback (most recent call last):
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 581, in _build_master
    ws.require(__requires__)
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 898, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 789, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (six 1.12.0 (/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages), Requirement.parse('six==1.11.0'), {'pymodbus'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/jonas/repos/myproject/.venv/bin/plc_server", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3126, in <module>
    @_call_aside
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3110, in _call_aside
    f(*args, **kwargs)
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3139, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 596, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/Users/jonas/repos/myproject/.venv/lib/python3.6/site-packages/pkg_resources/__init__.py", line 784, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'six==1.11.0' distribution was not found and is required by pymodbus
```

I think it is incorrect to label this issue as "not an issue". At least for my (pretty simple and conventional) use case this is a show stopper.

dhoomakethu, 2019-04-30:
Thanks for the update, I will update the requirement to `six >= 1.1.0` in coming versions. Meantime, you can do a `pip install six==1.1.0` to fix your issue.

mpf82, 2019-07-03:
Any updates on this? It's still ``six==1.11.0`` in the requirements.txt in all branches.

dhoomakethu, 2019-07-03:
It will be part of next release `2.3.0`.

fabaff, 2019-08-17:
Is there an ETA for 2.3.0?

dhoomakethu, 2019-08-19:
@fabaff Not sure, I am hopeful to clear some PR's this month and there should be a release next month but my hands are full and I can not commit a date. Thanks

dhoomakethu, 2019-10-07:
This should be fixed as part of [v2.3.0rc1](https://pypi.org/project/pymodbus/2.3.0rc1/) , please give it a spin and update.

closed 2019-10-07 by dhoomakethu