pymodbus-dev/pymodbus #388: ImportError: cannot import name 'BinaryModbusDecoder'

msoto42, 2019-02-23:
Hi, I've been testing pymodbus and I was trying to try the "modbus_mapper" example but when I run it I get the following error:
"ImportError: can not import name 'BinaryModbusDecoder'", when reviewing the import I see that in the file payload.py this class does not exist. How can i fix this?

msoto42, 2019-02-23:
CODE....

#from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance
#from pymodbus.transaction import ModbusRtuFramer
from modbus_mapper import csv_mapping_parser
from modbus_mapper import mapping_decoder
from pymodbus.payload import BinaryModbusDecoder

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)
template = ['address', 'type', 'size', 'name', 'function']
raw_mapping = csv_mapping_parser('input.csv',template)
mapping = mapping_decoder(raw_mapping)
client= ModbusClient(method = "rtu", port="/dev/ttyUSB0",timeout=1000, stopbits = 1, bytesize = 8, parity = 'N',baudrate= 19200)   
connection = client.connect()
index, size = 1, 100
result= client.read_holding_registers(index, size)
decoder = BinaryModbusDecoder.fromRegisters(result.registers)

dhoomakethu, 2019-02-25:
You will have to use `from pymodbus.payload import BinaryPayloadDecoder`. Sorry some examples are not up to date. More example on usage is [here](https://github.com/riptideio/pymodbus/blob/master/examples/common/modbus_payload.py)

closed 2021-03-08 by github-actions[bot]