pymodbus-dev/pymodbus #389: Parity issues

geoffbrucepayne, 2019-02-24:
### Versions

* Python: 2.7.13
* OS: Raspbian
* Pymodbus: 2.1.0
* Modbus Hardware (if used): various

### Pymodbus Specific
* Server: rtu - async

### Description

When I start my serial server, it seems to run fine when using 'N' parity, but when configured to run 'E' parity it seems to be operating as 'N'.  I'm currently running a simulator on Raspberry Pi which has my Pymodbus code and has been working fine running 9600-8N1.  However 8N1 is not compliant with the serial guidelines, so I have been trying to run 9600-8E1.  I'm connected to a Windows PC via a USB/serial adapter which goes null modem to another USB/serial on the Pi.  On my PC I'm running Modbus Poll from modbustools.com.  When I connect using 8N1 everything seems to work fine, but when I try to reconfigure the Pymodbus server to even parity, and run even parity on Modbus Poll, I get framing/parity errors on the PC application.  If I flip Modbus Poll to 8N1 it receives the responses ok.

Is there a way to validate the serial settings after the async server has been started?

### Code and Logs

```python
 # code and logs here.
            log.debug("serial settings: parity:{}".format(slave.ser.parity))
            StartSerialServer(context, identity=slave.identity, framer=framer,
                              port=slave.ser.name,
                              baudrate=slave.ser.baudrate,
                              bytesize=slave.ser.bytesize,
                              parity=slave.ser.parity,
                              stopbits=slave.ser.stopbits,
                              defer_reactor_run=False)

2019-02-24T22:54:03.258Z,[DEBUG],(MainThread),modbus_sim.run_async_server:992,serial settings: parity:E
2019-02-24T22:54:03.260Z,[INFO],(MainThread),async.StartSerialServer:310,Starting Modbus Serial Server on /dev/ttyUSB1
2019-02-24T22:54:03.268Z,[DEBUG],(MainThread),async.connectionMade:47,Client Connected [/dev/ttyUSB1]
2019-02-24T22:54:03.270Z,[DEBUG],(MainThread),async._is_main_thread:228,Running in Main thread
2019-02-24T22:54:15.676Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x1
2019-02-24T22:54:15.678Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x4
2019-02-24T22:54:15.680Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x0 0x0
2019-02-24T22:54:15.682Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x0 0x57
2019-02-24T22:54:15.684Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0xb1
2019-02-24T22:54:15.686Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0xf4
2019-02-24T22:54:15.691Z,[DEBUG],(MainThread),async._send:102,send: 0104ae0f01000000000000000000000000000000000000035c0000000000002670000000000000076c000000000000000000000000000000000000000000000000002200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000109a109acae5

```

geoffbrucepayne, 2019-03-22:
I'm concerned I might have a low level timing issue for the slave when I compare 2 different Master devices polling it.  In all cases I am running serial RTU 9600-8N1

Here is a debug log from a Master that seems to communicate reliably:
2019-03-21T23:55:21.526Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x1
2019-03-21T23:55:21.529Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x4 0x0 0x0
2019-03-21T23:55:21.532Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x0 0x1 0x31
2019-03-21T23:55:21.535Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0xca
2019-03-21T23:55:21.538Z,[DEBUG],(MainThread),async._send:102,send: 0104020f017d00

Another (2nd) master that seems to send a garbage byte prior to its initial poll and the slave server seems not to synchronize and therefore does not respond to subsequent polls:
2019-03-21T23:56:52.356Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0xfe
2019-03-21T23:56:52.370Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x1
2019-03-21T23:56:52.373Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x4 0x0
2019-03-21T23:56:52.376Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x0 0x0 0x1
2019-03-21T23:56:52.379Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x31 0xca
2019-03-21T23:56:53.524Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x1
2019-03-21T23:56:53.527Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x4 0x0
2019-03-21T23:56:53.531Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x0 0x0 0x1
2019-03-21T23:56:53.534Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x31 0xca
2019-03-21T23:56:54.677Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x1
2019-03-21T23:56:54.679Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x4 0x0
2019-03-21T23:56:54.682Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x0 0x0 0x1
2019-03-21T23:56:54.684Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x31
2019-03-21T23:56:54.686Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0xca

And the same (2nd) master when I stop then start the slave server in between the normal poll interval it seems to work:
2019-03-22T00:00:19.709Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x1
2019-03-22T00:00:19.710Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x4
2019-03-22T00:00:19.712Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x0 0x0
2019-03-22T00:00:19.713Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x0
2019-03-22T00:00:19.715Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x1
2019-03-22T00:00:19.717Z,[DEBUG],(MainThread),async.dataReceived:64,Data Received: 0x31 0xca
2019-03-22T00:00:19.719Z,[DEBUG],(MainThread),async._send:102,send: 0104020000b930

Can you provide any guidance on what might be happening and some suggestions how I might get server responses to the 2nd Master working reliably?

dhoomakethu, 2019-03-22:
@geoffbrucepayne apologies for the delay, I will try to poke around regarding the wrong parity, It should not be the case. Also there is no way at present to view the serial settings being applied unless you are hacking the code to print the serial object attributes or using a debugger to view the state.  Regarding your second comment, Could you please share the client code you are using to connect the serial server ? Please note the async client do not follow the exact modbus serial standards (3.5 frame sleep between transactions). Pymodbus client as such does not send any garbage value to the server as far as I know.

geoffbrucepayne, 2019-03-22:
@dhoomakethu I'm only using a pymodbus async server implementation as a [simulator](https://github.com/gbrucepayne/modbus_sim), no pymodbus client at this point.  The 2 clients I am using are third party tools/devices:
1st Client that I did most of my initial testing with: [Modbus Poll](https://modbustools.com/modbus_poll.html)
2nd Client I am having problems with is a physical device that claims strict Modbus serial compliance (in fact was the reason for my parity question, this device claims only to support none parity when using 2 stop bits)

To clarify the pymodbus code is not sending any garbage.  My question is really about how to understand the debug messages from the async server.  Does it also have the 3.5 frame sleep?

dhoomakethu, 2019-03-23:
This is a bug, thanks for the detailed explanation. The culprit appears to be this [line](https://github.com/riptideio/pymodbus/blob/master/pymodbus/server/async.py#L317) in the code. Pymodbus is only passing the `Port` and `Baudrate` info to the twisted serial port.

dhoomakethu, 2019-03-23:
@geoffbrucepayne Please give a try with [pymodbus-2.2.0](https://github.com/riptideio/pymodbus/tree/pymodbus-2.2.0) branch and let me know if that fixes the issue. Please note, this branch is compatible with python3.7 and hence the use of `async` is not permitted. You will have to do the import of your server as `from pymodbus.server.asynchronous import StartSerialServer` .

geoffbrucepayne, 2019-03-26:
@dhoomakethu can you clarify the main differences between synchronous and asynchronous?  Does synchronous better conform to the Modbus serial spec of 3.5 frame sleep?

dhoomakethu, 2019-03-26:
The main difference is synchronous server does not process requests in parallel and is sequential . Asynchronous servers can handle multiple requests at once and process . I am not quiet sure Both the servers can conform with modbus serial protocols. But when it comes to client the sync client does conform with the protocol.

AlexandreRio, 2019-07-04:
I'm experiencing the same issue, specifying a parity Even works with a synchronous slave but the parameter seems to be ignored with the async slave.
What workaround do you recommend? I need to update my register from a different thread (as detailed in https://pymodbus.readthedocs.io/en/latest/source/example/updating_server.html)

dhoomakethu, 2019-07-04:
This should be fixed with pymodbus 2.2.0.

closed 2019-07-04 by dhoomakethu