pymodbus-dev/pymodbus #392: Pymodbus RTU sync issue with consecutive readings/writings

makawi24, 2019-03-04:
### Versions

* Python: 3.7.2
* OS: Windows
* Pymodbus: 2.1.0 and 2.2.0 
* Modbus Hardware (if used): Regin

### Pymodbus Specific
* Client: rtu - sync

### Description
When reading or writing just one register, normally the operation always success.
Nevertheless, when there are consecutive readings or writings, many times fails.
I’ve tested in both pymodbus-2.1.0 and pymodbus-2.2.0 without success.
I've attached two examples of the issue.
Thank you
[Pymodbus RTU sync issue.pdf](https://github.com/riptideio/pymodbus/files/2926859/Pymodbus.RTU.sync.issue.pdf)

### Code and Logs
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
import time

client=ModbusClient(method='rtu',port='COM5',baudrate=9600,timeout=1,parity='N')
client.connect()

#reading HOLDING REGISTERS
rr=client.read_holding_registers(0,1,unit=1)
if not rr.isError():
    TsupSP = rr.registers[0]
    print ("Supply Setpoint: " + str(TsupSP/10) + " ºC")
else:
    print ("Supply Setpoint: Nan")

time.sleep(0.1)

#writing COILS
rq = client.write_coil(8, False, unit=1) # El coil correspondiente a la dirección 8 se pone a False
print("Function Code: " + str(rq.function_code))
print("IsError: " + str(rq.isError()))

client.close()

dhoomakethu, 2019-03-06:
@makawi24 Could you please provide the logs ? I am not able to reproduce this issue. Also can you check what happens if you increase the timeout to value > 1 ?

makawi24, 2019-03-07:
Hi dhoomakethu,
1) Please find attached the logging information. 
What I’ve noticed is that, when adding the logs, the success rate is much higher than the error rate. But without them is the opposite…
[Logging Information.pdf](https://github.com/riptideio/pymodbus/files/2941136/Logging.Information.pdf)

2) Increasing the time.sleep doesn’t solve the issue. Sometimes yes, sometimes not. It’s very random.

Always it fails seems to be because an incomplete message received.

Than you and best regards,

dhoomakethu, 2019-03-08:
Do you observe this only with the modbus hardware reported in this issue (Regin) or with other hardwares as well. I am not able to reproduce this issue with the existing modbus hardwares I am having with me. I will try to poke around more and see if I could reproduce and find a fix. Thanks.

transistorgrab, 2019-03-15:
I recently ported a logger application from python 2 to python 3 and in this progress I also got the latest version of pymodbus.
The script runs on a Raspberry Pi with Python 3.5.3 and collects data from a SMA solar inverter in an interval of 5 seconds.
The logger runs for a long time successfully. But after a day or so  I think I see the same error as the OP.
The error message is

> message = BinaryPayloadDecoder.fromRegisters(received.registers, byteorder=Endian.Big, wordorder=Endian.Big)
> AttributeError: 'ModbusIOException' object has no attribute 'registers'

The representative code is from line 792 in [PyModMon](https://github.com/transistorgrab/PyModMonLCD/blob/97f7e30c4d34b103ec2fb715979a66d59c50b441/pymodmon_cglcd_led_3.py#L792-L830)

To work around this error I added the lines from 817…830 (try block) but this only results in an error message from python after a few caught exceptions:
> *** Error in `python3': free(): invalid pointer: 0x00403eb0 ***
The error message is always the same with the same pointer hex value.

I hope this helps to catch the issue.

dhoomakethu, 2019-03-16:
@transistorgrab You will have to check for valid registers before feeding to `BinaryPayloadDecoder`. Perhaps something like
```
if not received.isError():
     message = BinaryPayloadDecoder.fromRegisters(received.registers, byteorder=Endian.Big, wordorder=Endian.Big)
```

transistorgrab, 2019-03-16:
Thanks for the hint.
I will try this and keep you updated.

transistorgrab, 2019-03-16:
Well... I started the script with your addition:
>            try:
                if not received.isError():
                    message = BinaryPayloadDecoder.fromRegisters(received.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                    message_errorcounter = 0
                if received.isError():
                    print (thisdate," Receive error!")

But I just got again the message from my try block "Received data not valid. Error count:1"
So it seems to me that pymodbus does not think there was an error during receiving but cannot decode the received data.
This was after about 3 1/2 hours after starting the script.

transistorgrab, 2019-03-17:
Update:
I added this line to my script in the except block:
`print ("Received is: ", received)`
This gives this output:
> Received is:  Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)

I have no idea why this does not trigger the lines in the try section:
```
try:
    if not received.isError():
        message = BinaryPayloadDecoder.fromRegisters(received.registers, byteorder=Endian.Big, wordorder=Endian.Big)
        message_errorcounter = 0
    if received.isError():
        print (thisdate," Receive error!")
except:
    message_errorcounter += 1
    thisdate = str(datetime.datetime.now()).partition('.')[0]
    thiserrormessage = thisdate + ': Received data not valid. Error count:' + str(message_errorcounter)
    print ("Received is: ", received)
    if (gui_active):
        messagebox.showerror('Data Error',thiserrormessage)
        return  ## prevent further execution of this function
    else:
        print(thiserrormessage)
        return  ## prevent further execution of this function
```

Any idea from your side?
The errors are quite sporadic and occur as single events only every few hours.

dhoomakethu, 2019-03-18:
The error is not raised !!! The read returns an `ModbusIOException` object in case of failure. There is a feature request #298 for the same.

makawi24, 2019-03-18:
Sorry for the delay, not easy to get a different controller.
Performing the tests with the second controller (from a different manufacturer) the results are similar (please find attached “Controller 2 results.pdf”).
Also tested with another USB-RS485 converter with also similar results.
I think I’ll make a condition that, in case of error, reread the value, and that’s all.
Thank you
[Controller 2 results.pdf](https://github.com/riptideio/pymodbus/files/2977903/Controller.2.results.pdf)

dhoomakethu, 2019-03-18:
@makawi24 thanks for the confirmation, I will try to reproduce the issue.

transistorgrab, 2019-04-01:
short update from my side: it seems that checking for .isError seems to work for now for me. :) 
Further crashes of my script seem to stem from some other imported module.

janiversen, 2022-05-25:
This works on 2.5.3 (at least in home assistant).

Feel free to open a new issue if you still have problems.

closed 2022-05-25 by janiversen