pymodbus-dev/pymodbus #393: DeprecationWarning: asyncio.async()

agn-7, 2019-03-06:
### Versions

* Python: 3.6.5
* OS: Ubuntu 16.04
* Pymodbus: 2.1.0
* Modbus Hardware (if used): Delta Gateway

### Pymodbus Specific
* Client: tcp - async

### Description

When I unplug Modbus Delta-Gateway device TCP port, I'm dealing with the following warning:

```
/usr/local/lib/python3.6/dist-packages/pymodbus/client/async/asyncio/__init__.py:319: DeprecationWarning: asyncio.async() function is deprecated, use ensure_future()
  asyncio.async(self._reconnect(), loop=self.loop)
```

dhoomakethu, 2019-03-06:
Please give it a try on 2.2.0rc4 and see if the issue is fixed .

agn-7, 2019-03-08:
Thanks for the response, I will try it by `sudo pip3 install pymodbus==2.2.0rc4`

agn-7, 2019-03-10:
With this update, I'm dealing with the following error:

```
from pymodbus.client.async.tcp import AsyncModbusTCPClient as ModbusClient
ModuleNotFoundError: No module named 'pymodbus.client.async.tcp'; 'pymodbus.client.async' is not a package
```

dhoomakethu, 2019-03-10:
Yes, it is expected. You will have to use `from pymodbus.client.asynchrnous.tcp import AsyncModbusTCPClient as ModbusClient` now. 

Please refer the [changelog](https://github.com/riptideio/pymodbus/releases/tag/v2.2.0rc1).

agn-7, 2019-03-10:
Now, there is not `ModbusTcpClient` Class:

```
from pymodbus.client.asynchronous import ModbusTcpClient as ModbusClient
ImportError: cannot import name 'ModbusTcpClient'
```

dhoomakethu, 2019-03-10:
Because your import is wrong, it should be 
`from pymodbus.client.asynchrnous.tcp import AsyncModbusTCPClient as ModbusClient`

agn-7, 2019-03-10:
- Missing an "o" in `asynchrnous` typing :-D
 - In changelog is `import ModbusTcpClient` also

However, the importing problem solved by `from pymodbus.client.asynchronous.tcp import AsyncModbusTCPClient as ModbusClient`

Then I encountered with `schedulers` import error which I solved that by: 

```
from pymodbus.client.asynchronous import schedulers
```

Apparently, my code is working and there isn't any warning.

Thanks for the response.

dhoomakethu, 2019-03-10:
Thanks wIll update the change log.

closed 2019-03-10 by dhoomakethu